aws/amazon-freertos #1829: Issue running nrf52840 FreeRTOS demo

Abz23, 2020-03-12:
Hi,
I am trying to setup the FreeRTOS demo for the nordic NRF52840 DK but with no luck so far.
After I manged to setup and configure the BLE demo on android studio, the application manages to connect using BLE with DK but then the following error keeps occurring ending the demo:

"UNKNOWN BLE EVENT 35"

I have followed the getting started instructions and have added the AWS endpoint and IoT thing name. Also, I have made sure to enable the MQTT and Bluetooth Low Energy GATT demos by including the defines as stated. I have included the outputs I see from the debugger on the nrf52840 DK and android studio below.

Any guidance would be greatly appreciated. 

0 0 [iot_thread] [INFO ][DEMO][0] ---------STARTING DEMO---------

The RAM must start at 200046f8, it's size must be 3b908
1 0 [iot_thread] [INFO ][INIT][0] SDK successfully initialized.
2 41 [iot_thread] [INFO ][DEMO][410] No networks connected for the demo. Waiting for a network connection. 
3 909 [BLE] [ERROR][BLE_HAL][9090] PM_UNKNOWN, 0.
4 909 [BLE] [ERROR][BLE_HAL][9090] sd_ble_gattc_exchange_mtu_request() on connection 0x0 returned busy, will retry.
5 909 [BLE] [INFO ][DEMO][9090] BLE Connected to remote device, connId = 0

6 909 [iot_thread] [INFO ][DEMO][9090] Successfully initialized the demo. Network type for the demo: 2
7 909 [iot_thread] [INFO ][MQTT][9090] MQTT library successfully initialized.
8 909 [iot_thread] [INFO ][DEMO][9090] MQTT demo client identifier is demo (length 4).
9 1031 [BLE] [ERROR][BLE_HAL][10310] UNKNOWN BLE EVENT 35.
10 1036 [iot_thread] [INFO ][MQTT][10360] Establishing new MQTT connection.
11 1036 [iot_thread] [INFO ][MQTT][10360] (MQTT connection 20017568, CONNECT operation 20017888) Waiting for operation completion.
12 1536 [iot_thread] [INFO ][MQTT][15360] (MQTT connection 20017568, CONNECT operation 20017888) Wait complete with result TIMEOUT.
13 1536 [iot_thread] [ERROR][MQTT][15360] Failed to establish new MQTT connection, error TIMEOUT.
14 1536 [iot_thread] [INFO ][MQTT][15360] Network connection closed on error.
15 1536 [iot_thread] [INFO ][MQTT][15360] (MQTT connection 20017568) Network connection destroyed.
16 1536 [iot_thread] [ERROR][DEMO][15360] MQTT CONNECT returned error TIMEOUT.
17 1536 [iot_thread] [INFO ][MQTT][15360] MQTT library cleanup done.
18 1536 [iot_thread] [INFO ][DEMO][15360] memory_metrics::freertos_heap::before::bytes::125128
19 1536 [iot_thread] [INFO ][DEMO][15360] memory_metrics::freertos_heap::after::bytes::106224
20 1536 [iot_thread] [INFO ][DEMO][15360] memory_metrics::demo_task_stack::before::bytes::19060
21 1536 [iot_thread] [INFO ][DEMO][15360] memory_metrics::demo_task_stack::after::bytes::18096
22 1536 [iot_thread] [ERROR][DEMO][15360] Error running demo.
23 1548 [iot_thread] [INFO ][INIT][15480] SDK cleanup done.
24 1548 [iot_thread] [INFO ][DEMO][15480] -------DEMO FINISHED-------

03/12 11:54:01: Launching 'app' on samsung SM-G950U.
$ adb shell am start -n "software.amazon.freertos.demo/software.amazon.freertos.demo.DeviceScanActivity" -a android.intent.action.MAIN -c android.intent.category.LAUNCHER
Connected to process 32174 on device 'samsung-sm_g950u-988b9d46345531314c'.
Capturing and displaying logcat messages from application. This behavior can be disabled in the "Logcat output" section of the "Debugger" settings page.
I/DecorView: createDecorCaptionView >> DecorView@663a1f0[], isFloating: false, isApplication: true, hasWindowDecorCaption: false, hasWindowControllerCallback: true
D/SignInView: Using defaults: backgroundColor = -12303292; logoResId = 2131099731
    Background Color : -12303292
    Logo : 2131099731
D/SignInView: Trying to create an instance of UserPoolSignInView
D/SignInView: Skipping creating the SignInButtons. No SignInbuttons were added to the view.
D/InputTransport: Input channel constructed: fd=85
D/ViewRootImpl@cebc244[SignInActivity]: setView = DecorView@663a1f0[SignInActivity] TM=true MM=false
D/ViewRootImpl@43ec642[AuthenticatorActivity]: MSG_WINDOW_FOCUS_CHANGED 0 1
D/InputMethodManager: prepareNavigationBarInfo() DecorView@b145a60[AuthenticatorActivity]
    getNavigationBarColor() -855310
D/ViewRootImpl@cebc244[SignInActivity]: dispatchAttachedToWindow
D/ViewRootImpl@cebc244[SignInActivity]: Relayout returned: old=[0,0][1080,2220] new=[0,0][1080,2220] result=0x7 surface={valid=true 543181852672} changed=true
D/OpenGLRenderer: eglCreateWindowSurface = 0x7e78744780, 0x7e78295010
D/OpenGLRenderer: eglDestroySurface = 0x7e7854d380, 0x7e784c2000
D/ViewRootImpl@43ec642[AuthenticatorActivity]: Relayout returned: old=[0,0][1080,2220] new=[0,0][1080,2220] result=0x5 surface={valid=false 0} changed=true
D/ViewRootImpl@cebc244[SignInActivity]: MSG_RESIZED: frame=Rect(0, 0 - 1080, 2220) ci=Rect(0, 81 - 0, 0) vi=Rect(0, 81 - 0, 0) or=1
    MSG_WINDOW_FOCUS_CHANGED 1 1
D/InputMethodManager: prepareNavigationBarInfo() DecorView@663a1f0[SignInActivity]
    getNavigationBarColor() -855310
D/InputMethodManager: prepareNavigationBarInfo() DecorView@663a1f0[SignInActivity]
    getNavigationBarColor() -855310
V/InputMethodManager: Starting input: tba=software.amazon.freertos.demo ic=com.android.internal.widget.EditableInputConnection@da720a5 mNaviBarColor -855310 mIsGetNaviBarColorSuccess true , NavVisible : true , NavTrans : false
D/InputMethodManager: startInputInner - Id : 0
I/InputMethodManager: startInputInner - mService.startInputOrWindowGainedFocus
D/InputTransport: Input channel constructed: fd=86
D/InputTransport: Input channel destroyed: fd=83
[... 350 lines cut ...]
D/FRD: ->->-> Characteristic changed for: MQTT_TX with data: a461770161646464656d6f61
    Processing incoming queue. size: 0
I/FRD: Handling Mqtt Message type : 1
I/FRD: Connecting to IoT using AWS credential: a21g22klfb6x4u-ats.iot.us-west-2.amazonaws.com
D/AWSMobileClient: Inspecting user state details
D/AWSMobileClient: waitForSignIn: userState:SIGNED_OUT
D/CognitoCachingCredentialsProvider: Loading credentials from SharedPreferences
    No valid credentials found in SharedPreferences
    Making a network call to fetch credentials.
I/System.out: (HTTPLog)-Static: isSBSettingEnabled false
I/System.out: (HTTPLog)-Static: isSBSettingEnabled false
D/CognitoCachingCredentialsProvider: Identity id is changed
    Saving identity id to SharedPreferences
D/CognitoCachingCredentialsProvider: Clearing credentials from SharedPreferences
I/System.out: (HTTPLog)-Static: isSBSettingEnabled false
I/System.out: (HTTPLog)-Static: isSBSettingEnabled false
I/FRD: mqtt connection status changed to: ConnectionLost
I/AmazonFreeRTOSManager: Stopping ble device scan
D/BluetoothAdapter: STATE_ON
D/BluetoothLeScanner: Stop Scan with callback
D/BluetoothGatt: onClientConnectionState() - status=19 clientIf=10 device=D0:3B:BE:E6:E8:EC
I/FRD: BLE connection state changed: 19; new state: BLE_DISCONNECTED
    Disconnected from GATT server.
I/DeviceScanFragment: BLE connection status changed to: BLE_DISCONNECTED
I/DeviceScanFragment: connect switch isChecked: OFF

ravibhagavandas, 2020-03-12:
```
Caused by: com.amazonaws.AmazonServiceException: 1 validation error detected: Value '6eaf0258-e8c5-467f-90de-6e212326e6d3' at 'identityPoolId' failed to satisfy constraint: Member must satisfy regular expression pattern: [\w-]+:[0-9a-f-]+ (Service: AmazonCognitoIdentity; Status Code: 400; Error Code: ValidationException; Request ID: c246f5d2-de6b-40c8-b106-2594e7fdcb5a)
```
The above error from Android SDK logs shows the Identity Pool ID configured in the application is of incorrect format. 

Identity pool id can be found when you click through following links from AWS console: `Cognito -> Manage User Pools -> Your User Pool name ->  General Settings -> Pool Id`
Identity pool id starts with the aws region name for example `us_west_2_`. 

More details about the same, can be found in the [documentation](https://docs.aws.amazon.com/freertos/latest/userguide/ble-demo.html#ble-sdk-app) under section **To configure the Android SDK demo application**.

Abz23, 2020-03-12:
Thank you for getting back to me so soon.

I have corrected that issues but now the App does not show the "Scan for a ble device" button, and it seems to get stuck almost instantly after logging in to AWS.

I am getting this new error were it keeps failing to "federate the tokens". Can you please shed some light on that?  
The error log from the android app is below.

W/AWSMobileClient: Failed to federate the tokens
    com.amazonaws.services.cognitoidentity.model.NotAuthorizedException: Token is not from a supported provider of this identity pool. (Service: AmazonCognitoIdentity; Status Code: 400; Error Code: NotAuthorizedException; Request ID: a8bb31c2-5feb-41fd-92c3-136d61771fa1)
        at com.amazonaws.http.AmazonHttpClient.handleErrorResponse(AmazonHttpClient.java:730)
        at com.amazonaws.http.AmazonHttpClient.executeHelper(AmazonHttpClient.java:405)
        at com.amazonaws.http.AmazonHttpClient.execute(AmazonHttpClient.java:212)
        at com.amazonaws.services.cognitoidentity.AmazonCognitoIdentityClient.invoke(AmazonCognitoIdentityClient.java:1658)
        at com.amazonaws.services.cognitoidentity.AmazonCognitoIdentityClient.getId(AmazonCognitoIdentityClient.java:739)
        at com.amazonaws.auth.AWSAbstractCognitoIdentityProvider.getIdentityId(AWSAbstractCognitoIdentityProvider.java:172)
        at com.amazonaws.mobile.client.AWSMobileClientCognitoIdentityProvider.refresh(AWSMobileClient.java:3600)
        at com.amazonaws.auth.CognitoCredentialsProvider.startSession(CognitoCredentialsProvider.java:678)
        at com.amazonaws.auth.CognitoCredentialsProvider.refresh(CognitoCredentialsProvider.java:631)
        at com.amazonaws.auth.CognitoCachingCredentialsProvider.refresh(CognitoCachingCredentialsProvider.java:510)
        at com.amazonaws.mobile.client.AWSMobileClient.federateWithCognitoIdentity(AWSMobileClient.java:1515)
        at com.amazonaws.mobile.client.AWSMobileClient.getUserStateDetails(AWSMobileClient.java:1046)
        at com.amazonaws.mobile.client.AWSMobileClient.waitForSignIn(AWSMobileClient.java:907)
        at com.amazonaws.mobile.client.AWSMobileClient.getCredentials(AWSMobileClient.java:359)
        at com.amazonaws.services.iot.AWSIotClient.invoke(AWSIotClient.java:9213)
        at com.amazonaws.services.iot.AWSIotClient.attachPolicy(AWSIotClient.java:589)
        at software.amazon.freertos.demo.AuthenticatorActivity.signinsuccessful(AuthenticatorActivity.java:114)
        at software.amazon.freertos.demo.AuthenticatorActivity.access$000(AuthenticatorActivity.java:22)
        at software.amazon.freertos.demo.AuthenticatorActivity$2.run(AuthenticatorActivity.java:66)
        at android.os.Handler.handleCallback(Handler.java:873)
        at android.os.Handler.dispatchMessage(Handler.java:99)
        at android.os.Looper.loop(Looper.java:214)
        at android.os.HandlerThread.run(HandlerThread.java:65)
D/AWSMobileClient: waitForSignIn: userState:SIGNED_OUT_USER_POOLS_TOKENS_INVALID

Abz23, 2020-03-12:
That seems to have fixed itself as soon as I restarted my phone.

However, the ble hal error still persists in the nrf52840. 

I have included the logs of the android app and nrf52840. Any help would be appreciated. 

Thanks

//////////////////////////////////////////////////////////////////////////////////////////////////////////////////
0 0 [iot_thread] [INFO ][DEMO][0] ---------STARTING DEMO---------

The RAM must start at 200044b0, it's size must be 3bb50
1 0 [iot_thread] [INFO ][INIT][0] SDK successfully initialized.
2 27 [iot_thread] [INFO ][DEMO][270] No networks connected for the demo. Waiting for a network connection. 
3 720 [BLE] [ERROR][BLE_HAL][7200] PM_UNKNOWN, 0.
4 720 [BLE] [ERROR][BLE_HAL][7200] sd_ble_gattc_exchange_mtu_request() on connection 0x0 returned busy, will retry.
5 720 [BLE] [INFO ][DEMO][7200] BLE Connected to remote device, connId = 0

6 721 [iot_thread] [INFO ][DEMO][7210] Successfully initialized the demo. Network type for the demo: 2
7 721 [iot_thread] [INFO ][MQTT][7210] MQTT library successfully initialized.
8 721 [iot_thread] [INFO ][DEMO][7210] MQTT demo client identifier is demo (length 4).
9 843 [BLE] [ERROR][BLE_HAL][8430] UNKNOWN BLE EVENT 35.
10 848 [iot_thread] [INFO ][MQTT][8480] Establishing new MQTT connection.
11 848 [iot_thread] [INFO ][MQTT][8480] (MQTT connection 20017568, CONNECT operation 20019e90) Waiting for operation completion.
12 1348 [iot_thread] [INFO ][MQTT][13480] (MQTT connection 20017568, CONNECT operation 20019e90) Wait complete with result TIMEOUT.
13 1348 [iot_thread] [ERROR][MQTT][13480] Failed to establish new MQTT connection, error TIMEOUT.
14 1348 [iot_thread] [INFO ][MQTT][13480] Network connection closed on error.
15 1348 [iot_thread] [INFO ][MQTT][13480] (MQTT connection 20017568) Network connection destroyed.
16 1348 [iot_thread] [ERROR][DEMO][13480] MQTT CONNECT returned error TIMEOUT.
17 1348 [iot_thread] [INFO ][MQTT][13480] MQTT library cleanup done.
18 1348 [iot_thread] [INFO ][DEMO][13480] memory_metrics::freertos_heap::before::bytes::125128
19 1348 [iot_thread] [INFO ][DEMO][13480] memory_metrics::freertos_heap::after::bytes::96488
20 1348 [iot_thread] [INFO ][DEMO][13480] memory_metrics::demo_task_stack::before::bytes::19060
21 1348 [iot_thread] [INFO ][DEMO][13480] memory_metrics::demo_task_stack::after::bytes::18096
22 1348 [iot_thread] [ERROR][DEMO][13480] Error running demo.
23 1354 [BLE] [INFO ][DEMO][13540]  Disconnected from BLE device. Stopping the counter update 

24 1354 [iot_thread] [INFO ][INIT][13540] SDK cleanup done.
25 1354 [iot_thread] [INFO ][DEMO][13540] -------DEMO FINISHED-------

//////////////////////////////////////////////////////////////////////////////////////////////////////////////////
Android app log
/////////////////////////////////////////////////////////////////////////////////////////////////////////////////
D/InputTransport: Input channel constructed: fd=87
    Input channel destroyed: fd=86
D/ViewRootImpl@4992b65[SignInActivity]: ViewPostIme pointer 0
D/ViewRootImpl@4992b65[SignInActivity]: ViewPostIme pointer 1
I/CognitoUserPoolsSignInProvider: Logged in. com.amazonaws.mobileconnectors.cognitoidentityprovider.tokens.CognitoIdToken@df82ff7
D/IdentityManager: SignInProviderResultAdapter.onSuccess(): Amazon Cognito Your User Pools provider sign-in succeeded.
    federate with provider: Populate loginsMap with token.
D/IdentityManager: SignInProviderResultAdapter.onCognitoSuccess()
D/AWSMobileClient: onUserSignedIn: Updating user state from drop-in UI
    _federatedSignIn: Putting provider and token in store
I/SignInActivity: Sign-in with Amazon Cognito Your User Pools succeeded.
D/SignInUI: Sign-in succeeded. The identity provider name is available here using: Amazon Cognito Your User Pools
D/ViewRootImpl@e61fb27[AuthenticatorActivity]: dispatchDetachedFromWindow
D/ViewRootImpl@e61fb27[AuthenticatorActivity]: Surface release. android.view.ViewRootImpl.doDie:7967 android.view.ViewRootImpl.die:7935 android.view.WindowManagerGlobal.removeViewLocked:497 android.view.WindowManagerGlobal.removeView:435 android.view.WindowManagerImpl.removeViewImmediate:124 android.app.ActivityThread.handleDestroyActivity:4776 android.app.servertransaction.DestroyActivityItem.execute:39 android.app.servertransaction.TransactionExecutor.executeLifecycleState:145 
D/InputTransport: Input channel destroyed: fd=82
W/ActivityThread: handleWindowVisibility: no activity for token android.os.BinderProxy@3d058cd
D/ViewRootImpl@4992b65[SignInActivity]: MSG_WINDOW_FOCUS_CHANGED 0 1
D/InputMethodManager: prepareNavigationBarInfo() DecorView@17e321[SignInActivity]
D/InputMethodManager: getNavigationBarColor() -855310
I/DecorView: createDecorCaptionView >> DecorView@e16b4c9[], isFloating: false, isApplication: true, hasWindowDecorCaption: false, hasWindowControllerCallback: true
D/CognitoCachingCredentialsProvider: Clearing credentials from SharedPreferences
    Identity id is changed
    Saving identity id to SharedPreferences
D/CognitoCachingCredentialsProvider: Clearing credentials from SharedPreferences
I/chatty: uid=11223(software.amazon.freertos.demo) Thread-9 identical 1 line
D/CognitoCachingCredentialsProvider: Clearing credentials from SharedPreferences
D/AuthActivity: waiting for AWSMobileClient Initialization
[... 225 lines cut ...]
I/FRD: onMTUChanged : 247 status: Success
I/FRD: Enabling Wifi provisioning
D/FRD: Ble operation is in progress. mqtt queue: 0 network queue: 1
I/FRD: Enabling MQTT Proxy
[... cut ...]
I/FRD: Connecting to IoT using AWS credential: a21g22klfb6x4u-ats.iot.us-west-2.amazonaws.com
I/AmazonFreeRTOSManager: Stopping ble device scan
D/BluetoothAdapter: STATE_ON
D/BluetoothLeScanner: Stop Scan with callback
D/BluetoothGatt: onClientConnectionState() - status=19 clientIf=10 device=D0:3B:BE:E6:E8:EC
I/FRD: BLE connection state changed: 19; new state: BLE_DISCONNECTED
    Disconnected from GATT server.
I/DeviceScanFragment: BLE connection status changed to: BLE_DISCONNECTED
I/DeviceScanFragment: connect switch isChecked: OFF

ravibhagavandas, 2020-03-12:
```W/FRD: There's no such service found with uuid: a9d7166a-d72e-40a9-a002-48044cc30100```
This UUID is for WiFi provisioning service, which is not enabled in Nordic (as it does not have a WiFi stack), so this should not be the cause of failure.

 ```UNKNOWN BLE EVENT 35```  is for one of the events which is not expected from our porting layer, so this message could be ignored.

I think the real issue is user is not logged into the cognito yet:
```
at com.amazonaws.mobile.client.internal.InternalCallback$1.run(InternalCallback.java:101) 
at java.lang.Thread.run(Thread.java:764) 
D/AWSMobileClient: Inspecting user state details
D/AWSMobileClient: Inspecting user state details
D/AuthActivity: onResult: SIGNED_OUT
I/AuthActivity: onResult: User did not choose to sign-in
D/AWSMobileClient: run: showSignIn completed
W/AWSMobileClient: Failed to federate the tokens
com.amazonaws.services.cognitoidentity.model.NotAuthorizedException: Token is not from a supported 
```
From the error it seems the federated token provided is invalid. Did you create a user id and password for the cognito signin ?

Abz23, 2020-03-13:
yes i have created user id and password for cognito signin and I have made sure to input the poolID and other details required correctly. So am not sure why this issue keeps occuring?

ravibhagavandas, 2020-03-13:
@Abz23 

There is a minor correction on how to obtain the identity pool id from the previous post:
Identity pool id can be obtained as follows: `Cognito -> Manage Identity Pools -> Federated Identities -> <Identity Pool Name> -> Edit Identity Pool  (on top right hand corner) `. Identity pool id should be listed here, it starts with `<aws_region>:`

Additionally, [awsconfiguration.json](https://github.com/aws/amazon-freertos-ble-android-sdk/blob/master/app/src/main/res/raw/awsconfiguration.json) in Android SDK demo app has instructions on how to fetch each configuration parameters from console. Could you confirm if all parameters are set as per the documentation in this file?

Abz23, 2020-03-16:
Hi,
I have double checked the parameters and I can confirm that they are correct.

This error below is my issue at the moment. 

E/AuthActivity: Exception caught: 
    com.amazonaws.AmazonServiceException: User: arn:aws:sts::566230376049:assumed-role/Cognito_DemoAppAuth_Role/CognitoIdentityCredentials is **not authorized to perform: iot:AttachPolicy on resource:** us-west-2:5b0d0c13-ce3a-46f0-8d22-a142378c8468 (Service: AWSIot; Status Code: 403; Error Code: AccessDeniedException; Request ID: 05e8f21e-cf8e-4a14-b32b-0c7cf793ea25)
        at com.amazonaws.http.AmazonHttpClient.handleErrorResponse(AmazonHttpClient.java:730)
        at com.amazonaws.http.AmazonHttpClient.executeHelper(AmazonHttpClient.java:405)
        at com.amazonaws.http.AmazonHttpClient.execute(AmazonHttpClient.java:212)
        at com.amazonaws.services.iot.AWSIotClient.invoke(AWSIotClient.java:9226)
        at com.amazonaws.services.iot.AWSIotClient.attachPolicy(AWSIotClient.java:589)
        at software.amazon.freertos.demo.AuthenticatorActivity.signinsuccessful(AuthenticatorActivity.java:114)
        at software.amazon.freertos.demo.AuthenticatorActivity.access$000(AuthenticatorActivity.java:22)
        at software.amazon.freertos.demo.AuthenticatorActivity$2.run(AuthenticatorActivity.java:66)
        at android.os.Handler.handleCallback(Handler.java:873)
        at android.os.Handler.dispatchMessage(Handler.java:99)
        at android.os.Looper.loop(Looper.java:214)
        at android.os.HandlerThread.run(HandlerThread.java:65)

ravibhagavandas, 2020-03-16:
From the error I think the authenticated identity is missing permissions to attach a policy to publish/subscribe to AWS IoT.
Section **To create and attach an IAM policy to the authenticated identity** within the [documention](https://docs.aws.amazon.com/freertos/latest/userguide/ble-demo.html#ble-sdk-app) tells how to attach the policy to authenticated identity roles.

closed 2020-04-13 by dcgaws