aws/amazon-freertos #1830: Including whole Amazon FreeRTOS library rather than only a part of it.

heshananupama, 2020-03-12:
Hi,

First of all, I know that you would think that why someone would need such a thing when I am gonna ask this question.

So my need is to have all the AmazonFreeRTOS library included in my binary though I am using few functions from it. (for example, lets say I am using MQTT demo application that is already included. Although it may only use few functions from the actual library, I need to put all the library inside the final executable rather than only the functions that is been used.) Yeah! you may definitely think that what's the reason to do so. Basically, for my research work, I need to build an executable so that it include all the Amazon FreeRTOS in it. It would be very convenient and easy for me if I have a way to put all the library rather than trying to use each and every part of it manually in my application.

Through the system workbench, I have been using '-Wl,--whole-archive' option to instruct the linker but I was unable to achieve my goal. To be specific, the demo application I am trying to build is for STMicro Electronics. Is there any way that I could get to my goal? May be trying with linker or a compiler flag, or any other way?

Thank you so much!

jnz86, 2020-03-12:
Just a head's up, but you never wrote which compiler/IDE you're using. Also I'd recommend stackoverflow as a generic question.

heshananupama, 2020-03-12:
I also posted on stack-overflow but did not got the answer that I was needed. The IDE is ST Microelectronics recommended System Workbench (According to the user guide) which is actually an eclipse distribution.

TheIronNinja, 2020-03-13:
I'd suggest reading the thread at #832 as it's pretty illustrating about how to use Amazon FreeRTOS as an entire library. If you follow along you'll get there, but note that it's no trivial task. And also, beware of differences in platforms, as I believe most of the comments there are from an ESP32 perspective.

heshananupama, 2020-03-13:
Thank you for the reply. I went through the link and it seems complex to me may be because I am a beginner to the area. Other than that, isn't there a way to compile via the IDE itself? Thanks

TheIronNinja, 2020-03-13:
You can use the [Getting Started](https://docs.aws.amazon.com/freertos/latest/userguide/getting_started_st.html) guide for your device as an starting point, but maybe you'll have to explore beyond that. I'm not familiar with this device, so I'm sorry I can't help more than that.

dachalco, 2020-03-14:
Hello @heshananupama 

Please note that thread #832 has been summarized in a more approachable manner in [Using FreeRTOS in Your Own CMake Project for ESP32](https://docs.aws.amazon.com/freertos/latest/userguide/getting_started_espressif.html#getting_started_espressif_cmake_project). 

The STM32L4 does support cmake builds as shown in the _Using CMake with FreeRTOS_ section of the [STM32L4 Getting Started Guide](https://docs.aws.amazon.com/freertos/latest/userguide/getting_started_st.html). Please note that you may need to modify some of the steps in [Using FreeRTOS in Your Own CMake Project for ESP32](https://docs.aws.amazon.com/freertos/latest/userguide/getting_started_espressif.html#getting_started_espressif_cmake_project) guide to accommodate the ST board and additional cmake settings may be necessary. 

Finally, if you really want to include all symbols/code inside your final executable, including unused symbols/code, you'll probably need a few things: 
- Use static libraries for linking as they are linked in compilation, versus dynamic libraries which are linked in runtime through separate *.so/*.dll files.

- Be sure that the linker does not strip unused symbols/code. You may need to alter linker options.

If you can, would you please elaborate more on your higher-level goal? Perhaps with more details we can assist you with an easier solution that does not require linking unused symbols/code.

heshananupama, 2020-03-15:
Thank you very much for the reply. My goal is to create a demo application, including all the amazon FreeRTOS library rather than only a part of the library. I know CMake has been integrated much recent versions of FreeRTOS and I for my project, I need to build demo applications with a number of older versions too. Due to this fact, I am using System Workbench as I could simply create the executable regardless of older versions or newer versions. 

I think  found a solution-: There is an linker option through the IDE to explicitly say not to discard the unused sections (-Wl --gc-sections) and I feel like it works. )

dachalco, 2020-03-16:
@heshananupama 
It's our pleasure to help where we can! Glad you may have found a sufficient solution. If you want to be very sure that some symbol is in your final compiled ELF, you can use a command line tool such `readelf` to inspect contents. 

If you have any further questions please feel free to post on this thread.

heshananupama, 2020-03-17:
Yup Thanks a lot!

closed 2020-03-17 by heshananupama