aws/amazon-freertos #1831: Common IO doesn't get build by cmake

teuteuguy, 2020-03-13:
**Describe the bug**
When building projects with cmake, common-io doesnt get picked up in the build process.

Note: yes ESP32 port is not final, but have been working on this code on the side, but despite the lack of code, the build process still doesnt take the module into account.

**System information**
- Which hardware board or part numbers? ESP32
- IDE used: vscode
- Operating System: MacOS
- Version of FreeRTOS: latest master
- Project: Custom Application

**Expected behavior**
The common-io should be picked up by the build process.

**Found solution for now**
When comparing the common-io CMakeLists.txt with the wifi one, there are differences in that file.
Making changes to that file, to mimic the wifi one, fixes the problem.
- afr_module(INTERFACE) to afr_module()
- adding the metadata
- changing from INTERFACE to PUBLIC in the module include dirs
- INTERFACE TO PRIVATE in the module dependencies

teuteuguy, 2020-03-13:
Changing https://github.com/aws/amazon-freertos/blob/master/libraries/abstractions/common_io/CMakeLists.txt to the following implementation, seems to solve the problem:

afr_module()

afr_set_lib_metadata(ID "common_io")
afr_set_lib_metadata(DESCRIPTION "This library implements an abstraction layer for I2C, SPI and UART.")
afr_set_lib_metadata(DISPLAY_NAME "COMMON IO")
afr_set_lib_metadata(CATEGORY "BSP")
afr_set_lib_metadata(VERSION "1.0.0")
afr_set_lib_metadata(IS_VISIBLE "true")

set(inc_dir "${CMAKE_CURRENT_LIST_DIR}/include")
set(test_dir "${CMAKE_CURRENT_LIST_DIR}/test")

afr_module_sources(
    ${AFR_CURRENT_MODULE}
    INTERFACE
        "${inc_dir}/iot_i2c.h"
        "${inc_dir}/iot_uart.h"
        "${inc_dir}/iot_spi.h"
)

afr_module_include_dirs(
    ${AFR_CURRENT_MODULE}
    PUBLIC
        "${inc_dir}"
)

afr_module_dependencies(
    ${AFR_CURRENT_MODULE}
    PRIVATE AFR::common_io::mcu_port
)

# Common I/O tests
afr_test_module()
afr_module_sources(
    ${AFR_CURRENT_MODULE}
    INTERFACE
        "${test_dir}/iot_test_common_io_internal.h"
        "${test_dir}/iot_test_common_io.c"
        "${test_dir}/test_iot_i2c.c"
        "${test_dir}/test_iot_uart.c"
        "${test_dir}/test_iot_spi.c"
)

afr_module_include_dirs(
    ${AFR_CURRENT_MODULE}
    INTERFACE "${test_dir}"
)

afr_module_dependencies(
    ${AFR_CURRENT_MODULE}
    INTERFACE
        AFR::common_io
)

tgsong, 2020-03-13:
Confirmed the issue is due to the use of `INTERFACE` in `afr_module`, when `INTERFACE` is used, CMake will not create a build target, so this could be a mistake. @qiutongs Can you verify the issue?

lundinc2, 2020-04-03:
Noting this breaks the ST cmake build. So we will need to address that in the update

qiutongs, 2020-04-23:
@teuteuguy Feel free to re-open this if there is still problem.

closed 2020-04-23 by qiutongs