aws/amazon-freertos #1832: ERROR: Handshake failed with error code -29312

KunjalRupala, 2020-03-13:
**Describe the bug**
I am trying to run a demo code of greengrass connectivity but whenever I try to flash it, I get "ERROR: Handshake failed with error code -29312"

- Hardware board : ESP32
- Operating System  : Linux
- Version of FreeRTOS : 201912.00-116-gc393d4658
- Project : Greengrass

I am attaching a screenshot of the same.

[image]

Link to log: https://gist.github.com/KunjalRupala/ddcd43c88166cd0642c3e330e0018ee8

Thank you!

sameerdasarwad, 2020-03-13:
look like something wrong in certificates  but cant say exact reason  if you on mqtt debug will be useful , turn mqtt debug for additional logs  in config-> aws_mqtt_config.h

dachalco, 2020-03-13:
@sameerdasarwad 
Thank you for your suggestions. It's great to see the community working together!

@KunjalRupala 
The error in red indicates an attempt to retrieve an object from NVS memory that either does not exist, or is at least empty. Excluding keys and confidential data, could you please provide more details on your configurations in 
`amazon-freertos/vendors/espressif/boards/esp32/aws_demos/config_files/iot_pkcs11_config.h`, 
or of a similar pkcs11 configuration file in use?

@lundinc2 Could you also please take a look at this?

KunjalRupala, 2020-03-14:
@sameerdasarwad 
I tried what you suggested but I think it's just the same as it was. I am attaching my logs here.
https://gist.github.com/KunjalRupala/a5f728d68b14e581555e3a6c2b67c9d7

@dachalco 
Here am attaching my pkcs11 configuration file.
https://gist.github.com/KunjalRupala/82c0839936496111442b54c7b89ea97d

sameerdasarwad, 2020-03-14:
@dachalco I'm happy to help 
@KunjalRupala  i just see logs found nothing but i deal with similar  problem ,  
can you enable mbedTLS debugging from menu config ->  component config -> mbedTLS

KunjalRupala, 2020-03-16:
@dachalco Here it is https://gist.github.com/KunjalRupala/bcc928800338c3355cdf1d2d84b117e8

KunjalRupala, 2020-03-16:
The demo code runs as expected when I shift from my router to mobile network. It shows this error only when I am connected to my router.

dachalco, 2020-03-16:
@KunjalRupala 

If it works on your mobile network but not the router, it may be a firewall issue or other router setting. Do you notice any settings that differ between your mobile and router, such as port restrictions?

Some additional info:
- mbedTLS debug messages can be enabled by defining [this macro](https://github.com/aws/amazon-freertos/blob/master/libraries/3rdparty/mbedtls/include/mbedtls/config.h#L2121), and following [these instructions.](https://tls.mbed.org/kb/development/debugging-tls)
- A tool like wireshark can be used to filter and view these packet transactions.

lundinc2, 2020-04-09:
@KunjalRupala I am going to close this issue, as it has become inactive. Please reopen it should you need further assistance.

closed 2020-04-09 by lundinc2