aws/amazon-freertos #1833: difference between Aws Iot Shadow _Get() and  SHADOW _Get() how they works

sameerdasarwad, 2020-03-14:
This is not any issue its just informational question i know i can put this on stack overflow but it is better place where i can find perfect answer,
i just want know there working and definition  if both are same then why two ?

dachalco, 2020-03-17:
Hi @sameerdasarwad 

`SHADOW_Get` belongs to a higher level [`aws_shadow.h API`](https://github.com/aws/amazon-freertos/blob/master/libraries/c_sdk/aws/shadow/include/aws_shadow.h#L415) that uses the [`aws_iot_shadow.h API`](https://github.com/aws/amazon-freertos/blob/master/libraries/c_sdk/aws/shadow/include/aws_iot_shadow.h#L337) containing `AwsIotShadow_Get`.

They both can be used to get data related to [`AwsIotShadowDocumentInfo_t`](https://github.com/aws/amazon-freertos/blob/master/libraries/c_sdk/aws/shadow/include/types/aws_iot_shadow_types.h#L454): 
```c
typedef struct AwsIotShadowDocumentInfo
{
    const char * pThingName; /**< @brief The Thing Name associated with this Shadow document. */
    size_t thingNameLength;  /**< @brief Length of #AwsIotShadowDocumentInfo_t.pThingName. */

    IotMqttQos_t qos;        /**< @brief QoS when sending a Shadow get or update message. See #IotMqttPublishInfo_t.qos. */
    uint32_t retryLimit;     /**< @brief Maximum number of retries for a Shadow get or update message. See #IotMqttPublishInfo_t.retryLimit. */
    uint32_t retryMs;        /**< @brief First retry time for a Shadow get or update message. See IotMqttPublishInfo_t.retryMs. */

    union
    {
        /* Valid for Shadow get. */
        struct
        {
            /**
             * @brief Function to allocate memory for an incoming Shadow document.
             *
             * This only needs to be set if #AWS_IOT_SHADOW_FLAG_WAITABLE is passed to
             * @ref shadow_function_get.
             */
            void * ( *mallocDocument )( size_t );
        } get; /**< @brief Valid members for @ref shadow_function_get. */

        /* Valid for Shadow update. */
        struct
        {
            const char * pUpdateDocument; /**< @brief The Shadow document to send in the update. */
            size_t updateDocumentLength;  /**< @brief Length of Shadow update document. */
        } update;                         /**< @brief Valid members for @ref shadow_function_update. */
    } u;                                  /**< @brief Valid member depends on operation type. */
} AwsIotShadowDocumentInfo_t;
```
however `SHADOW_Get` only utilizes a subset of `AwsIotShadowDocumentInfo_t` as it uses [`ShadowOperationsParams_t`](https://github.com/aws/amazon-freertos/blob/master/libraries/c_sdk/aws/shadow/include/aws_shadow.h#L144) to store its results.
```c
typedef struct ShadowOperationParams
{
    /** @brief Thing name of Shadow. */
    const char * pcThingName;

    /**
     * @brief Input parameter for #SHADOW_Update; output parameter of #SHADOW_Get.
     * - For #SHADOW_Update, set this to point to the Shadow update document.
     * - For #SHADOW_Get, this is the output parameter pointing to a buffer which
     * holds the Shadow document. If #SHADOW_Get fails, this will be set to NULL.
     * - This member is ignored by #SHADOW_Delete. */
    const char * pcData;

    /**
     * @brief Input parameter for #SHADOW_Update; output parameter of #SHADOW_Get.
     * - For #SHADOW_Update, set this to length of the Shadow update document.
     * - For #SHADOW_Get, this is the output parameter specifying the length of
     * #ShadowOperationParams_t.pcData. If #SHADOW_Get fails, this will be set
     * to @c 0.
     * @note If the received Shadow document is longer than this value, the
     * document will be truncated. A warning message will be printed if
     * #shadowENABLE_DEBUG_LOGS is @c 1.
     * - This member is ignored by #SHADOW_Delete. */
    uint32_t ulDataLength;

    /**
     * @brief Output parameter of #SHADOW_Get.
     * - This member is ignored by #SHADOW_Update.
     * - For #SHADOW_Get, this is the output parameter for the MQTT Buffer Handle.
     * This buffer must be returned to the MQTT Client after the user is done
     * with the received document by calling #SHADOW_ReturnMQTTBuffer.
     * - This member is ignored by #SHADOW_Delete. */
    MQTTBufferHandle_t xBuffer;

    /**
     * @brief Controls whether topic subscriptions remain active after a Shadow
     * operation completes.
     *
     * Allows the MQTT subscriptions of the operation to remain active if set
     * to @c 1, saving time if the same operation is performed again. Set this
     * value to @c 0 to deactivate the operation's MQTT subscriptions after the
     * operation completes.
     * @warning Users may be billed for extraneous messages received on an
     * operation's MQTT topics. If other clients are publishing to the same topics,
     * it is best to deactivate the subscriptions. */
    uint8_t ucKeepSubscriptions;

    /**
     * @brief MQTT qos when publishing from the Shadow Client @b to the Shadow Service.
     * @note The Shadow services always publishes to the Shadow Client with qos 1. */
    MQTTQoS_t xQoS;
} ShadowOperationParams_t;
```

You can see `SHADOW_Get` mapping these data members [here](https://github.com/aws/amazon-freertos/blob/master/libraries/c_sdk/aws/shadow/src/aws_shadow.c#L500) and [here](https://github.com/aws/amazon-freertos/blob/master/libraries/c_sdk/aws/shadow/src/aws_shadow.c#L521), for example. 

Besides their different interfaces, a key difference is that `SHADOW_Get` is a **_blocking_** function that internally calls [`AwsIotShadow_TimedGet`](https://github.com/aws/amazon-freertos/blob/master/libraries/c_sdk/aws/shadow/src/aws_shadow.c#L511) to implement a timeout whereas `AwsIotShadow_Get` is **_non-blocking_** and can invoke a callback once the _Thing_'s data is retrieved.

As far as when to use one or the other, that depends on if/how you want to handle timeouts for retrieving the _Thing_'s data.

sameerdasarwad, 2020-03-19:
Hi thanks for valuable information i was little confused while looking into conveyor, drink dispenser and aws shadow demo thanks for clarification

abhidixi11, 2020-03-19:
Hello @sameerdasarwad,
Thanks for  letting us know that the information helped. I'm closing this issue, if you still have questions regarding this issue,  please feel free to re-open the issue or create a new one.
Thanks !

closed 2020-03-19 by abhidixi11