aws/amazon-freertos #1837: Bug: Race Condition in StreamBuffer and MessageBuffer

dereks-igor, 2020-03-16:
The FreeRTOS StreamBuffer (and thus, also MessageBuffer) has a race condition in versions V10.2.1 and below.

**Symptom:**

When calling 

```cpp
xStreamBufferReceive(xStreamBuffer, pvRxData, xBufferLengthBytes, xTicksToWait);
```
, the function will sometimes immediately return a value of zero to indicate zero bytes read. It is similar as if the `xTicksToWait` timeout has expired with no data, however, the timeout value is not respected (the calling Task does not block for "xTicksToWait" ticks waiting for data). Depending on the application code, this could look like false hardware timeouts, or in time-sensitive synchronous data streams, like a broken data stream.

Here is an example result we produced on an ARM system. It was with two tight looping Tasks and a 100KB StreamBuffer, reading data samples at approximately 1 KHz. We could reproduce this intermittently, roughly one out every ten test runs.

In the example timeline below, "block to wait for a sample" means calling `xStreamBufferReceive()` to wait for data. That eventually results in a call to `portYIELD_WITHIN_API()`, which is shown as a "CONTEXT SWITCH" below.

"Write sample" and "notify the blocked Reader task" means calling `xStreamBufferSend()`, the function where both of those things occur. The race condition is within that function (`xStreamBufferSend()`), between lines 562 and 571.

```
  Reader Task:                        Writer Task:
  -------------                       -------------
                                      (Kickoff writing loop)
  (Kickoff reading loop)
  block to wait for a sample

                [CONTEXT SWITCH via portYIELD_WITHIN_API()]

                                      write sample 1 into the StreamBuffer (stream_buffer.c:562)
                                      notify the blocked Reader task of sample 1 (stream_buffer.c:571)
                                      (loop / repeat for another sample)
                                      write sample 2 into the StreamBuffer (stream_buffer.c:562)

                   [CONTEXT SWITCH via SysTick_Handler()]

  read sample 1
  (loop / repeat for another sample)
  read sample 2 -- StreamBuffer is now empty
  (loop / repeat for another sample)
  block to wait for a sample

                [CONTEXT SWITCH via portYIELD_WITHIN_API()]

                                      notify the blocked Reader task of sample 2 (stream_buffer.c:571)
                                      (loop / repeat for another sample)

                    [CONTEXT SWITCH via SysTick_Handler()]

  error: got a TaskNotify, but the StreamBuffer is empty!
```

`xStreamBufferSendFromISR()` doesn't have this issue because the SysTick interrupt has the lowest priority (it's 15 on ARM) and so won't preempt your application's ISR.

**Root Cause:**

The error case mentioned above was attempted to be guarded against in `xStreamBufferSend()` with this conditional:

```cpp
if( prvBytesInBuffer( pxStreamBuffer ) >= pxStreamBuffer->xTriggerLevelBytes )
{
	sbSEND_COMPLETED( pxStreamBuffer );
}
```

This protects against notifying if the data has already been consumed by the reading thread. Unfortuntely, after the sending thread calls `prvBytesInBuffer()`, a SysTick interrupt can occur which wakes up the other reader thread to consume the buffer before getting sent its notification with `sbSEND_COMPLETED` (which calls `xTaskNotify()`).

**Workaround:**

Here is the default (and most commonly used) definition of `sbSEND_COMPLETED`, which is the user-customizable macro that sends the `xTaskNotify()` to the receiving thread:

```cpp
#ifndef sbSEND_COMPLETED
	#define sbSEND_COMPLETED( pxStreamBuffer )											\
		vTaskSuspendAll();																\
		{																				\
			if( ( pxStreamBuffer )->xTaskWaitingToReceive != NULL )						\
			{																			\
				( void ) xTaskNotify( ( pxStreamBuffer )->xTaskWaitingToReceive,		\
									  ( uint32_t ) 0,									\
									  eNoAction );										\
				( pxStreamBuffer )->xTaskWaitingToReceive = NULL;						\
			}																			\
		}																				\
		( void ) xTaskResumeAll();
#endif /* sbSEND_COMPLETED */
```

Note that this macro protects itself against a SysTick race by using `vTaskSuspendAll()`/`xTaskResumeAll()`, which prevents any Task switching in this block of code. The race condition is fixed by moving the check for `prvBytesInBuffer()` into this protected section, so that the SysTick can't invoke the race condition:

```cpp
#ifndef sbSEND_COMPLETED
	#define sbSEND_COMPLETED( pxStreamBuffer )											\
		vTaskSuspendAll();																\
		{																				\
			if( ( pxStreamBuffer )->xTaskWaitingToReceive != NULL && 						\
  			  prvBytesInBuffer( pxStreamBuffer ) >= pxStreamBuffer->xTriggerLevelBytes) \
			{																			\
				( void ) xTaskNotify( ( pxStreamBuffer )->xTaskWaitingToReceive,		\
									  ( uint32_t ) 0,									\
									  eNoAction );										\
				( pxStreamBuffer )->xTaskWaitingToReceive = NULL;						\
			}																			\
		}																				\
		( void ) xTaskResumeAll();
#endif /* sbSEND_COMPLETED */
```

Existing users can fix their deployment of FreeRTOS by customizing the macro above within their build system, using either the corrected default version as shown above, or (if it's already been customized) by adding the test for `prvBytesInBuffer()` within their pre-existing customized version.

For a permanent fix, the upstream FreeRTOS source code should update the default `sbSEND_COMPLETED` to the above example, and also remove the unprotected and extraneous call to `prvBytesInBuffer()` (and also it's conditional if statement) from the function `xStreamBufferSend()`.

RichardBarry, 2020-03-17:
Thanks for the detailed information.

I'm not keen on moving the prvBytesInBuffer() call inside the macro as it would not be backward compatible for people who have created their own macros, and would put additional burden on people writing their own in the future.

The first thing I need to do is create a test that exhibits this behaviour, then we can try to figure out a generic solution, if possible.

As an aside, we will leave this issue posted here now for continuity as we work it to a conclusion - but future issues with the kernel itself can be posted in this repo: https://github.com/FreeRTOS/FreeRTOS-Kernel

RichardBarry, 2020-03-17:
To help us reliably replicating the observed behavior I would be grateful if you could attach you test file - interested in seeing the parameters used to create the stream buffer, the priority of the respective send and receive tasks, and the send and receive tasks themselves (so I can see how many bytes are being written at a time).....I have something running already but not seen the behaviour yet.

dereks, 2020-03-17:
@RichardBarry Thank you for your response. Unfortunately we saw this on a custom commercial board so I can't just copy/paste our code. Also, the app has some other threads launched, although they should have all been idle. 

I'll try to answer your questions, and if you still can't get a repro then I'll try to make a simple test app on one of my STM Discovery boards. But I don't have a lot of spare time on my hands this month.

> the parameters used to create the stream buffer

```cpp
#define STREAM_BUFFER_SIZE_BYTES 100000
#define STREAM_BUFFER_TRIGGER_LEVEL_BYTES 2

g_stream_buffer_handle = xStreamBufferCreate( STREAM_BUFFER_SIZE_BYTES, STREAM_BUFFER_TRIGGER_LEVEL_BYTES );
```

> the priority of the respective send and receive tasks
```
#define configMINIMAL_STACK_SIZE                 ((uint16_t)128)

#define READER_TASK_STACK_SIZE    (configMINIMAL_STACK_SIZE*14)
#define READER_TASK_PRIORITY 1

#define WRITER_TASK_STACK_SIZE           (configMINIMAL_STACK_SIZE*2)
#define WRITER_TASK_PRIORITY 1
```
There are 6 Tasks in total, and they all have priority 1. I believe the other 4 tasks were sleeping during most of the test runs.

The error was seen with two different interrupt-driven writers: one from an I2C bus (IRQ priority 5) and one from a SPI bus (IRQ priority 5). The SysTick IRQ is priority 15 (the lowest priority).

The bug disappears on a third writer that uses `xStreamBufferSendFromISR()` instead of `xStreamBufferSend()`, because the within an ISR the SysTick interrupt doesn't fire. (Also, applying the fix mentioned above resolves the problem.)

Here are hand-scrubbed versions of some code. This isn't the literal code we run but shows the main architecture.

The writer task busy-loops buffering up to N samples of data.

```cpp
void writer_task(void *ignored) {

  writer_event_t event;

  while(1) {
    // Wait for a software-triggered launch:
    xQueueReceive(g_event_queue, &event, portMAX_DELAY);

    // Launch a sampling loop. This task is consumed until the number of
    // samples have been buffered (or failed w/timeout):
    unsigned int sample_count = 0;
    while (sample_count < event.data.sample_count) {
      buffer_sample();
      sample_count++;
    }

  }

  return;
}

static void buffer_sample(void) {
  uint32_t timestamp;
  uint16_t value_16bits;

  // We expect to block (sleeping) here until a value is available from
  // the data bus. It's an interrupt-driven bus read that uses xTaskNotifyFromISR().
  hardware_bus_read(&timestamp, &value_16bits);

#define STREAM_BUFFER_TIMEOUT_MS 2000
  xStreamBufferSend(g_stream_buffer_handle, &timestamp, 4, STREAM_BUFFER_TIMEOUT_MS);
  xStreamBufferSend(g_stream_buffer_handle, &value_16bits, 2, STREAM_BUFFER_TIMEOUT_MS);
}
```

The reader task is a little more complicated. It is a generic event handling loop:

```cpp
void reader_task(void *ignored) {

  task_init();
  
  reader_event_t event;

  while (1) {

    // Get the next event.
    xQueueReceive(event_queue, &event, portMAX_DELAY);

    log_event(&event);

    handle_event(&event);

  }
}
```

However, unlike the writer task (which busy-loops buffering N samples as fast as possible) the reader task processes one new DATA_READY event in the event queue for each sample. It looks like this:

```cpp

// handle_event() eventually calls handle_state_started():
static void handle_state_started(reader_event_t *event) {
  
  switch (event->type) {

    // ...snip...

    case READER_EVENT_DATA_READY:

      g_context.sample_countdown--;

      handle_data();  // Consume a buffered sample and timestamp with xStreamBufferReceive()

      if (g_context.sample_countdown == 0) {
        // Done.
        set_state(READER_STATE_STOPPED); 
      } else {
        // Pump the event queue with DATA_READY events until done.
        // This gives other threads a chance to submit "stop" or "restart" events.

        reader_data_ready();   // Put another DATA_READY event on the queue.
      }

      break;
  }

}
```

The reader's `handle_data()` function basically boils down to this (after some other application logic):

```cpp
byte_count = xStreamBufferReceive( g_stream_buffer_handle, 
                                   (uint8_t *)timestamp,
                                   4, 
                                   STREAM_BUFFER_TIMEOUT_MS );

byte_count += xStreamBufferReceive( g_stream_buffer_handle, 
                                   (uint8_t *)&value_16bits,
                                   2, 
                                   STREAM_BUFFER_TIMEOUT_MS );
```

> not seen the behaviour yet.

It is easier to repro when taking many samples (1000 samples or 10000 samples). But we did get a repro roughly 1 out of 10 times with just 200 samples. In none of our tests did we completely fill the 100000 byte StreamBuffer -- it was empty at the start of each test, and I don't think we tested with more than 1000 samples (because even 200 was sufficient to repro).

I do not think the hardware bus interrupts have anything to do with it -- I would first ignore `hardware_bus_read()` and just try just putting any random value in the StreamBufffer from RAM. We saw the bug when writing 6 bytes per sample (as shown above) and also when reading and writing 8 bytes per sample (writing a `uint32_t value_32bits` instead of the `uint16_t value_16bits` shown above).

Please let me know if you get a repro...

dereks-igor, 2020-03-17:
@RichardBarry Thank you for your response. Unfortunately we saw this on a custom commercial board so I can't just copy/paste our code. Also, the app has some other threads launched, although they should have all been idle. 

I'll try to answer your questions, and if you still can't get a repro then I'll try to make a simple test app on one of my STM Discovery boards. But I don't have a lot of spare time on my hands this month.

> the parameters used to create the stream buffer

```cpp
#define STREAM_BUFFER_SIZE_BYTES 100000
#define STREAM_BUFFER_TRIGGER_LEVEL_BYTES 2

g_stream_buffer_handle = xStreamBufferCreate( STREAM_BUFFER_SIZE_BYTES, STREAM_BUFFER_TRIGGER_LEVEL_BYTES );
```

> the priority of the respective send and receive tasks
```
#define configMINIMAL_STACK_SIZE                 ((uint16_t)128)

#define READER_TASK_STACK_SIZE    (configMINIMAL_STACK_SIZE*14)
#define READER_TASK_PRIORITY 1

#define WRITER_TASK_STACK_SIZE           (configMINIMAL_STACK_SIZE*2)
#define WRITER_TASK_PRIORITY 1
```
There are 6 Tasks in total, and they all have priority 1. I believe the other 4 tasks were sleeping during most of the test runs.

The error was seen with two different interrupt-driven writers: one from an I2C bus (IRQ priority 5) and one from a SPI bus (IRQ priority 5). The SysTick IRQ is priority 15 (the lowest priority).

The bug disappears on a third writer that uses `xStreamBufferSendFromISR()` instead of `xStreamBufferSend()`, because the within an ISR the SysTick interrupt doesn't fire. (Also, applying the fix mentioned above resolves the problem.)

Here are hand-scrubbed versions of some code. This isn't the literal code we run but shows the main architecture.

The writer task busy-loops buffering up to N samples of data.

```cpp
void writer_task(void *ignored) {

  writer_event_t event;

  while(1) {
    // Wait for a software-triggered launch:
    xQueueReceive(g_event_queue, &event, portMAX_DELAY);

    // Launch a sampling loop. This task is consumed until the number of
    // samples have been buffered (or failed w/timeout):
    unsigned int sample_count = 0;
    while (sample_count < event.data.sample_count) {
      buffer_sample();
      sample_count++;
    }

  }

  return;
}

static void buffer_sample(void) {
  uint32_t timestamp;
  uint16_t value_16bits;

  // We expect to block (sleeping) here until a value is available from
  // the data bus. It's an interrupt-driven bus read that uses xTaskNotifyFromISR().
  hardware_bus_read(&timestamp, &value_16bits);

#define STREAM_BUFFER_TIMEOUT_MS 2000
  xStreamBufferSend(g_stream_buffer_handle, &timestamp, 4, STREAM_BUFFER_TIMEOUT_MS);
  xStreamBufferSend(g_stream_buffer_handle, &value_16bits, 2, STREAM_BUFFER_TIMEOUT_MS);
}
```

The reader task is a little more complicated. It is a generic event handling loop:

```cpp
void reader_task(void *ignored) {

  task_init();
  
  reader_event_t event;

  while (1) {

    // Get the next event.
    xQueueReceive(event_queue, &event, portMAX_DELAY);

    log_event(&event);

    handle_event(&event);

  }
}
```

However, unlike the writer task (which busy-loops buffering N samples as fast as possible) the reader task processes one new DATA_READY event in the event queue for each sample. It looks like this:

```cpp

// handle_event() eventually calls handle_state_started():
static void handle_state_started(reader_event_t *event) {
  
  switch (event->type) {

    // ...snip...

    case READER_EVENT_DATA_READY:

      g_context.sample_countdown--;

      handle_data();  // Consume a buffered sample and timestamp with xStreamBufferReceive()

      if (g_context.sample_countdown == 0) {
        // Done.
        set_state(READER_STATE_STOPPED); 
      } else {
        // Pump the event queue with DATA_READY events until done.
        // This gives other threads a chance to submit "stop" or "restart" events.

        reader_data_ready();   // Put another DATA_READY event on the queue.
      }

      break;
  }

}
```

The reader's `handle_data()` function basically boils down to this (after some other application logic):

```cpp
byte_count = xStreamBufferReceive( g_stream_buffer_handle, 
                                   (uint8_t *)timestamp,
                                   4, 
                                   STREAM_BUFFER_TIMEOUT_MS );

byte_count += xStreamBufferReceive( g_stream_buffer_handle, 
                                   (uint8_t *)&value_16bits,
                                   2, 
                                   STREAM_BUFFER_TIMEOUT_MS );
```

> not seen the behaviour yet.

It is easier to repro when taking many samples (1000 samples or 10000 samples). But we did get a repro roughly 1 out of 10 times with just 200 samples. In none of our tests did we completely fill the 100000 byte StreamBuffer -- it was empty at the start of each test, and I don't think we tested with more than 1000 samples (because even 200 was sufficient to repro).

I do not think the hardware bus interrupts have anything to do with it -- I would first ignore `hardware_bus_read()` and just try just putting any random value in the StreamBufffer from RAM. We saw the bug when writing 6 bytes per sample (as shown above) and also when reading and writing 8 bytes per sample (writing a `uint32_t value_32bits` instead of the `uint16_t value_16bits` shown above).

Please let me know if you get a repro...

dereks-igor, 2020-03-17:
Also of note: We were reading samples at roughly 1 KHz (+/- 20%). If you simply populate 100000 bytes of random data from RAM, that might happen too fast and just immediately fill the StreamBuffer until it's full. So I recommend putting in some artificial delay, maybe 1 ms, between each write.

(Maybe the fact that our sample rate was close to the SysTick rate (1 ms) had something to do with us getting a repro.)

RichardBarry, 2020-03-18:
Thanks - will report back with suggestions.

RichardBarry, 2020-03-18:
Well I've not been able to create a reliable test to catch this, but I can force the issue by simply adding a taskYIELD() between

if( prvBytesInBuffer( pxStreamBuffer ) >= pxStreamBuffer->xTriggerLevelBytes )

and

sbSEND_COMPLETED( pxStreamBuffer );

....then it happens every time.  In some of the coverage tests we do similar things by adding long null loops at certain points in the code to hold execution up to make it more likely to hit the case where an interrupt occurs at that point.

....anyway, now to decide what to do.  The stream buffers are intended to be lightweight so we don't want to impact that design goal - one option is to document that this might happen but in that case we would need a mechanism to communicate that it happened.

RichardBarry, 2020-03-19:
The following **may** fix the issue although it is not fully tested.  Grateful if you can take a look and even give it a try.  It adds a do/while loop that doesn't let the function exit if zero bytes are received without there also having been a timeout:

`
size_t xStreamBufferReceive( StreamBufferHandle_t xStreamBuffer,
							 void *pvRxData,
							 size_t xBufferLengthBytes,
							 TickType_t xTicksToWait )
{
StreamBuffer_t * const pxStreamBuffer = xStreamBuffer;
size_t xReceivedLength = 0, xBytesAvailable, xBytesToStoreMessageLength;
TimeOut_t xTimeOut;
	
	configASSERT( pvRxData );
	configASSERT( pxStreamBuffer );

	/* Note the time on entry of this function to ensure time outs are handled
	correctly if the task enters and exits the blocked state multiple times.
	That could happen if a context switch occurs between the sending task
	checking to see if there is data in the buffer and sending a notification
	to the receiving task.  If that were to happen then the receiving task
	might drain the buffer before receiving the notification to say there was
	more data in the buffer - resulting in zero bytes received even though the
	timeout had not expired. */
	vTaskSetTimeOutState( &xTimeOut );
	
	/* This receive function is used by both message buffers, which store
	discrete messages, and stream buffers, which store a continuous stream of
	bytes.  Discrete messages include an additional
	sbBYTES_TO_STORE_MESSAGE_LENGTH bytes that hold the length of the
	message. */
	if( ( pxStreamBuffer->ucFlags & sbFLAGS_IS_MESSAGE_BUFFER ) != ( uint8_t ) 0 )
	{
		xBytesToStoreMessageLength = sbBYTES_TO_STORE_MESSAGE_LENGTH;
	}
	else
	{
		xBytesToStoreMessageLength = 0;
	}

	do
	{
		if( xTicksToWait != ( TickType_t ) 0 )
		{		
			/* Checking if there is data and clearing the notification state must be
			performed atomically. */
			taskENTER_CRITICAL();
			{
				xBytesAvailable = prvBytesInBuffer( pxStreamBuffer );

				/* If this function was invoked by a message buffer read then
				xBytesToStoreMessageLength holds the number of bytes used to hold
				the length of the next discrete message.  If this function was
				invoked by a stream buffer read then xBytesToStoreMessageLength will
				be 0. */
				if( xBytesAvailable <= xBytesToStoreMessageLength )
				{
					/* Clear notification state as going to wait for data. */
					( void ) xTaskNotifyStateClear( NULL );

					/* Should only be one reader. */
					configASSERT( pxStreamBuffer->xTaskWaitingToReceive == NULL );
					pxStreamBuffer->xTaskWaitingToReceive = xTaskGetCurrentTaskHandle();
				}
				else
				{
					mtCOVERAGE_TEST_MARKER();
				}
			}
			taskEXIT_CRITICAL();

			if( xBytesAvailable <= xBytesToStoreMessageLength )
			{
				/* Wait for data to be available. */
				traceBLOCKING_ON_STREAM_BUFFER_RECEIVE( xStreamBuffer );
				( void ) xTaskNotifyWait( ( uint32_t ) 0, ( uint32_t ) 0, NULL, xTicksToWait );
				pxStreamBuffer->xTaskWaitingToReceive = NULL;

				/* Recheck the data available after blocking. */
				xBytesAvailable = prvBytesInBuffer( pxStreamBuffer );
			}
			else
			{
				mtCOVERAGE_TEST_MARKER();
			}
		}
		else
		{
			xBytesAvailable = prvBytesInBuffer( pxStreamBuffer );
		}

		/* Whether receiving a discrete message (where xBytesToStoreMessageLength
		holds the number of bytes used to store the message length) or a stream of
		bytes (where xBytesToStoreMessageLength is zero), the number of bytes
		available must be greater than xBytesToStoreMessageLength to be able to
		read bytes from the buffer. */
		if( xBytesAvailable > xBytesToStoreMessageLength )
		{
			xReceivedLength = prvReadMessageFromBuffer( pxStreamBuffer, pvRxData, xBufferLengthBytes, xBytesAvailable, xBytesToStoreMessageLength );

			/* Was a task waiting for space in the buffer? */
			if( xReceivedLength != ( size_t ) 0 )
			{
				traceSTREAM_BUFFER_RECEIVE( xStreamBuffer, xReceivedLength );
				sbRECEIVE_COMPLETED( pxStreamBuffer );
			}
			else
			{
				mtCOVERAGE_TEST_MARKER();
			}
		}
		else
		{
			traceSTREAM_BUFFER_RECEIVE_FAILED( xStreamBuffer );
			mtCOVERAGE_TEST_MARKER();
		}
		
		/* See the comment where vTaskInternalSetTimeOutState() is called above. */
	} while( ( xReceivedLength == 0 ) && ( xTaskCheckForTimeOut( &xTimeOut, &xTicksToWait ) == pdFALSE ) );

	return xReceivedLength;
}
`

dereks-igor, 2020-03-24:
@RichardBarry Thank you for the proposed fix. I have tested it in our environment and it does indeed resolve the symptom.

However, I disagree with this fix, for the following two reasons:

1. It does not fix the race condition. It simply works around it by checking whether or not the race occurred. This could have unintended side effects. For example, it could result in repeatedly waking up the second processor on SMP systems, as in the FreeRTOS Demo file `MessageBufferAMP.c`, which would result in reduced battery life.
2. It changes the behavior of `xStreamBufferReceive()`. Currently, it is possible to have a Task that blocks forever (sleeping), waiting for bytes from a StreamBuffer. That task can then programmatically be kicked awake with a call to `xTaskNotify()` if you want to send it an out-of-band signal to halt, reset, or perform error recovery (for example by a watchdog timer, or by a bus overflow interrupt, or by a manual button press for a reset and/or test mode). For example, this type of code now becomes impossible with the proposed change:

```cpp
void hardware_bus_buffer_task(void *ignored) {

  size_t byte_count;
  uint8_t chunk_buffer[CHUNK_SIZE];

  while (1) {

    byte_count = xStreamBufferReceive(stream_buffer, chunk_buffer, CHUNK_SIZE, portMAX_DELAY);

    if (byte_count > 0) {
      hardware_bus_write(chunk_buffer, byte_count);
    } else {
      /* An error occurred. Reset the hardware bus. */
      g_error_count++;
      reset_hardware_bus();
    }

  }  /* while (1) */
}
```

In my opinion this change is less desirable than moving the `prvBytesInBuffer()` call inside the `sbSEND_COMPLETED` macro.

As an alternative third option for a fix, I propose the following change to the function `xStreamBufferSend()`, starting at line 568. This just wraps the existing call to `sbSEND_COMPLETED` with `vTaskSuspendAll()`/`xTaskResumeAll()`:

```cpp
if( xReturn > ( size_t ) 0 )
{
	traceSTREAM_BUFFER_SEND( xStreamBuffer, xReturn );

	/* Ensure we don't context switch while notifying a waiting task: */
	vTaskSuspendAll();  

	/* Was a task waiting for the data? */
	if( prvBytesInBuffer( pxStreamBuffer ) >= pxStreamBuffer->xTriggerLevelBytes )
	{
		sbSEND_COMPLETED( pxStreamBuffer );
	}
	else
	{
		mtCOVERAGE_TEST_MARKER();
	}
	
	/* The waiting task (if any) has been notified. Resume normal context switching. */
	( void ) xTaskResumeAll();
}
```

This is safe to do because calls to `vTaskSuspendAll()`/`xTaskResumeAll()` can be nested, so the fact that there is already a call to those functions within the default `sbSEND_COMPLETED` macro is harmless. In fact, the burden on the user is reduced, because they no longer need to remember to have those calls in there themselves.  (And the default macro definition can remove those redundant calls as a minor cleanup.) 

I have tested all three of the proposed fixes on our hardware (including the one above) and they all resolve the symptom. But the new proposed fix (above) has the following advantages:

- It is backwards compatible with existing customized `sbSEND_COMPLETED` macros that may already have calls to `vTaskSuspendAll()`/`xTaskResumeAll()`.
- It does not change the behavior of `xStreamBufferReceive()`.
- It eliminates the race condition, rather than just working around it.

Thoughts?

RichardBarry, 2020-03-30:
I like the third option too, although this does move away from the 'lockless' buffer idea.  I will have to think a little more about the deficiencies you call out regarding the suggestion I posted, and the implication of restricting what people can do inside the sbSEND_COMPLETED() macro as it is called with the scheduler suspended.   In the default case (where a user does not provide their own sbSEND_COMPLETED() macro), we could remove the suspend/resume all calls from the macro if the macro is called with the scheduler already suspended (despite its name, vTaskSuspendAll() really suspends the scheduler, not the tasks).

Viatorus, 2021-07-25:
Hello, what is the state of this bug? Is there still a raise condition in the newest version?

micheledarold, 2022-01-04:
@RichardBarry @dereks-igor what about this issue? is still unresolved?

paulbartell, 2022-04-22:
I believe this was fixed in FreeRTOS-Kernel V10.4.4 with commit [6685c042cb3b3811e2965d3dce59c42e4566648f](https://github.com/FreeRTOS/FreeRTOS-Kernel/commit/6685c042cb3b3811e2965d3dce59c42e4566648f).

closed 2022-04-22 by paulbartell

dereks-igor, 2022-04-23:
> I believe this was fixed in FreeRTOS-Kernel V10.4.4 

@paulbartell Negative, it is not fixed. Please re-open this bug. The race condition still exists exactly as described under "**Root Cause**" in this Issue's Description.

In the new commit, the **unguarded check for a waiting task** is on line 651. This happens **before the vTaskSuspendAll()** (in the `sbSEND_COMPLETED` macro) on line 653, so 

- if you get a SysTick interrupt between lines 651 and 653, and 
- that SysTick results in the waiting task getting CPU time where it empties the StreamBuffer, 

you'll get an incorrect TaskNotify event. The call to xStreamBufferReceive() will unblock but there will be zero bytes available.

After the discussion thread above, I still propose that the best fix is to simply move the check for a waiting task into guarded section, thus eliminating the race.

```
#ifndef sbSEND_COMPLETED
	#define sbSEND_COMPLETED( pxStreamBuffer )											\
		vTaskSuspendAll();																\
		{																				\
			if( ( pxStreamBuffer )->xTaskWaitingToReceive != NULL && 						\
  			  prvBytesInBuffer( pxStreamBuffer ) >= pxStreamBuffer->xTriggerLevelBytes) \
			{																			\
				( void ) xTaskNotify( ( pxStreamBuffer )->xTaskWaitingToReceive,		\
									  ( uint32_t ) 0,									\
									  eNoAction );										\
				( pxStreamBuffer )->xTaskWaitingToReceive = NULL;						\
			}																			\
		}																				\
		( void ) xTaskResumeAll();
#endif /* sbSEND_COMPLETED */
```

dereks-igor, 2022-04-29:
@paulbartell Ping? This bug should be re-opened. It is not resolved.