aws/amazon-freertos #1845: ESP32 BLE Wi-Fi provisioning example disconnects after 20 seconds when using BLE app

ReinierH, 2020-03-19:
**Describe the bug**
Provisioning Wi-Fi on the ESP32 manually using a BLE app (LightBlue) causes the app to disconnect after approximately 20 seconds. After further investigation I noticed that the LightBlue app automatically performs a read request when opening the RX_CHARACTERISTIC (0x03) characteristic. 

Apparently iPhones do not accept that read requests won't get answered which result in the iPhone disconnecting from the device. 

TLDR;
Reading to the iot_ble_data_transfer RX_CHARACTERISTIC causes the iPhone to disconnect after approximately 20 seconds.

**System information**
- Which hardware board or part numbers?
ESP32_devkitc module
iPhone 6 and 8
- Version of FreeRTOS (run ``git describe --tags`` to find it)
201912.00-121-g57bd0a856
- Project [Custom Application|MQTT|Shadow|Greengrass|OTA|Device Defender|TCP Echo|Sub pub Demo|Simple TCP Server]
IOT_BLE_ENABLE_WIFI_PROVISIONING

**Expected behavior**
Expected the iPhone not to disconnect when performing a read request to the RX_CHARACTERISTIC (a9d7166ad72e40a9a00248044cc30103).

**To reproduce**
Steps to reproduce the behavior:

1. Perform a READ request to the RX_CHARACTERISTIC using a BLE scanner app (i.e. Lightblue) using an iPhone
2. Wait 20 seconds
3. App will disconnect from the ESP32

**Code to reproduce the bug**
Added a **IotBle_SendResponse()** in the case of a **eBLERead** event
**amazon-freertos/libraries/c_sdk/standard/ble/src/services/mqtt_ble/iot_ble_data_transfer.c:735**
```
static void _RXMesgCharCallback( IotBleAttributeEvent_t * pEventParam )
{
    IotBleAttributeData_t attrData = { 0 };
    IotBleEventResponse_t resp =
    {
        .pAttrData      = &attrData,
        .rspErrorStatus = eBTRspErrorNone,
        .eventStatus    = eBTStatusFail,
        .attrDataOffset = 0
    };
    IotBleDataTransferService_t * pService;
    bool status = false;
    IotBleDataChannelBuffer_t recvBuffer = { 0 };
    /* ADDED eBLERead */
    if( ( pEventParam->xEventType == eBLERead ) || ( pEventParam->xEventType == eBLEWrite )  || ( pEventParam->xEventType == eBLEWriteNoResponse ) )
    {
        pService = _getServiceFromHandle( pEventParam->pParamWrite->attrHandle );

        if( ( pService != NULL ) &&
            ( pService->channel.isOpen == true ) )
        {
            recvBuffer.pBuffer = ( uint8_t * ) pEventParam->pParamWrite->pValue;
            recvBuffer.head = pEventParam->pParamWrite->length;
            recvBuffer.tail = 0;
            pService->channel.pReceiveBuffer = &recvBuffer;

            if( pService->channel.callback != NULL )
            {
                pService->channel.callback( IOT_BLE_DATA_TRANSFER_CHANNEL_DATA_RECEIVED,
                                            &pService->channel,
                                            pService->channel.pContext );
            }

            resp.eventStatus = eBTStatusSuccess;
        }

        if( pEventParam->xEventType == eBLEWrite )
        {
            resp.pAttrData->handle = pEventParam->pParamWrite->attrHandle;
            resp.pAttrData->pData = ( uint8_t * ) &status;
            resp.pAttrData->size = 1;
            IotBle_SendResponse( &resp, pEventParam->pParamWrite->connId, pEventParam->pParamWrite->transId );
        }
        /* ADDED THIS TO PREVENT IPHONES FROM DISCONNECTING */
        if( pEventParam->xEventType == eBLERead )
        {
            resp.pAttrData->handle = pEventParam->pParamRead->attrHandle;
            resp.pAttrData->pData = ( uint8_t * ) &status;
            resp.pAttrData->size = 1;
            IotBle_SendResponse( &resp, pEventParam->pParamRead->connId, pEventParam->pParamRead->transId );
        }
    }
}
```

ravibhagavandas, 2020-03-19:
@ReinierH 

This characteristic (RX) is  used to write data to FreeRTOS (peripheral) device and its not meant for reading data. To read from the data transfer service, you can use the TX characteristic. 

Also, I noticed you are using a different APP to perform WiFi provisioning using iPhone. Did you consider using [FreeRTOS BLE companion IOS SDK](https://github.com/aws/amazon-freertos-ble-ios-sdk) for this? It has a builtin WiFi provisioning library and is meant to work out of the box without any extra implementation.

ReinierH, 2020-03-20:
@ravibhagavandas 

I understand there is no reason to read this characteristic, nevertheless I wouldn't expect it to disconnect when you're able to read to the characteristic. 

Regarding the native iOS app, our app is written in React Native which forced us to reverse engineer the protocol and implement it into our app.

ravibhagavandas, 2020-03-23:
A better approach would be to remove the read permission and read property for this characteristic, as there is no requirement for the mobile app to read that characteristic at all. We will raise a PR for this change.

PR #1860 (2020-03-30): BLE data transfer: Remove read permission on RX char
Remove read permission on RX char.
PR Fixes #1845 

Description
-----------
Read permission is not required for RX characteristics as its used to write data to the device.

Added unit tests to catch this behavior.

Checklist:
----------

- [x] I have tested my changes. No regression in existing tests.
- [ ] My code is Linted.

By submitting this pull request, I confirm that you can use, modify, copy, and redistribute this contribution, under the terms of your choice.
closed 2020-03-30 by dachalco