aws/amazon-freertos #1846: Stack overflow when making HTTPS request

TheIronNinja, 2020-03-20:
**Describe the bug**
I'm trying to make use of a CVM for my device. It needs to make a request to an endpoint and save the response, but it aborts with a stack overflow error when trying to make the HTTPS connection.
You can find the backtrace below.

**System information**
- ESP32 DEVKIT-C
- Windows
- Custom project with HTTP, MQTT and OTA. It crashes before MQTT or OTA are enabled though.

**Expected behavior**
What I'm trying to do is to use the code in the HTTPS sync download demo to download a certificate from a custom endpoint. When it reaches [this](https://github.com/aws/amazon-freertos/blob/master/demos/https/iot_demo_https_s3_download_sync.c#L359) line of the demo code it crashes with the following traceback:

```
***ERROR*** A stack overflow in task main has been detected.
abort() was called at PC 0x40088a40 on core 0
0x40088a40: vApplicationStackOverflowHook at {project path}/FreeRTOS/vendors/espressif/esp-idf/components/esp32/panic.c:715

ELF file SHA256: e2f56ba4b548185eda6be273a9a9fb967498c5803073e03b4a285bda0dc5c24b

Backtrace: 0x400887fc:0x3ffb9b10 0x40088a29:0x3ffb9b30 0x40088a40:0x3ffb9b50 0x400894c1:0x3ffb9b70 0x40088e38:0x3ffb9b90 0x40088ecd:0x3ffb9bb0 0x401317fe:0x3ffb9bf0 0x40131e4c:0x3ffb9c10 0x40130b02:0x3ffb9c40 0x40130d09:0x3ffb9c70 0x40130e31:0x3ffb9cc0 0x4012dd7d:0x3ffb9cf0 0x400dc2e0:0x3ffb9d10 0x40127338:0x3ffb9d40 0x40127d96:0x3ffb9fa0 0x400da4de:0x3ffb9fc0 0x400d198f:0x3ffbaae0 0x400d8079:0x3ffbab00
0x400887fc: invoke_abort at {project path}/FreeRTOS/vendors/espressif/esp-idf/components/esp32/panic.c:715

0x40088a29: abort at {project path}/FreeRTOS/vendors/espressif/esp-idf/components/esp32/panic.c:715

0x40088a40: vApplicationStackOverflowHook at {project path}/FreeRTOS/vendors/espressif/esp-idf/components/esp32/panic.c:715

0x400894c1: vTaskSwitchContext at {project path}/FreeRTOS/freertos_kernel/tasks.c:5068

0x40088e38: _frxt_dispatch at {project path}/FreeRTOS/freertos_kernel/portable/ThirdParty/GCC/Xtensa_ESP32/portasm.S:406

0x40088ecd: vPortYield at {project path}/FreeRTOS/freertos_kernel/portable/ThirdParty/GCC/Xtensa_ESP32/portasm.S:555

0x401317fe: xSendEventStructToIPTask at {project path}/FreeRTOS/libraries/freertos_plus/standard/freertos_plus_tcp/source/FreeRTOS_IP.c:854

0x40131e4c: FreeRTOS_bind at {project path}/FreeRTOS/libraries/freertos_plus/standard/freertos_plus_tcp/source/FreeRTOS_Sockets.c:3212

0x40130b02: prvCreateDNSSocket at {project path}/FreeRTOS/libraries/freertos_plus/standard/freertos_plus_tcp/source/FreeRTOS_DNS.c:179

0x40130d09: prvGetHostByName at {project path}/FreeRTOS/libraries/freertos_plus/standard/freertos_plus_tcp/source/FreeRTOS_DNS.c:179

0x40130e31: FreeRTOS_gethostbyname at {project path}/FreeRTOS/libraries/freertos_plus/standard/freertos_plus_tcp/source/FreeRTOS_DNS.c:541

0x4012dd7d: SOCKETS_GetHostByName at {project path}/FreeRTOS/libraries/abstractions/secure_sockets/freertos_plus_tcp/iot_secure_sockets.c:211

0x400dc2e0: IotNetworkAfr_Create at {project path}/FreeRTOS/libraries/abstractions/platform/freertos/iot_network_freertos.c:370

0x40127338: _createHttpsConnection at {project path}/FreeRTOS/libraries/c_sdk/standard/https/src/iot_https_client.c:3222

0x40127d96: IotHttpsClient_Connect at {project path}/FreeRTOS/libraries/c_sdk/standard/https/src/iot_https_client.c:3222

0x400da4de: HTTPKeyProvisioning at {project path}/components/key-provisioning/src/key-provisioning.c:1448

0x400d198f: app_main at {project path}/src/main.c:88

0x400d8079: main_task at {project path}/FreeRTOS/vendors/espressif/esp-idf/components/esp32/cpu_start.c:565
```

**Additional context**
I am calling this function after initializing the network, the IoTSDK and connecting to the WiFi, but before doing the key provisioning, as this has to become the production-ready key provisioning. The MQTT and OTA initializations are yet to be done at this point.

Thank you!

lundinc2, 2020-03-20:
@TheIronNinja I am sorry to hear you are running into this issue. 

Do you mind logging the amount of stack and heap you have available when you reach [this](https://github.com/aws/amazon-freertos/blob/29d006df5798c8b257ea20e03038eaa1f0862f31/demos/https/iot_demo_https_s3_download_sync.c#L356) point? 

The API for ESP32 would be xPortGetFreeHeapSize() and uxTaskGetStackHighWaterMark(). 

Since you are getting a stack overflow, the fix could be increasing the size of the stack for the task running the above code, by default it should be democonfigDEMO_STACKSIZE.

TheIronNinja, 2020-03-21:
Thanks for the response.

As I tried to print those values I got another stack overflow, so I couldn't really see them. I tried to increase the democonfigDEMO_STACKSIZE to more than double but to avail.

BUT what I've tried now is to call the function as a separated thread using Iot_CreateDetachedThread(), which has allowed me to print the values AND continue running the program further than before. These are the results:

```
10 334 [iot_thread] Free Heap available: 156900
11 334 [iot_thread] Free Stack available: 2036
```

My program is still not working, but not because of a stack overflow. Anyway, I still don't understand why increasing the size of the stack didn't work, as it was apparently the correct solution.

lundinc2, 2020-03-24:
Hello @TheIronNinja,

I apologize for the delayed response, that is odd indeed. What version of Amazon FreeRTOS are you using? Do you have some large object allocated on the stack in your custom code? 2036 is a relatively low amount of stack available, as this code appears early in the callstack.

I ran the HTTPS sync demo on commit ee50e8da46f7cd1c5ed7888bf056989ee3a0bf8d, and the data I got was:

Free Stack Size Available: 4108
Free Heap Size Available: 127896

Perhaps you can try increasing the stack further? For instance, to run the unchanged HTTPS sync download it took me about ~4,288 bytes in stack. I agree, using a separate thread is a good solution here. 

Please let me know if there's any way I can assist you further.

TheIronNinja, 2020-04-01:
Sorry for the delay.

I'm using the one Amazon provides when downloading from the Console, not sure how to check which version it is.

I am checking the heap and stack available right at the start of my code, so I'd assume it should be mostly empty. This is the code, in case you want to check it:

```
int app_main( void )
{
    /* Perform any hardware initialization that does not require the RTOS to be
     * running.  */

    prvMiscInitialization();
    configPRINTF( ( "Firmware version %u.%u.%u\r\n",
                    xAppFirmwareVersion.u.x.ucMajor,
                    xAppFirmwareVersion.u.x.ucMinor,
                    xAppFirmwareVersion.u.x.usBuild ) );

    configPRINTF( ( "Free heap: %d\r\n", xPortGetFreeHeapSize()) );
    configPRINTF( ( "Free stack: %d\r\n", uxTaskGetStackHighWaterMark(NULL)) );
    configPRINTF( ("Initializing FreeRTOS+TCP stack\r\n") );
```

prvMiscInitialization is the same function as in the demos, only initializing nvs, uart and the logging task.

This is what I'm seeing exactly with this code, again:

```
0 0 [main] Firmware version 1.0.0
2 1 [main] Free heap: 211732
3 1 [main] Free stack: 2308
1 0 [main] Initializing FreeRTOS+TCP stack
```

lundinc2, 2020-04-01:
Hello @TheIronNinja,

The stack size you are showing at this point in time is actually for a different task, then the one running the demo code in the default configuration. 

The call [here](https://github.com/aws/amazon-freertos/blob/e984d9463b7cb594434f07e8435869bab8787ba0/vendors/espressif/boards/esp32/aws_demos/application_code/main.c#L137) will call the demo task entry point, which starts [here](https://github.com/aws/amazon-freertos/blob/e984d9463b7cb594434f07e8435869bab8787ba0/demos/demo_runner/iot_demo_runner.c#L90).

dcgaws, 2020-04-09:
@TheIronNinja any success in pulling stack numbers for the task making the HTTPS call? As pointed out above, getting free heap numbers would be helpful as well, before and after initiating the TLS negotiation, for example.

closed 2020-04-27 by TheIronNinja