aws/amazon-freertos #1856: Support for Raspberry?

dangviethieu, 2020-03-24:
Hello all. I'm look forward about the project to connect and send data from Raspberry to Phone app (iOS, Android) use BLE, and then from Phone app, it will publish that data to AWS use MQTT message.
I already used ESP32 follow: https://aws.amazon.com/vi/blogs/iot/using-bluetooth-low-energy-with-amazon-freertos-on-espressif-esp32/
However, freertos isn't support Raspberry. Anyone can help me any idea? Thanks very much !!!

gordonwang0, 2020-03-24:
Hello,

FreeRTOS does not support Raspberry.  Since Rasberry can run Linux, I suggest using something such as BlueZ for BLE on Raspberry.

https://www.bluetooth.com/blog/bluez-on-raspberry-pi-update/

htibosch, 2020-03-25:
Hello, afaik, there are only third-party ports of FreeRTOS on Raspberry PI. You can Google them.
[James Walmsley](https://github.com/jameswalmsle) maintains one of those FreeRTOS ports.
Using Linux on a PI is great: it is well documented and it installs very easily. I installed the command-line only version (using `ssh`).

dangviethieu, 2020-03-25:
Yes thanks all. I sent data successfully to android app from rasp use the document at
https://scribles.net/creating-ble-gatt-server-uart-service-on-raspberry-pi

lundinc2, 2020-03-25:
@dangviethieu has your question been satisified? Is it okay if I mark this issue as resolved?

dangviethieu, 2020-03-26:
@lundinc2 Yes, I will close the issue. Thanks all

closed 2020-03-26 by dangviethieu