aws/amazon-freertos #1858: FreeRTOS window simulator not working on EC2 window server (all types 2019.2016/2012)

JackSpade12, 2020-03-26:
**Describe the bug**
FreeRTOS window board is not working on EC2 window instance sue to DHCP not returning an offer.

**System information**
- board: pc.window
- visual studio 2019
- Windows Server 2019
- 
- Project MQTT. 

**Expected behavior**
The demo should start and send MQTT messages 

**Screenshots or console output**

**To reproduce**

1. Download Window FreeRTOS Greengrass simulator (https://console.aws.amazon.com/freertos)
2. add the MQTT Demo and follow the instructions to replace aws_clientcredential.h , aws_clientcredential_keys.h with the credentials zip
3. run CMake GUI based on AWS instructions
4. run build in AWS VisualStudio ( set aws_demo as startup project)

**Additional context**
The same exact process works fine on a personal computer running Window 10

Thank you!

htibosch, 2020-03-26:
HI @JackSpade12, are you running this project from AWS/FreeRTOS?

    projects/pc/windows/visual_studio/aws_demos

Normally, in the WinSim application, I would expect more logging from the FreeRTOS+TCP stack.

Maybe you can enable debug logging as well, in your `FreeRTOSIPConfig.h`:

~~~c
#define ipconfigHAS_DEBUG_PRINTF    1
#define FreeRTOS_debug_printf( X )    configPRINTF( X )
~~~
And then run it again and post some logging?

JackSpade12, 2020-03-26:
Thanks will do it and send the output @htibosch 
my gut feeling is that WinPcap is not supporting a cloud VPN setting.

https://www.winpcap.org/misc/faq.htm#Q-6

JackSpade12, 2020-03-26:
Here is the console with debug

0 0 [None] FreeRTOS_IPInit
1 0 [None] vTaskStartScheduler
2 1 [IP-task] prvIPTask started

The following network interfaces are available:

Interface 1 - rpcap://\Device\NPF_{54B31D7E-36BF-4BBE-9AB2-106A939CD78C}
              (Network adapter 'Xen Project' on local host)

The interface that will be opened is set by "configNETWORK_INTERFACE_TO_USE", which
should be defined in FreeRTOSConfig.h
Attempting to open interface number 1.
Successfully opened interface number 1.
3 79 [IP-task] prvInitialiseDHCP: start after 250 ticks
4 79 [IP-task] vDHCPProcess: discover
5 5329 [IP-task] vDHCPProcess: discover
6 5329 [IP-task] vDHCPProcess: timeout 10000 ticks

JackSpade12, 2020-03-26:
tried using static IP but I am not sure how to set it up
getting this answer now

0 0 [None] FreeRTOS_IPInit
1 0 [None] vTaskStartScheduler
2 1 [IP-task] prvIPTask started

The following network interfaces are available:

Interface 1 - rpcap://\Device\NPF_{54B31D7E-36BF-4BBE-9AB2-106A939CD78C}
              (Network adapter 'Xen Project' on local host)
The interface that will be opened is set by "configNETWORK_INTERFACE_TO_USE", which
should be defined in FreeRTOSConfig.h
Attempting to open interface number 1.
Successfully opened interface number 1.
3 40 [IP-task] Write certificate...
4 48 [IP-task]

IP Address: xxxxxxx
5 48 [IP-task] Subnet Mask: 255.255.240.0
6 48 [IP-task] Gateway Address: xxxxxx
7 48 [IP-task] DNS Server Address: 208.67.222.222

8 48 [iot_thread] [INFO ][DEMO][48] ---------STARTING DEMO---------

9 49 [iot_thread] [INFO ][INIT][49] SDK successfully initialized.
10 49 [iot_thread] [INFO ][DEMO][49] Successfully initialized the demo. Network type for the demo: 4
11 49 [iot_thread] [INFO ][MQTT][49] MQTT library successfully initialized.
12 49 [iot_thread] [INFO ][DEMO][49] MQTT demo client identifier is WinSim (length 6).
13 1050 [iot_thread] [ERROR][NET][1050] Failed to resolve a30hnc7meo510z-ats.iot.us-east-1.amazonaws.com.
14 1050 [IP-task] FreeRTOS_closesocket[0 to 0.0.0.0:0]: buffers 60 socks 0
15 1051 [iot_thread] [ERROR][MQTT][1051] Failed to establish new MQTT connection, error NETWORK ERROR.
16 1051 [iot_thread] [ERROR][DEMO][1051] MQTT CONNECT returned error NETWORK ERROR.
17 1051 [iot_thread] [INFO ][MQTT][1051] MQTT library cleanup done.
18 1051 [iot_thread] [INFO ][DEMO][1051] memory_metrics::freertos_heap::before::bytes::2086532
19 1051 [iot_thread] [INFO ][DEMO][1051] memory_metrics::freertos_heap::after::bytes::2074360
20 1051 [iot_thread] [INFO ][DEMO][1051] memory_metrics::demo_task_stack::before::bytes::1908
21 1051 [iot_thread] [INFO ][DEMO][1051] memory_metrics::demo_task_stack::after::bytes::1908
22 1051 [iot_thread] [ERROR][DEMO][1051] Error running demo.
23 1053 [iot_thread] [INFO ][INIT][1053] SDK cleanup done.
24 1053 [iot_thread] [INFO ][DEMO][1053] -------DEMO FINISHED-------

htibosch, 2020-03-27:
> I updated the bug report with the debug output @htibosch

Thank you!

> my gut feeling is that WinPcap is not supporting a cloud VPN setting

To me it looks like there is no communication yet at all.

Is "Network adapter 'Xen Project' on local host" the correct adapter, the one that is connected to a LAN that leads to the Internet? Doesn't the logging show an adapter like Realtek?

Maybe you can comment out the call to `DEMO_RUNNER_RunDemos()`, so the logging stays quiet.
And then, from a different PC, try ping to the WinSim program? Maybe just disable `ipconfigUSE_DHCP` so it takes the static IP-address immediately?

Is `172.31.0.2` really the address of a DNS? Can you ping to e.g. "freertos.net"? And when you run WireShark, do you see that that mentioned DNS gives an answer?

( sorry to ask stupid questions, but I want to exclude every possible mis-configuration )

JackSpade12, 2020-03-27:
@htibosch 

> Is "Network adapter 'Xen Project' on local host" the correct adapter, the one that is connected to a LAN that leads to the Internet? Doesn't the logging show an adapter like Realtek?
'Xen Project' is the only adapter that shows on AWS cloud window instance.

lundinc2, 2020-03-27:
Hello @JackSpade12,

We have run the FreeRTOS windows simulator successfully on EC2 previously, so it is possible to do.

I suggest relaxing your inbound rules to be very permissive while you test, and trying a different port, as well as double checking that your credentials are configured correctly.

I might also suggest testing with one of the winsim TCP demos found here https://github.com/FreeRTOS/FreeRTOS/tree/master/FreeRTOS-Plus/Demo/FreeRTOS_Plus_TCP_Minimal_Windows_Simulator. That would help rule out a lot of factors such as AWS IoT Core credentials, and TLS.

JackSpade12, 2020-03-28:
@lundinc2 thanks for the reply
I tried  the FreeRTOS_Plus_TCP_Minimal_Windows_Simultaor

same exact output not establishing connection with DHCP
```
0 0 [None] Seed for randomiser: 1585358013
1 0 [None] Random numbers: 000055F5 000014DB 00007668 00004DDB
2 0 [None] FreeRTOS_IPInit
3 0 [None] vTaskStartScheduler
4 1 [IP-task] prvIPTask started

The following network interfaces are available:

Interface 1 - rpcap://\Device\NPF_{54B31D7E-36BF-4BBE-9AB2-106A939CD78C}
              (Network adapter 'Xen Project' on local host)

The interface that will be opened is set by "configNETWORK_INTERFACE_TO_USE", which
should be defined in FreeRTOSConfig.h
Attempting to open interface number 1.
Successfully opened interface number 1.
5 35 [IP-task] prvInitialiseDHCP: start after 250 ticks
6 35 [IP-task] vDHCPProcess: discover
7 5285 [IP-task] vDHCPProcess: discover
8 5285 [IP-task] vDHCPProcess: timeout 10000 ticks
```

htibosch, 2020-03-28:
I must admit that I know nothing about `AWS EC2`, except that it gives you a virtual machine in the cloud. I don't know how this works, but can other AWS developers confirm that it is compatible with Winpcap?
It looks like the WinSim applications has no communication at all.

JackSpade12, 2020-03-28:
thanks for looking into it @htibosch & @lundinc2 
The impetus for it is for establishing a continuous deployment pipeline the tests the deployment first on a window  EC2 instance using the window simulator before deploying to the target device.

I think it is an essential piece for this project's success.

tgsong, 2020-03-30:
Hi, @JackSpade12 @htibosch Our internal CI team has the same need for this, but as far as I remember, we have never successfully run the Windows Simulator with network on a EC2 instance. @lundinc2 Can you provide more context? I never heard we made any progress on this. So far what we can do is to run some unit tests that do not require network connection.

lundinc2, 2020-03-30:
@tgsong, @JackSpade12,

I apologize, I was mistaken, we have not been able to get the simulator running on an EC2 instance.

dan4thewin, 2020-04-01:
Hi, @JackSpade12 .  I think it might be useful to test that the security groups and ACLs are not at fault, separate from the winsim.  Can you make a connection to the IoT MQTT broker using a mosquitto client and publish and receive messages?  Note, to use a mosquitto client, you'll need to provide as arguments a valid thing certificate and key, a CA file, and the hostname and port of your AWS IoT endpoint.

JackSpade12, 2020-04-03:
@dan4thewin its a bit hard for me to try the Mosquito broker setup. 
Can you confirm you can run the FreeRTOS window simulator on an AWS Windows EC2 instance? From reading the above, it seems the contributors team could not get it to work.

tgsong, 2020-04-03:
Just to give more insights here, like @htibosch pointed out, the EC2's network adapter is not a real hardware, for some reason WinPcap doesn't work with it. Our team has spent effort trying to figure it out, but no progress yet. This is a known issue because of the limitation of EC2 or Winpcap, it has nothing to do with security configurations. We should document this limitation.

dcgaws, 2020-04-09:
To summarize, we have not been able to get WinPcap to work with the EC2 network adapter. (The gap appears to be equivalent to that of injecting full packets into NDIS drivers for Wi-Fi interfaces: it's not allowed by the Windows kernel. But I'm somewhat speculating that this is analogous.) Anyway, lacking new insights, I'm closing this issue.

closed 2020-04-09 by dcgaws