aws/amazon-freertos #1859: common-io test cases will have cascading failures as resources are not freed up

osloapps-max, 2020-03-26:
**Describe the bug**
For most of the common-io test cases, the resources are not freed up following a failed test case. This will cause the following tests to fail as the resource would still be considered "in-use".

**System information**
- Which hardware board or part numbers? Texas Instruments SimpleLink family
- IDE used: none
- Operating System: Windows
- Version of FreeRTOS (run ``git describe --tags`` to find it): 201912.00-151-ga6a459b5e
- Project: common-io porting

**Expected behavior**
Using the test framework "Setup" and "Tear-down" stages to allocate and free the resource before and after a test case.

**To reproduce**
Steps to reproduce the behavior:

Run generic test group (for example ADC) and force an early error.

**Additional context**

I am currently working in ports for the Texas Instruments Simplelink family (CC32XX, MSP432, CC13X2 and CC26X2). I'm also available to do/provide work on implementing these fixes (moving to using the setup/ tear-down stages) as I'm fixing these issues anyway to be able to test the ports properly.

lundinc2, 2020-03-26:
@osloapps-max thank you for raising this issue and offer to help. @qiutongs and I are available should you need any help / assistance.

osloapps-max, 2020-03-31:
Thanks @lundinc2  and @qiutongs . I'm currently working on refactoring the test cases along side my effort of implementing the ports. I'm taking it upon myself to also re-arrange / fix a few test cases that I found to be wrong (like flash test performing out of bounce writes etc.), I hope this is OK.

I'm also putting in quite some effort into the watchdog tests and reworking them so that they support implementations that cannot actually stop the timer.

qiutongs, 2020-04-06:
@osloapps-max Thank you for reporting this problem to us. I totally understand the problem of cascading failure. And I actually did solve it by using TEST_PROTECT() of the [Unity test framework](https://github.com/ThrowTheSwitch/Unity/blob/master/docs/UnityGettingStartedGuide.md). I guess you noticed this problem in ADC test. Unfortunately, we only applied TEST_PROTECT in some peripherals like UART, SPI and I2C, not all of them. Sorry about the confusion.

TEST_PROTECT works like a "try-finally". In below example, close will always be called even if there is test failure inside the if block.
```
open()

if( TEST_PROTECT() ) {
     // other test code and assertion
}

close()
```

The way of fix you suggested (setup and teardown) is also a valid alternative. But at this moment, I would suggest to make minimal changes if possible. Please consider that and let me know your thoughts.

I will go ahead reviewing your pull requests.

osloapps-max, 2020-04-07:
@qiutongs 
I'm updating the pull-requests to follow the TEST_PROTECT flow. Will do this for any other test APIs that I'm touching that do not have this already.

qiutongs, 2020-04-07:
I appreciate that and I am sorry about the extra work you have done. I will look at all the PRs and approve them if they are good.

qiutongs, 2020-04-09:
Thanks for keep pushing the fix or improvement for tests. We are actively reviewing and merging them.

qiutongs, 2020-04-30:
Three PRs are still pending review. Thank you for the patience.

tsensor https://github.com/aws/amazon-freertos/pull/1932
timer https://github.com/aws/amazon-freertos/pull/1887
flash https://github.com/aws/amazon-freertos/pull/1883

qiutongs, 2020-05-19:
We have merged all the PRs you have listed. We really appreciate that!

closed 2020-05-19 by qiutongs