aws/amazon-freertos #1864: POSIX port hangs on vTaskEndScheduler()

KKoovalsky, 2020-03-28:
**Describe the bug**
When running a Posix port the call to `vTaskEndScheduler()` hangs on `prvSuspendSelf()`.

**System information**
- Run on PC x86_64, Ubuntu 18.04.4 LTS
- Used FreeRTOS-kernel, commit hash: `464695a4f`
- Project: local unit tests

**Expected behavior**
The program is gracefully finished after a call to the `vTaskEndScheduler()`.

**To reproduce**
Steps to reproduce the behavior:

1. Set up and download the CMake project with `CMakeLists.txt`:
```
cmake_minimum_required(VERSION 3.10)

include(ExternalProject)

project(JunglesOsHelpers)

set(CMAKE_EXPORT_COMPILE_COMMANDS ON)

set(CMAKE_CXX_STANDARD 17)
set(CMAKE_C_FLAGS ${CMAKE_C_FLAGS} -Wall)
set(CMAKE_CXX_FLAGS ${CMAKE_CXX_FLAGS} -Wall)

set(FREERTOS_KERNEL_ROOT_DIR ${CMAKE_BINARY_DIR}/freertos_kernel_repo)

ExternalProject_Add(freertos_kernel_repo
    GIT_REPOSITORY https://github.com/FreeRTOS/FreeRTOS-Kernel.git
    GIT_SHALLOW 1
    GIT_PROGRESS 1
    SOURCE_DIR ${FREERTOS_KERNEL_ROOT_DIR}
    CONFIGURE_COMMAND "" 
    BUILD_COMMAND ""
    INSTALL_COMMAND ""
    TEST_COMMAND ""
)

set(FREERTOS_KERNEL_PORT_DIR ${FREERTOS_KERNEL_ROOT_DIR}/portable/ThirdParty/GCC/Posix)
set(FREERTOS_KERNEL_SOURCES
    ${FREERTOS_KERNEL_PORT_DIR}/port.c
    ${FREERTOS_KERNEL_ROOT_DIR}/portable/MemMang/heap_3.c
    ${FREERTOS_KERNEL_ROOT_DIR}/tasks.c
    ${FREERTOS_KERNEL_ROOT_DIR}/list.c
    ${FREERTOS_KERNEL_ROOT_DIR}/queue.c
    ${FREERTOS_KERNEL_ROOT_DIR}/timers.c
)
set_source_files_properties(${FREERTOS_KERNEL_SOURCES} PROPERTIES GENERATED TRUE)
add_library(freertos_kernel STATIC ${FREERTOS_KERNEL_SOURCES})
add_dependencies(freertos_kernel freertos_kernel_repo)
target_include_directories(freertos_kernel PUBLIC
    ${CMAKE_SOURCE_DIR}/test/freertos
    ${FREERTOS_KERNEL_PORT_DIR}
    ${FREERTOS_KERNEL_ROOT_DIR}/include
)
target_link_libraries(freertos_kernel pthread)

set(CATCH_ROOT_DIR ${CMAKE_BINARY_DIR}/catch2_repo)
list(APPEND CATCH2_CMAKE_ARGS 
    "-DCATCH_BUILD_TESTING=OFF" "-DCATCH_INSTALL_DOCS=OFF" "-DCATCH_INSTALL_HELPERS=OFF"
    "-DCMAKE_INSTALL_PREFIX:PATH=${CATCH_ROOT_DIR}"
)
ExternalProject_Add(catch2_repo
    GIT_REPOSITORY https://github.com/catchorg/Catch2.git
    CMAKE_ARGS ${CATCH2_CMAKE_ARGS}
)
add_library(catch2 INTERFACE IMPORTED)
add_dependencies(catch2 catch2_repo)
target_include_directories(catch2 INTERFACE ${CATCH_ROOT_DIR}/include)

add_executable(TestBuild test/freertos/main.cpp)
include_directories(${CMAKE_SOURCE_DIR})
target_link_libraries(TestBuild freertos_kernel catch2)
```

2. Configure FreeRTOS with `FreeRTOSConfig.h`:
```
#ifndef FREERTOS_CONFIG_H
#define FREERTOS_CONFIG_H

#define configUSE_PREEMPTION                       1
#define configUSE_PORT_OPTIMISED_TASK_SELECTION    0
#define configMAX_PRIORITIES                       ( 7 )
#define configTICK_RATE_HZ                         ( 1000 )                  
#define configMINIMAL_STACK_SIZE                   ( ( unsigned short ) 60 )
#define configTOTAL_HEAP_SIZE                      ( ( size_t ) ( 2048U * 1024U ) )
#define configMAX_TASK_NAME_LEN                    ( 15 )
#define configUSE_TRACE_FACILITY                   0
#define configUSE_16_BIT_TICKS                     0
#define configIDLE_SHOULD_YIELD                    1
#define configUSE_CO_ROUTINES                      0
#define configUSE_MUTEXES                          0
#define configUSE_RECURSIVE_MUTEXES                0
#define configQUEUE_REGISTRY_SIZE                  0
#define configUSE_APPLICATION_TASK_TAG             1
#define configUSE_COUNTING_SEMAPHORES              0
#define configUSE_ALTERNATIVE_API                  0
#define configNUM_THREAD_LOCAL_STORAGE_POINTERS    3      /* FreeRTOS+FAT requires 2 pointers if a CWD is supported. */
#define configRECORD_STACK_HIGH_ADDRESS            0

/* Hook function related definitions. */
#define configUSE_TICK_HOOK                        0
#define configUSE_IDLE_HOOK                        0
#define configUSE_MALLOC_FAILED_HOOK               0
#define configCHECK_FOR_STACK_OVERFLOW             0

/* Software timer related definitions. */
#define configUSE_TIMERS                           1
#define configTIMER_TASK_PRIORITY                  ( configMAX_PRIORITIES - 1 )
#define configTIMER_QUEUE_LENGTH                   5
#define configTIMER_TASK_STACK_DEPTH               ( configMINIMAL_STACK_SIZE * 2 )

/* Event group related definitions. */
#define configUSE_EVENT_GROUPS                     0

/* Run time stats gathering definitions. */
#define configGENERATE_RUN_TIME_STATS    0

/* Co-routine definitions. */
#define configUSE_CO_ROUTINES                   0
#define configMAX_CO_ROUTINE_PRIORITIES         ( 2 )

#define configSUPPORT_DYNAMIC_ALLOCATION        1
#define configSUPPORT_STATIC_ALLOCATION         1

/* Set the following definitions to 1 to include the API function, or zero
 * to exclude the API function. */
#define INCLUDE_vTaskPrioritySet                1
#define INCLUDE_uxTaskPriorityGet               1
#define INCLUDE_vTaskDelete                     1
#define INCLUDE_vTaskCleanUpResources           0
#define INCLUDE_vTaskSuspend                    1
#define INCLUDE_vTaskDelayUntil                 1
#define INCLUDE_vTaskDelay                      1
#define INCLUDE_uxTaskGetStackHighWaterMark     1
#define INCLUDE_xTaskGetSchedulerState          1
#define INCLUDE_xTimerGetTimerTaskHandle        0
#define INCLUDE_xTaskGetIdleTaskHandle          0
#define INCLUDE_xQueueGetMutexHolder            1
#define INCLUDE_eTaskGetState                   1
#define INCLUDE_xEventGroupSetBitsFromISR       1
#define INCLUDE_xTimerPendFunctionCall          1
#define INCLUDE_xTaskGetCurrentTaskHandle       1
#define INCLUDE_xTaskAbortDelay                 1

#define configUSE_STATS_FORMATTING_FUNCTIONS    0

#define configINCLUDE_DEMO_DEBUG_STATS       0

#define configCOMMAND_INT_MAX_OUTPUT_SIZE    0

/* The platform that FreeRTOS is running on. */
#define configPLATFORM_NAME    "Linux"

#endif /* FREERTOS_CONFIG_H */
```
3. Run the code:

```cpp
#define CATCH_CONFIG_RUNNER
#include "catch2/catch.hpp"

#include <algorithm>
#include <iostream>
#include <thread>

#include "FreeRTOS.h"

extern "C" void vApplicationGetIdleTaskMemory(StaticTask_t** ppxIdleTaskTCBBuffer,
                                              StackType_t** ppxIdleTaskStackBuffer,
                                              uint32_t* pulIdleTaskStackSize)
{
    static StaticTask_t xIdleTaskTCB;
    static StackType_t uxIdleTaskStack[configMINIMAL_STACK_SIZE];
    *ppxIdleTaskTCBBuffer = &xIdleTaskTCB;
    *ppxIdleTaskStackBuffer = uxIdleTaskStack;
    *pulIdleTaskStackSize = configMINIMAL_STACK_SIZE;
}

extern "C" void vApplicationGetTimerTaskMemory(StaticTask_t** ppxTimerTaskTCBBuffer,
                                               StackType_t** ppxTimerTaskStackBuffer,
                                               uint32_t* pulTimerTaskStackSize)
{
    static StaticTask_t xTimerTaskTCB;
    static StackType_t uxTimerTaskStack[configTIMER_TASK_STACK_DEPTH];
    *ppxTimerTaskTCBBuffer = &xTimerTaskTCB;
    *ppxTimerTaskStackBuffer = uxTimerTaskStack;
    *pulTimerTaskStackSize = configTIMER_TASK_STACK_DEPTH;
}

// --------------------------------------------------------------------------------------------------------------------
// --------------------------------------------------------------------------------------------------------------------

int main(int argc, char* argv[])
{
    auto t{std::thread([]() {
        std::cout << "TEST FreeRTOS: starting scheduler" << std::endl;
        vTaskStartScheduler();
        std::cout << "TEST FreeRTOS: scheduler stopped" << std::endl;
    })};

    using namespace std::chrono_literals;
    while (xTaskGetSchedulerState() != taskSCHEDULER_RUNNING) std::this_thread::sleep_for(100ms);
    std::cout << "TEST FreeRTOS: scheduler started. Running tests." << std::endl;
    int result = Catch::Session().run(argc, argv);

    std::cout << "TEST FreeRTOS: stopping scheduler" << std::endl;
    vTaskEndScheduler();
    std::cout << "TEST FreeRTOS: scheduler end commissioned" << std::endl;
    t.join();
    std::cout << "TEST FreeRTOS: finished, leaving" << std::endl;

    return result;
}
```
4. Compile and execute the program:
```
mkdir build && cd build && cmake .. && make && ./TestBuild
```
5. See the output:

```
> TEST FreeRTOS: starting scheduler
> TEST FreeRTOS: scheduler started. Running tests.
> =======================================================================
> test cases: 1 | 1 passed
> assertions: - none -
> 
> TEST FreeRTOS: stopping scheduler
> TEST FreeRTOS: scheduler stopped
>  # "TEST FreeRTOS: scheduler end commissioned" is not printed - hangs in vTaskEndScheduler
```

**Additional context**
I have added some `printf` calls in the `portable/ThirdParty/GCC/Posix/port.c` file and could see that the hang is on `vPortEndScheduler` -> `prvSuspendSelf` -> `sigwait( &xResumeSignals, &iSig );`.

I hope I have misconfigured something. ;)

Thank you!

dvrabel, 2020-03-28:
Are you running on a real Linux system? Windows Subsystem for Linux (WSL) v1 does not work due to bugs in WSLv1 wrt sigwait() with blocked signals.

vTaskEndScheduler() can only be called from a FreeRTOS task, you can't use it to stop the scheduler from another OS thread.

KKoovalsky, 2020-03-28:
Yep, running on a real Linux system.

Ok, so I have slightly modified the main function:

```cpp
int main(int argc, char* argv[])
{
    static int result{1};

    using programs_args = std::pair<int, char**>;
    programs_args arg{argc, argv};
    xTaskCreate(
        [](void* arg_pv) {
            auto arg_p{static_cast<programs_args*>(arg_pv)};
            auto [argc, argv] = *arg_p;
            std::cout << "TEST FreeRTOS: Running tests." << std::endl;
            result = Catch::Session().run(argc, argv);
            std::cout << "TEST FreeRTOS: Stopping scheduler" << std::endl;
            vTaskEndScheduler();

            std::cout << "TEST FreeRTOS: Scheduler end commissioned, but SHALL NOT GET HERE." << std::endl;
            vTaskDelete(nullptr);
        },
        "TestRunner",
        2048,
        &arg,
        1,
        nullptr);

    std::cout << "TEST FreeRTOS: Starting scheduler" << std::endl;
    vTaskStartScheduler();
    std::cout << "TEST FreeRTOS: Scheduler stopped, finished, leaving." << std::endl;

    return result;
}
```

With that approach, `valgrind` sees some "possibly lost" sections:
```
kacper ~/Workspace/JunglesOsHelpers/build $ valgrind --leak-check=full ./TestBuild ==27732== Memcheck, a memory error detector
==27732== Copyright (C) 2002-2017, and GNU GPL'd, by Julian Seward et al.
==27732== Using Valgrind-3.13.0 and LibVEX; rerun with -h for copyright info
==27732== Command: ./TestBuild
==27732== 
TEST FreeRTOS: Starting scheduler
TEST FreeRTOS: Running tests.
===============================================================================
test cases: 1 | 1 passed
assertions: - none -

TEST FreeRTOS: Stopping scheduler
==27732== Thread 2:
==27732== Syscall param setitimer(&value->it_interval) points to uninitialised byte(s)
==27732==    at 0x5A6FE57: setitimer (syscall-template.S:78)
==27732==    by 0x1E0F84: vPortEndScheduler (in /home/kacper/Workspace/JunglesOsHelpers/build/TestBuild)
==27732==    by 0x1DE589: vTaskEndScheduler (in /home/kacper/Workspace/JunglesOsHelpers/build/TestBuild)
==27732==    by 0x144B4F: main::{lambda(void*)#1}::operator()(void*) const (in /home/kacper/Workspace/JunglesOsHelpers/build/TestBuild)
==27732==    by 0x144BDC: main::{lambda(void*)#1}::_FUN(void*) (in /home/kacper/Workspace/JunglesOsHelpers/build/TestBuild)
==27732==    by 0x1E1362: prvWaitForStart (in /home/kacper/Workspace/JunglesOsHelpers/build/TestBuild)
==27732==    by 0x4E436DA: start_thread (pthread_create.c:463)
==27732==    by 0x5ABB88E: clone (clone.S:95)
==27732==  Address 0x698abe0 is on thread 2's stack
==27732==  in frame #1, created by vPortEndScheduler (???:)
==27732== 
==27732== Syscall param rt_sigaction(act->sa_mask) points to uninitialised byte(s)
==27732==    at 0x4E4E95D: __libc_sigaction (sigaction.c:62)
==27732==    by 0x1E0FAF: vPortEndScheduler (in /home/kacper/Workspace/JunglesOsHelpers/build/TestBuild)
==27732==    by 0x1DE589: vTaskEndScheduler (in /home/kacper/Workspace/JunglesOsHelpers/build/TestBuild)
==27732==    by 0x144B4F: main::{lambda(void*)#1}::operator()(void*) const (in /home/kacper/Workspace/JunglesOsHelpers/build/TestBuild)
==27732==    by 0x144BDC: main::{lambda(void*)#1}::_FUN(void*) (in /home/kacper/Workspace/JunglesOsHelpers/build/TestBuild)
[... 37 lines cut ...]
==27732==    at 0x4C31988: calloc (vg_replace_malloc.c:711)
==27732==    by 0x40134A6: allocate_dtv (dl-tls.c:286)
==27732==    by 0x40134A6: _dl_allocate_tls (dl-tls.c:530)
==27732==    by 0x4E44227: allocate_stack (allocatestack.c:627)
==27732==    by 0x4E44227: pthread_create@@GLIBC_2.2.5 (pthread_create.c:644)
==27732==    by 0x1E0E2C: pxPortInitialiseStack (in /home/kacper/Workspace/JunglesOsHelpers/build/TestBuild)
==27732==    by 0x1DDB2F: prvInitialiseNewTask (in /home/kacper/Workspace/JunglesOsHelpers/build/TestBuild)
==27732==    by 0x1DD87A: xTaskCreateStatic (in /home/kacper/Workspace/JunglesOsHelpers/build/TestBuild)
==27732==    by 0x1DFEA7: xTimerCreateTimerTask (in /home/kacper/Workspace/JunglesOsHelpers/build/TestBuild)
==27732==    by 0x1DE523: vTaskStartScheduler (in /home/kacper/Workspace/JunglesOsHelpers/build/TestBuild)
==27732==    by 0x144C85: main (in /home/kacper/Workspace/JunglesOsHelpers/build/TestBuild)
==27732== 
==27732== LEAK SUMMARY:
==27732==    definitely lost: 0 bytes in 0 blocks
==27732==    indirectly lost: 0 bytes in 0 blocks
==27732==      possibly lost: 864 bytes in 3 blocks
==27732==    still reachable: 16,544 bytes in 2 blocks
==27732==         suppressed: 0 bytes in 0 blocks
==27732== Reachable blocks (those to which a pointer was found) are not shown.
==27732== To see them, rerun with: --leak-check=full --show-leak-kinds=all
==27732== 
==27732== For counts of detected and suppressed errors, rerun with: -v
==27732== Use --track-origins=yes to see where uninitialised values come from
==27732== ERROR SUMMARY: 6 errors from 5 contexts (suppressed: 0 from 0)
```

alfred2g, 2020-04-02:
Hi All
Can we have some more info about the issue ?
@dvrabel did you investigate

Thanks, 
Alfred

KKoovalsky, 2020-04-04:
Actually, the main topic of this issue has been resolved - I'm able to finish the scheduler, thus finish the tests.

Anyway, there is this problem with `valgrind` as you can see in my previous post. Should I start a separate issue, related specifically to memory leaks, or can we continue with that?

alfred2g, 2020-04-04:
A separate issue would be preferable, as to keep the title relevant, and for better search results.
I will close this issue.. please go ahead and open a separate one
Thanks for reporting the issues

closed 2020-04-04 by alfred2g