aws/amazon-freertos #1867: A base project

Wtpaul, 2020-03-31:
Hi all.
I am having great trouble knowing where to start (after running all demos)
I would love a base project... that is ready for 'my tasks' running alongside all the great stuff aws and freertos have on demo.. in reality I dont want a single demo... i need the ability to enable ota and shadow and wifi provisioning and ... etc.
I don't have much experience with cmake and tobe honest it is the setup of the project structure that is really beating me...
If a little time could be placed on creating a (my project) directory in the correct place with a few commented functions/ tasks launching all the good aws stuff and a place where i can launch my stuff... this would be epic.

Thank you in advance.
W

**Is your feature request related to a problem? Please describe.**  
A clear and concise description of what the problem is. What are the use cases of the feature to be added? 

**Describe the solution you would like.**  
A clear and concise description of what you want to happen.

**Additional context**  
Add any other information (including rationale, requirements, or references) about the feature request here.

Thank you!

lundinc2, 2020-03-31:
Hello @Wtpaul,

Thank you for raising this issue. This issue is similar to https://github.com/aws/amazon-freertos/issues/1535. I have noted this as a "+1" to the feature request we are tracking. 

In the meantime, is there anything specific I can help you troubleshoot, in order to get your application started?

Wtpaul, 2020-03-31:
Thanks for quick reply @lundinc2 

Have just read #1535 -- sorry hadn't seen this before posting.
@monkeytronics this is **exactly** my experience.
I have years of c experience... but trying to put a project together without using demo code... but using all of the features from demo code _have_ driven me crazy over last weeks!!

I expect that only one mqtt needs created for the ota and shadow tasks, etc... but the demo code launches mqtt from within the same task as the rest of the demo... a good first step might be to extract the mqtt link init from the demo task allowing multiple tasks to make use of the comms to the aws servers. (if this is not a correct assumption please say)

from #1535 you have said this is being worked on... is this days, weeks or longer away?

Thanks in advance

dcgaws, 2020-04-09:
@Wtpaul based on my own experience, I suggest making and testing the changes you need first in our simulator (or, lacking that, a third-party one). As you say, our libraries are designed to be able to share an MQTT connection, but our demos are feature-specific. If you were to start with the OTA demo, say, and refactor the MQTT connection logic into your app startup logic, you could then proceed by adding Shadow, for example. But, frankly, doing that for the first time directly on a developer board or module, rather than in a simulator or emulator, sounds needlessly painful. 

My second suggestion is to use an IDE rather than make/cmake. For what it's worth, that guidance is consistent with the feedback that we get from our silicon partners as well as most line-of-business embedded developer customers. The FreeRTOS team uses cmake primarily to facilitate test automation across a wide variety of MCU boards; few of our individual customers have that need.

If there are more specifics you can share about hardware or issues you're running into, then please do.

closed 2020-04-13 by dcgaws