aws/amazon-freertos #1874: Occourence of IP event even if device is disconnected from WiFi network

shubhamkulkarni97, 2020-04-03:
**Describe the bug**
When a Wi-Fi network is disconnected, after around 10 minutes of disconnection `vApplicationIPNetworkEventHook` is called with `eNetworkUp` which triggers the IP event.

DHCP lease time is set to 2 minutes.

**System information**
- On ESP32 platform
- Version of FreeRTOS is 201912.00-164-ge984d9463
- Issue is observed with aws_demos as well as aws_tests when using FreeRTOS TCP stack (works fine with LwIP networking stack).

**Expected behavior**
IP event should not trigger when Wi-Fi is disconnected.

**To reproduce**
Steps to reproduce the behavior:

1. Reduce DHCP lease time to 2 minutes.

**Code to reproduce the bug**
I'm attaching a patch which will help reproduce this issue.
[bug_ip_event.patch.zip](https://github.com/aws/amazon-freertos/files/4425604/bug_ip_event.patch.zip)

I'm also attaching console log for the same.

[image]

Thank you!

dcgaws, 2020-04-09:
@shubhamkulkarni97 thank you, I'm pinging the author of the FreeRTOS+TCP library for further comment.

htibosch, 2020-04-11:
Hello Shubham, I tried out the **aws_tests** application with both lwIP and FreeRTOS+TCP.
I applied your patch that stops WiFi and avoids executing the tests.:
~~~c
    WIFI_Disconnect();
    vTaskDelay(portMAX_DELAY);
~~~
Indeed, once the lease time is over, the FreeRTOS+TCP will try to apply for a DHCP extension. As a consequence, there will be unwanted WiFi events.

I will prepare a patch for this. It's a change in FreeRTOS_DHCP, around line 491:
~~~c
 case eLeasedAddress :
 
 +   if( FreeRTOS_IsNetworkUp() )
 +   {
         /* Resend the request at the appropriate time to renew the lease. */
         prvCreateDHCPSocket();
 
         if( xDHCPData.xDHCPSocket != NULL )
         {
             xDHCPData.xDHCPTxTime = xTaskGetTickCount();
             xDHCPData.xDHCPTxPeriod = dhcpINITIAL_DHCP_TX_PERIOD;
             prvSendDHCPRequest();
             xDHCPData.eDHCPState = eWaitingAcknowledge;
 
             /* From now on, we should be called more often */
             vIPReloadDHCPTimer( dhcpINITIAL_TIMER_PERIOD );
         }
+    }
+    else
+    {
+        FreeRTOS_printf( ( "DHCP: lease time finished but network is down\n" ) );
+        vIPReloadDHCPTimer( pdMS_TO_TICKS( 5000 ) );
+    }
     break;
~~~

htibosch, 2020-04-11:
Something else: once the ESP32 was connected, I rebooted my WiFi router. My laptop reconnected within a minute.
The ESP32 application however, never reconnected again. I tried this both with the lwIP and with FreeRTOS+TCP stack. Is that expected behaviour?

Thanks, Hein

htibosch, 2020-04-11:
I created a [patch #53](https://github.com/FreeRTOS/FreeRTOS/pull/53) that should fix this issue.

shubhamkulkarni97, 2020-04-14:
Hi @htibosch , thank you for the fix.

Regarding the reconnection issue, it is expected that ESP32 will not reconnect to the WiFi network.

Reconnection should be handled by the application.

After disconnection `eNetworkStateDisabled` occurs. https://github.com/aws/amazon-freertos/blob/1c82352a62006ed716a9c7c787e2e40a0022d7d2/vendors/espressif/boards/esp32/ports/wifi/iot_wifi.c#L138

To attempt reconnection, `WIFI_ConnectAP` should be called from the application when eNetworkStateDisabled occurs.

gkwicker, 2020-05-04:
Hi @shubhamkulkarni97, I'm going to close this issue as we are tracking the fix [in FreeRTOS PR #53](https://github.com/FreeRTOS/FreeRTOS/pull/53).  If you need anything else for this, feel free to open a new issue and thank you for bringing it to our attention.

closed 2020-05-04 by gkwicker