aws/amazon-freertos #1876: Valgrind outputs errors for POSIX port

KKoovalsky, 2020-04-04:
**Describe the bug**
When running a Posix port and using `vTaskEndScheduler()` `valgrind` raises memory leaks.

**System information**
- Run on PC x86_64, Ubuntu 18.04.4 LTS
- Used FreeRTOS-kernel, commit hash: `464695a4f`
- Project: local unit tests

**Expected behavior**
No errors are found by `valgrind`.

**To reproduce**
Steps to reproduce the behavior:

1. Configure FreeRTOS with `FreeRTOSConfig.h`:
```
#ifndef FREERTOS_CONFIG_H
#define FREERTOS_CONFIG_H

#define configUSE_PREEMPTION                       1
#define configUSE_PORT_OPTIMISED_TASK_SELECTION    0
#define configMAX_PRIORITIES                       ( 7 )
#define configTICK_RATE_HZ                         ( 1000 )                  
#define configMINIMAL_STACK_SIZE                   ( ( unsigned short ) 60 )
#define configTOTAL_HEAP_SIZE                      ( ( size_t ) ( 2048U * 1024U ) )
#define configMAX_TASK_NAME_LEN                    ( 15 )
#define configUSE_TRACE_FACILITY                   0
#define configUSE_16_BIT_TICKS                     0
#define configIDLE_SHOULD_YIELD                    1
#define configUSE_CO_ROUTINES                      0
#define configUSE_MUTEXES                          0
#define configUSE_RECURSIVE_MUTEXES                0
#define configQUEUE_REGISTRY_SIZE                  0
#define configUSE_APPLICATION_TASK_TAG             1
#define configUSE_COUNTING_SEMAPHORES              0
#define configUSE_ALTERNATIVE_API                  0
#define configNUM_THREAD_LOCAL_STORAGE_POINTERS    3      /* FreeRTOS+FAT requires 2 pointers if a CWD is supported. */
#define configRECORD_STACK_HIGH_ADDRESS            0

/* Hook function related definitions. */
#define configUSE_TICK_HOOK                        0
#define configUSE_IDLE_HOOK                        0
#define configUSE_MALLOC_FAILED_HOOK               0
#define configCHECK_FOR_STACK_OVERFLOW             0

/* Software timer related definitions. */
#define configUSE_TIMERS                           1
#define configTIMER_TASK_PRIORITY                  ( configMAX_PRIORITIES - 1 )
#define configTIMER_QUEUE_LENGTH                   5
#define configTIMER_TASK_STACK_DEPTH               ( configMINIMAL_STACK_SIZE * 2 )

/* Event group related definitions. */
#define configUSE_EVENT_GROUPS                     0

/* Run time stats gathering definitions. */
#define configGENERATE_RUN_TIME_STATS    0

/* Co-routine definitions. */
#define configUSE_CO_ROUTINES                   0
#define configMAX_CO_ROUTINE_PRIORITIES         ( 2 )

#define configSUPPORT_DYNAMIC_ALLOCATION        1
#define configSUPPORT_STATIC_ALLOCATION         1

/* Set the following definitions to 1 to include the API function, or zero
 * to exclude the API function. */
#define INCLUDE_vTaskPrioritySet                1
#define INCLUDE_uxTaskPriorityGet               1
#define INCLUDE_vTaskDelete                     1
#define INCLUDE_vTaskCleanUpResources           0
[... 124 lines cut ...]
==7996==    by 0x1E3657: prvInitialiseNewTask (in /home/kacper/Workspace/JunglesOsHelpers/build/TestBuild)
==7996==    by 0x1E33A2: xTaskCreateStatic (in /home/kacper/Workspace/JunglesOsHelpers/build/TestBuild)
==7996==    by 0x1E6B63: xTimerCreateTimerTask (in /home/kacper/Workspace/JunglesOsHelpers/build/TestBuild)
==7996==    by 0x1E404B: vTaskStartScheduler (in /home/kacper/Workspace/JunglesOsHelpers/build/TestBuild)
==7996==    by 0x1454F1: main (in /home/kacper/Workspace/JunglesOsHelpers/build/TestBuild)
==7996== 
==7996== LEAK SUMMARY:
==7996==    definitely lost: 0 bytes in 0 blocks
==7996==    indirectly lost: 0 bytes in 0 blocks
==7996==      possibly lost: 864 bytes in 3 blocks
==7996==    still reachable: 16,544 bytes in 2 blocks
==7996==         suppressed: 0 bytes in 0 blocks
==7996== Reachable blocks (those to which a pointer was found) are not shown.
==7996== To see them, rerun with: --leak-check=full --show-leak-kinds=all
==7996== 
==7996== For counts of detected and suppressed errors, rerun with: -v
==7996== Use --track-origins=yes to see where uninitialised values come from
==7996== ERROR SUMMARY: 6 errors from 5 contexts (suppressed: 0 from 0)
kacper ~/Workspace/JunglesOsHelpers/build (master) $ 
```

**Additional context**
Originates in: https://github.com/aws/amazon-freertos/issues/1864#issue-589565527

Thank you!

dcgaws, 2020-04-09:
@alfred2g can you please comment? Thank you!

alfred2g, 2020-04-10:
Hi @KKoovalsky 
I will be looking into the issue, I will get back to you if I have any questions

Thanks,

alfred2g, 2020-04-20:
I fixed the first 2 errors of uninitialized memory, I will push the commit soon.

I will continue looking into the possibly lost memory leak issues, for some reason, pthread_join is not being called to free the memory

alfred2g, 2020-04-20:
I ran locally with valgrind, no trace of any memory leaks,

==21880== LEAK SUMMARY:
==21880==    definitely lost: 0 bytes in 0 blocks
==21880==    indirectly lost: 0 bytes in 0 blocks
==21880==      possibly lost: 0 bytes in 0 blocks
==21880==    still reachable: 20,784 bytes in 7 blocks
==21880==         suppressed: 0 bytes in 0 blocks
==21880== 
==21880== For counts of detected and suppressed errors, rerun with: -v
==21880== ERROR SUMMARY: 0 errors from 0 contexts (suppressed: 0 from 0)
Built target run-test

 Could you please share the test you are performing, if you can provide a runnable program source with a makefile/cmake it would be great 

Thanks,

Alfred

alfred2g, 2020-04-26:
Hi,
Pushed a pull request to address the uninitialized variables
 
https://github.com/FreeRTOS/FreeRTOS-Kernel/pull/56

Alfred

KKoovalsky, 2020-04-27:
Hello,

sorry for such a late reply. You could clone, build and run the project in this repository:
https://github.com/KKoovalsky/JunglesOsHelpers

The code I pasted in this issue comes from that project.

alfred2g, 2020-04-28:
Hi @KKoovalsky, 
Thanks for the link, i was able to reproduce the exact same valgrind error you were seeing.

I think this is a case of a false positive, as the memory is coming from pthread_create in 3 places.
* The timer thread (https://github.com/FreeRTOS/FreeRTOS-Kernel/blob/master/timers.c#L548)
* The task thread (https://github.com/FreeRTOS/FreeRTOS-Kernel/blob/master/tasks.c#L3392)
* The user thread

which are not supposed to end (first 2 with infinite loops), and suspend scheduler is not exiting as well (we could change that to exit if necessary)  main is exiting before the threads

Thanks,

Alfred

KKoovalsky, 2020-04-28:
@alfred2g do you know whether there is a way to somehow suppress those errors or get rid of them? The thing is that I could ignore them, but it's really hard to figure out which errors come from my application and which come from the FreeRTOS kernel.

alfred2g, 2020-04-28:
You might replace the infinite loop with a global variable that it could check to exit the loop, then try to pthread join it
you could also make it return from vTaskEndScheduler, and kill the thread, then try to join it

Second way, is to disallow dynamic memory allocation in the kernel, like using xTaskCreateStatic for example (and change the config file to turn configSUPPORT_DYNAMIC_ALLOCATION off)

Lastly, you could have a script to parse the results ,and ignore these 3 false positives

Hope this helps

alfred2g, 2020-04-28:
Code to fix the uninitialized references [Pull Request] (https://github.com/FreeRTOS/FreeRTOS-Kernel/pull/56)

I will close the issue, Please reopen if anything changes.

closed 2020-04-28 by alfred2g