aws/amazon-freertos #1925: WebRTC support for Freertos

kirantpatil, 2020-04-21:
We would like to create ring door bell using TI CC3220SF and for which we need Webrtc as advertised [here](https://aws.amazon.com/kinesis/video-streams/).

**Describe the solution you would like.**  
Add WebRTC protocol to FreeRTOS

**Additional context**  
The more protocols are added for example RTMP ... is much better build innovative solutions.

Thank you!

kirantpatil, 2020-04-21:
We are really really awaiting this protocol inclusion, since we are planning to use Amazon Kinesis as mentioned in https://aws.amazon.com/kinesis/video-streams/ to build solutions and sell in the market. We could not build our solution with our HW because it supports only RTSP protocol which not many streaming providers supports.

What can we do to escalate this feature request such that it will be built and delivered with at most highest priority ?

yourslab, 2020-04-26:
Have you considered using https://github.com/awslabs/amazon-kinesis-video-streams-webrtc-sdk-c?

kirantpatil, 2020-04-27:
@yourslab, 

we haven't. I will check with our development team about it.

Thanks.

aggarg, 2020-04-28:
Thank your for this feature request. As you rightly pointed out, currently WebRTC protocol is not supported in FreeRTOS. We value your feedback as it helps us in determining the priorities of the features we implement.

Thank you once again.

zekageri, 2021-09-13:
Still nothing in this request?