aws/amazon-freertos #1940: Examples of Custom Projects or Documentation

jsinglet, 2020-04-27:
In following the documentation, it seems that the "correct" way to develop your custom application is to drop freertos in a directory, create a src directory at the top level, and build from there. Something like this:

```
freertos
src
CmakeLists.txt
```

However, there is no MWE for how to do this or what should go in the main.c file. I've managed to cobble together things by looking at the demos but I think much more needs to be done to explain this. For example, I learned that there are various functions that must be implemented for it to compile, for example: `vApplicationDaemonTaskStartupHook`. I've combed through the documentation and I cannot find any indication of what the requirements for standalone applications are. Also, no mention of the details of `app_main` in the case of the ESP32, which is the entry point that is expected. 

I think what is needed is a way to understand the structure of the application and specific instructions on how to pull in and use each component rather than trying to work it out from the demo code. 

Additionally, there is a graphic in the documentation that suggests that the way one develops an app is by creating a custom distribution of freertos in the console, downloading it, and then developing your app. It is not clear however what that last step means. Is the intention that you modify the demos? 

I think without these items it will be very hard to use amazon-freertos for many people.

aggarg, 2020-04-28:
>I think what is needed is a way to understand the structure of the application and specific instructions on how to pull in and use each component rather than trying to work it out from the demo code.

Thank for providing the feedback about our documentation. We strive to continuously improve our documentation to best serve our customers and we look forward to the feedback. 

Meanwhile, you can look at this project created by my teammate @tgsong to demostrate how to set up a CMake project for ESP32 and use FreeRTOS as an external library: https://github.com/tgsong/amazon-freertos-examples

> Additionally, there is a graphic in the documentation that suggests that the way one develops an app is by creating a custom distribution of freertos in the console, downloading it, and then [...]

Yes, the demos are provided to show how to use our libraries and depending on your requirements, may serve to be a starting point. 

Thanks.

jsinglet, 2020-04-28:
thanks @aggarg -- for what it's worth, this was also helpful to me and possibly to future searchers: https://github.com/kevinresol/amazon-freertos-hello-world

> Yes, the demos are provided to show how to use our libraries and depending on your 
> requirements, may serve to be a starting point.

I think the confusion for me and perhaps others is if the intent is to just modify the demos where they exist and base your application off of those. 

From an outsider perspective, a LOT of text is dedicated to talking about the demos and just a few paragraphs about how one might go about developing their own app, which is probably what most people are reading for. 

So perhaps these could be helpful a suggestions from the outside looking in:

- What I've seen others do (like espressif) is provide example projects that show their kit fully integrated with with a main program that uses the framework. A lot can go wrong setting up all these toolchains to work and just having something you can clone and get to work as a starter project would be very helpful. 

- Things like STMCube make it really easy to get a project started. This takes a code generation approach, but it might not be a bad concept for how to allow people to quickly and reliably create projects. The output of a system like this would probably a directory with a git submodule pointing to freertos, and makefiles with the correct components, boilerplate and so on already added. 

- The demos are helpful but they way they are integrated with amazon freertos is quite opaque and confusing. They feel like they are written at the level that an internal unit test is. It is a fair bit of work to go plumbing around the demos directory. Having them stand alone, apart from the core repo, might be a good starting point for making them more approachable.

- As a side point, although code reuse is good, it also makes the examples not self contained. There is quite a bit of code reuse and cross module coupling within the demos. One has to do quite a bit of work to understand what is being pulled in from the "demo library" and what is part of the "demo."

aggarg, 2020-04-28:
Thank you for all the suggestions - these are really good suggestions both about the documentation and the demo code. We will try to improve both.

Thanks.

BasemTarek, 2020-05-03:
Hello,
I still cannot understand how i can develop my custom app using Amazon FreeRTOS, and i did not understand how this project can help me do that.
> https://github.com/tgsong/amazon-freertos-examples 
I've tried to use it without any enabled demo but i get the following error:
`#error "No demo to run. One demo should be enabled"`

Is this mean that the only way to develop my application is to modify the demo code..!?
Can you give me some insight about how can i do that, where is the point that i can start from to do that..?

> I think what is needed is a way to understand the structure of the application and specific instructions on how to pull in and use each component rather than trying to work it out from the demo code.

 What i need is to use AFreeRTOS for esp32 to connect through AWS Greengrass Core device to AWS IoT Core to publish sensor readings. And i cannot modify the `iot_demo_mqtt.c` to publish my own messages to my specific topic as long as i want, the demo publishes this specific payload `#define PUBLISH_PAYLOAD_FORMAT                   "Hello Basem %d!"` for a limited number of times and i don't know how to modify this number even if i modify the following `#define TOPIC_FILTER_COUNT                       ( 4 )` i get some errors.

Thank you.

aggarg, 2020-05-04:
If I understand correctly, you want to publish some data to AWS IoT via Greengrass. You will need to setup a Greengrass core first. Assuming that you have a Greengrass core running, publishing data from your edge device requires the following two steps:

* Discover the Greengrass core.
* Publish the data over MQTT.

The following demo shows how to do the above two steps: https://github.com/aws/amazon-freertos/blob/master/demos/greengrass_connectivity/aws_greengrass_discovery_demo.c

This page documents the above demo: https://docs.aws.amazon.com/freertos/latest/userguide/gg-demo.html

Hope that helps.

Thanks.

yanjos-dev, 2020-08-12:
I'm going to close this issue since it has become inactive. Please reopen this issue if you need any further assistance.

closed 2020-08-12 by yanjos-dev