aws/amazon-freertos #1960: Documentation for regenerating untrusted OTA certificates in tools/ota_e2e_tests/credentials

rpcme, 2020-04-30:
**Is your feature request related to a problem? Please describe.**  

The documentation for recreating the untrusted certificate/key pairs for e2e testing either do not exist or the documentation is not easy to find.  Right now, the committed certificate/key pair fails to import with the following error at tag 202002.00.  The resulting ARN is required for e2e and IDT test configuration.

```bash
     #+begin_src bash :tangle /tmp/upload_untrusted.sh
SRC=/home/X/src/amazon-freertos-pic32mzef-ota
OTA_TOOLS=${SRC}//tools/ota_e2e_tests
KEY_UNTRUSTED=${OTA_TOOLS}/credentials/untrusted-ecdsa-sha256-signer.key.pem
CERT_UNTRUSTED=${OTA_TOOLS}/credentials/untrusted-ecdsa-sha256-signer.crt.pem

aws acm import-certificate \
    --certificate fileb://${CERT_UNTRUSTED} \
    --private-key fileb://${KEY_UNTRUSTED}
     #+end_src

     #+RESULTS:
An error occurred (ValidationException) when calling the ImportCertificate operation: The certificate is expired. You can Import a certificate only during its validity period.
```
**Describe the solution you would like.**  
Documentation in README.md on how to recreate the key/certificate pair according to testing guidelines.

tgsong, 2020-05-05:
Hi @rpcme thanks for reporting the issue, I submitted a PR #1972 for this. Meanwhile, you can find the document for how to generate certificate here, https://docs.aws.amazon.com/freertos/latest/userguide/ota-code-sign-cert.html.

muneebahmed10, 2020-05-11:
Hi @rpcme 

Thank you for creating this issue. I will close it as PR #1972 has been merged.

closed 2020-05-11 by muneebahmed10