aws/amazon-freertos #1969: Espressif projects build with LWIP by default (no longer possible to build with native FreeRTOS_plus_tcp  stack)

bgintz-rsc, 2020-05-04:
**Describe the bug**
Espressif ESP32 builds default to  #define CONFIG_TCPIP_LWIP 1 even without specifying -DAFR_ESP_LWIP=1

**System information**
- Espressif ESP32
- Can replicate using command line
- MacOS
- Version of FreeRTOS: 201912.00-280-gd88755b0
**Expected behavior**
Previously - to build using the LWIP stack it was necessary to add -DAFR_ESP_LWIP=1 to the cmake command line options. If this was not added then the native FreeRTOS_plus_tcp stack was used.  Now the project is built with lwip even without requesting it. 
**To reproduce**
Steps to reproduce the behavior:

1. Run cmake as described here: https://docs.aws.amazon.com/freertos/latest/userguide/getting_started_espressif.html
2. Verify that the code is built using lwip instead of FreeRTOS_plus_tcp

**Additional context**
Has support for the native stack been deprecated from ESP32 builds?  If so where is this documented

Thank you!

Kindest regards,
Bobby

tgsong, 2020-05-05:
Hi @bgintz-rsc , Espressif switched to use lwip by default as in this PR, https://github.com/aws/amazon-freertos/pull/1886. The native FreeRTOS plus TCP stack was not deprecated, to enable it, you can pass the `-DAFR_ESP_FREERTOS_TCP=1` flag to cmake. Our public documentation has not been updated yet since this change is not yet released.

shubhamkulkarni97, 2020-05-05:
Hi @bgintz-rsc ,

Many features (like SoftAP provisioning) had dependencies on LwIP networking stack. Hence, we have switched the default stack to LwIP.

Adding to it, we also found some bugs in FreeRTOS TCP stack, like:
* https://github.com/aws/amazon-freertos/pull/1696
* https://github.com/aws/amazon-freertos/issues/1874

Following are the features of LwIP stack:
* Lower memory consumption.
* Zero copy receive path.
* Simultaneous multi interface and Posix API support.

Native SDK for ESP32 (ESP-IDF) uses LwIP networking stack.

We recommend using LwIP stack as it is tested on ESP32 by our internal team.

However, if you want to use FreeRTOS TCP stack, you can pass `-DAFR_ESP_FREERTOS_TCP=1 ` flag to CMake.

Thanks!

htibosch, 2020-05-05:
@shubhamkulkarni97 :

> Adding to it, we also found some bugs in FreeRTOS TCP stack, like:
> 
>     #1696

You posted a PR that was adequate to repair this. Thanks a lot for that.

>    #1874

I already responded with [PR #53](https://github.com/FreeRTOS/FreeRTOS/pull/53) that should repair it. I'm still waiting for a response.

> Following are the features of LwIP stack:
> 
>     Lower memory consumption.
>     Simultaneous multi interface and Posix API support.
>     Zero copy receive path.

You have stated this before and I already responded to it [here](https://github.com/aws/amazon-freertos/pull/1886#issuecomment-610875684).
We are working hard to get the new multi-interface version tested and released.

FreeRTOS+TCP is smaller and simpler than lwIP, and that has advantages. It is fully integrated with FreeRTOS. If it has the features needed for your project, I recommend using it. 
And if it misses features, feel free to use any other TCP/IP stack that you like.

bgintz-rsc, 2020-05-05:
Thank you for the answer!!!

bgintz-rsc, 2020-05-05:
@htibosch @tgsong @shubhamkulkarni97 

Is there any functionality in the native stack that is NOT supported by the LWIP stack?  i.e. any downside to switching stacks at this time?

Thanks

htibosch, 2020-05-05:
> The -DAFR_ESP_FREERTOS_TCP=1 option should help us do so

I can confirm that the option works fine.

> Is there any functionality in the native stack that is NOT supported by the LWIP stack?

FreeRTOS+TCP doesn't have major features ( functionalities )  that lwIP doesn't have.

muneebahmed10, 2020-05-11:
Hi @bgintz-rsc since your question has been resolved, I will close this issue. Please reopen or create a new issue for further questions

closed 2020-05-11 by muneebahmed10