aws/amazon-freertos #1973: SIM800L TCP communication support

ahmedwahdan, 2020-05-05:
Is there any current implementation for supporting TCP communication using SIM800 (GPRS modems)?
If not, what are the files/fiunctions/configurations that could be wrapped to be able to use my current implementation?

Thank you!

htibosch, 2020-05-05:
I just had a quick look at the SIM800 AT commands. That allows to open a TCP connection, meaning that the device already has a built-in TCP/IP{ stack.
 Do you know if it has a possibility to pass a raw Ethernet II packets?

ahmedwahdan, 2020-05-05:
@htibosch I don't know much about the SIM800 functionality, I use a driver written by a colleague, which use the TCP layer of the modem.

nickfaughey, 2020-05-05:
See my earlier comment on #22 - basically you need to bring your own `aws_secure_sockets.c` functionality, and it should be a fairly clean swap.

ahmedwahdan, 2020-05-06:
@nickfaughey  yeah, I see that comment and will start implement this layer.
Thanks a lot!

closed 2020-05-06 by ahmedwahdan

ahmedwahdan, 2020-05-07:
@nickfaughey  another question before starting the implementation, could we use different broker other than AWS IOT? because we are developing our back-end environment.

nickfaughey, 2020-05-07:
I don't see why not, but that's independent of the communication layer. I'm not sure how much of the broker logic in amazon-freertos is AWS-flavored, but theoretically it should all be swappable.