aws/amazon-freertos #1974: Handshake Failed

WaterDemo, 2020-05-05:
Hello, I want to ask a question. 
I used the development board CC3220SF-LAUNCHXL to burn AmazonFreeRTOS. When I run code in debug mode on CCS, I can communicate with the server normally. But when I burned the bin file into the board with uniflash, the following error occurred：

[image]

Why did this happen? Looking forward to your reply. Thank you！

tgsong, 2020-05-06:
Error -456 seems like related to the root certificate,
https://github.com/aws/amazon-freertos/blob/8106b1d280daec3d0b4aa33d08a9e9785f7baeab/vendors/ti/SimpleLink_CC32xx/v2_10_00_04/source/ti/drivers/net/wifi/errors.h#L126
How did you burn the bin file to board? Can you try following this doc? https://docs.aws.amazon.com/freertos/latest/userguide/burn-initial-firmware-ti.html

WaterDemo, 2020-05-06:
It solved my problem perfectly, thank you very much!

closed 2020-05-06 by WaterDemo