aws/amazon-freertos #1976: OTA demo on TI CC3235SF

HansHabraken, 2020-05-05:
I'm trying to run the OTA demo on the CC3235SF launchpad board from Texas Instrument. I know that this chip is not officially supported by Amazon FreeRTOS, but normally this should work, because it uses the same SDK (Simplelink CC32XX SDK) as the supported CC3220SF chip.

When running the demo, everything seems to work just fine. The demo can download the firmware image over MQTT and saves it on the SFLASH. The image is then set to the PENDING_COMMIT state and a reboot is executed, to test the new image. After the reboot, the chip does not startup and nothing is showing up in the output console. To get the chip out of this 'lock' state, I need to manual reset the chip what results in a rollback of the downloaded firmware 
image.

Here are the console logs:
[ota_demo_output.txt](https://github.com/aws/amazon-freertos/files/4579723/ota_demo_output.txt)

I have tested the code on a CC3220SF and it works like expected!

I contacted TI about this problem and they said that it should work if I use to correct service pack, certificates and certificate catalog.

Does anyone know what the problem is or can someone point me in the right direction to solve it?

Thanks in advance!
Hans

HansHabraken, 2020-05-11:
Hey

I've set a breakpoint on the requested line and I can confirm that this will reset the device. When I comment the line, an error message is displayed indicating that a software reset is not possible and that a hardware reset must be performed (using the reset switch SW1). After I perform the hardware reset, the device comes in the same 'lock state' as with a software reset. This also confirms that the MAP_PRCMHibernateCycleTrigger does reset the microcontroller.

Here are the logs with some more information in it:
[screenlog.txt](https://github.com/aws/amazon-freertos/files/4611358/screenlog.txt)

I currently use the latest version of the simpleLink SDK (simplelink_cc32xx_sdk_4_10_00_07). But I have also tested it with the previous version (simplelink_cc32xx_sdk_3_40_00_05). You are correct that the service pack is different for CC3235SF device. I have flashed the correct service pack on the device (tools/cc32xx_tools/servicepack-cc3x35/sp_4.6.0.1_3.1.0.5_3.1.0.25.bin)

Greetings
Hans

pvyawaha, 2020-05-15:
Hello,

Thank you for trying the steps to debug it further. We are working on getting this platform to debug this issue further and I will share updates here.

HansHabraken, 2020-05-28:
Hey

Is it possible to provide me with an update about this issue?

Thanks
Hans

pvyawaha, 2020-05-29:
Hello, 

We have received the hardware for this platform and work is in progress. I will update here soon with solution.

lerebel103, 2020-07-07:
Hi,

Has there been some progress on this since please? I am keen to see if there is anything I can do to help track this down. 

Just as you've described, the firmware doesn't seem to be written to flash with CC3235SF. If it helps someone, on CC3220SF I have a little more luck, however the reset process does not work either and I don't believe this relates to `amazon-freertos`, to this effect I posted on the TI forums: https://e2e.ti.com/support/wireless-connectivity/wifi/f/968/p/919422/3400829#3400829
 
Regards,
Will.

pvyawaha, 2020-07-13:
Hello,

We are observing the same issue as you described in the TI Forum. The call to `sl_Stop` does not return any error but still call to `PRCMHibernateCycleTrigger ` does not seem to reset the device. From the documentation `PRCMHibernateCycleTrigger` seems to be correct function to reset the SOC after calling `sl_Stop`. If we do not have any update on the TI forum I will try to get support on this issue using our channel.

lerebel103, 2020-08-12:
@pvyawaha 

Oh this is interesting, as per post on this [TI issue](https://e2e.ti.com/support/wireless-connectivity/wifi/f/968/t/926457), could it be the signature that needs to be sha256?

Apologies, I am unable to verify this over the next few weeks, but hope it can help someone.

Regards,
Will.

pvyawaha, 2021-06-07:
Hello , 

Can you please confirm if using SHA 256 solve the issue and this ticket can be closed?

HansHabraken, 2021-06-09:
Hey,

Yes, using SHA 256 solved this issue for us. Closing this issue now.

closed 2021-06-09 by HansHabraken