aws/amazon-freertos #1979: Espressif China Mobile (Arris Router) fix appears to not be merged into AFR ESP-IDF repo

bgintz-rsc, 2020-05-05:
See comment also reported on  Espressif site's issue: 

https://github.com/espressif/esp-idf/issues/4843

This fix appears not to have been completely merged into the AFR version of the ESP-IDF repo which is located at: https://github.com/espressif/esp-afr-sdk

The reason for questioning this is that the the lib submodule (under components) in the above repo points to a detached HEAD at commit d8ed359 instead of v3.3 which includes the update to the wifi lib files. If we build the current master in the amazon-freertos repo it appears that this fix will not be included.

We are unable to build at the desired commit with the fix as cmake performs a repo refresh which overrides our selected commit.

Please advise.

tgsong, 2020-05-06:
I'm not aware of the plan on Espressif side to merge this fix. @mahavirj could you provide some details here or point to someone who knows this specific issue? Meanwhile, @bgintz-rsc as a temporary workaround, you can comment out this line https://github.com/aws/amazon-freertos/blob/master/tools/cmake/afr.cmake#L38 to stop cmake from checking out the submodule.

mahavirj, 2020-05-06:
@bgintz-rsc I am following up on this internally, shall keep you posted at https://github.com/espressif/esp-idf/issues/4843

bgintz-rsc, 2020-05-06:
Thank you Mahavirj!

tgsong, 2020-05-06:
closing this one as it's being tracked in espressif/esp-idf#4843

closed 2020-05-06 by tgsong

bgintz-rsc, 2020-05-12:
@mahavirj -  The commit still doe not appear to be merged in the air repo as the HEAD of the esp-idf still points to commit ee3a98507 and therefore requires the workaround above to build.  I do not wish to test this as a patch as suggested in the Espressif issue because we build our code from the AFR repo.   We are engaging a third party at our expense to test this fix for us.  It appears that the bt submodule is also at a HEAD older than the latest 3.3. 

Please advise - thank you