aws/amazon-freertos #1990: Unable to Build FreeRTOS.

dev-techshlok, 2020-05-10:
When i run the build command given in AWS RTOS getting started page it shows this error. 
I am compiling for ESP32. 

```

root@iot-virtual-machine:~/esp/amazon-freertos# ls
CHANGELOG.md	CMakeFiles	    CONTRIBUTING.md  doc	      LICENSE	     README.md	vendors
checksums.json	CMakeLists.txt	    demos	     freertos_kernel  PreLoad.cmake  tests
CMakeCache.txt	CODE_OF_CONDUCT.md  directories.txt  libraries	      projects	     tools
root@iot-virtual-machine:~/esp/amazon-freertos# cmake -DVENDOR=espressif -DBOARD=esp32_devkitc -DCOMPILER=xtensa-esp32 -S . -B demos/
CMake Error at CMakeLists.txt:5 (afr_demo_module):
  Unknown CMake command "afr_demo_module".

CMake Warning (dev) in CMakeLists.txt:
  No cmake_minimum_required command is present.  A line of code such as

    cmake_minimum_required(VERSION 3.10)

  should be added at the top of the file.  The version specified may be lower
  if you wish to support older CMake versions for this project.  For more
  information run "cmake --help-policy CMP0000".
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Configuring incomplete, errors occurred!
See also "/root/esp/amazon-freertos/CMakeFiles/CMakeOutput.log".
See also "/root/esp/amazon-freertos/CMakeFiles/CMakeError.log".
root@iot-virtual-machine:~/esp/amazon-freertos# 

```

muneebahmed10, 2020-05-10:
Hi @dev-techshlok 

The `demos/` directory should not be your build directory. Can you try creating a new directory? E.g. from `~/esp/amazon-freertos/` run:
```
mkdir build
cmake -DVENDOR=espressif -DBOARD=esp32_devkitc -DCOMPILER=xtensa-esp32 -S . -B build
```

dev-techshlok, 2020-05-11:
Hello @muneebahmed10, Thanks for the quick response.
I tried making directory but it gives below message.

```
root@iot-virtual-machine:~/esp/amazon-freertos# mkdir build
root@iot-virtual-machine:~/esp/amazon-freertos# cmake -DVENDOR=espressif -DBOARD=esp32_devkitc -DCOMPILER=xtensa-esp32 -S . -B build
CMake Error: The source directory "/root/esp/amazon-freertos/build" does not appear to contain CMakeLists.txt.
Specify --help for usage, or press the help button on the CMake GUI.
root@iot-virtual-machine:~/esp/amazon-freertos# 

```

dev-techshlok, 2020-05-11:
Hello @muneebahmed10 issue is resolved now. it was a problem of some cmake version and miss-configurations.

Thank you for your support.
I am closing this issue

closed 2020-05-11 by dev-techshlok