aws/amazon-freertos #1994: Unable to see any mqtt msg on iot dashboard.

dev-techshlok, 2020-05-11:
I have build the rtos. Flashed the mqtt default code in esp32. 
Also able to see monitor output. 
So far able to see connection made in Successful connections section in monitor of aws dashboard.
But not being able to receive any mqtt messages on topic #.

The output goes like this.

```

 (2255) WIFI: SYSTEM_EVENT_STA_CONNECTED
I (5105) event: sta ip: 192.168.1.144, mask: 255.255.255.0, gw: 192.168.1.1
I (5105) WIFI: SYSTEM_EVENT_STA_GOT_IP
GAP procedure initiated: stop advertising.
4 508 [iot_thread] [INFO ][DEMO][5080] Successfully initialized the demo. Network type for the demo: 1
GAP procedure initiated: stop advertising.
5 508 [iot_thread] [INFO ][MQTT][5080] MQTT library successfully initialized.
6 508 [iot_thread] [INFO ][DEMO][5080] MQTT demo client identifier is five (length 4).
E (5585) PKCS11: failed nvs get file size 4354 0
7 1486 [iot_thread] [INFO ][MQTT][14860] Establishing new MQTT connection.
8 1486 [iot_thread] [INFO ][MQTT][14860] Anonymous metrics (SDK language, SDK version) will be provided to AWS IoT. Recompile with AWS_IOT_MQTT_ENABLE_METRICS set to 0 to disable.
9 1486 [iot_thread] [INFO ][MQTT][14860] (MQTT connection 0x3ffb5cf8, CONNECT operation 0x3ffb5e14) Waiting for operation completion.
10 1527 [iot_thread] [INFO ][MQTT][15270] (MQTT connection 0x3ffb5cf8, CONNECT operation 0x3ffb5e14) Wait complete with result SUCCESS.
11 1527 [iot_thread] [INFO ][MQTT][15270] New MQTT connection 0x3ffb3e10 established.
12 1527 [iot_thread] [INFO ][MQTT][15270] (MQTT connection 0x3ffb5cf8) SUBSCRIBE operation scheduled.
13 1527 [iot_thread] [INFO ][MQTT][15270] (MQTT connection 0x3ffb5cf8, SUBSCRIBE operation 0x3ffba118) Waiting for operation completion.
14 1568 [iot_thread] [INFO ][MQTT][15680] (MQTT connection 0x3ffb5cf8, SUBSCRIBE operation 0x3ffba118) Wait complete with result SUCCESS.
15 1568 [iot_thread] [INFO ][DEMO][15680] All demo topic filter subscriptions accepted.
16 1568 [iot_thread] [INFO ][DEMO][15680] Publishing messages 0 to 1.
17 1568 [iot_thread] [INFO ][MQTT][15680] (MQTT connection 0x3ffb5cf8) MQTT PUBLISH operation queued.
18 1568 [iot_thread] [INFO ][MQTT][15680] (MQTT connection 0x3ffb5cf8) MQTT PUBLISH operation queued.
19 1568 [iot_thread] [INFO ][DEMO][15680] Waiting for 2 publishes to be received.
20 1609 [iot_thread] [INFO ][DEMO][16090] MQTT PUBLISH 0 successfully sent.
21 1609 [iot_thread] [INFO ][DEMO][16090] Incoming PUBLISH received:
Subscription topic filter: aws/things/five/shadow/update/topic/1
Publish topic name: aws/things/five/shadow/update/topic22 1609 [iot_thread] [INFO ][MQTT][16090] (MQTT connection 0x3ffb5cf8) MQTT PUBLISH operation queued.

```

muneebahmed10, 2020-05-11:
Hi @dev-techshlok 

You can enable debug logs for MQTT [here](https://github.com/aws/amazon-freertos/blob/master/vendors/espressif/boards/esp32/aws_demos/config_files/iot_config.h#L51). Are you able to successfully run the MQTT demo when you do not make any changes to it (i.e. leaving `IOT_DEMO_MQTT_TOPIC_PREFIX` as `"iotdemo"`)?

dev-techshlok, 2020-05-12:
Hello @muneebahmed10  Yes it compiles ok when i leave the ` IOT_DEMO_MQTT_TOPIC_PREFIX as "iotdemo"` 
But still not messages on test mqtt dashboard.
Is there any guide from amazon on running demos in details.

muneebahmed10, 2020-05-12:
Hi @dev-techshlok 

You can find instructions for building and running the FreeRTOS demo project for Espressif [here](). You can find more detailed information for the MQTT demo in the docs for the [AWS IoT Device SDK for Embedded C](https://docs.aws.amazon.com/freertos/latest/lib-ref/c-sdk/mqtt/mqtt_demo.html), with which the library in FreeRTOS shares an API.

Based on the output you have posted, it looks like the demo runs successfully and the client receives the publishes from the server. If you can't see the messages on the test client in the AWS IoT console, ensure that your permissions and policies are configured correctly as described [here](https://docs.aws.amazon.com/freertos/latest/userguide/freertos-account-and-permissions.html) and [here](https://docs.aws.amazon.com/freertos/latest/userguide/get-started-freertos-thing.html). 

Additionally, to aid in troubleshooting, you can try using another MQTT client, like this [sample](https://github.com/aws/aws-iot-device-sdk-python-v2/blob/master/samples/pubsub.py) in the [AWS IoT SDK for Python](https://github.com/aws/aws-iot-device-sdk-python-v2), on your host machine. If you use the same endpoint and credentials as your IoT Thing, you can determine if the issue lies in the configuration in the cloud, or on your device, based on whether or not the publish messages appear in the AWS IoT Console under the `#` topic.

dev-techshlok, 2020-05-13:
Hello @muneebahmed10 
Thank you for your support.

Found the solution from above process.

It was that i had my endpoint configured to different (south) region and RTOS configuration to east region.
Changed that to south all worked well.

I am closing the issue.

closed 2020-05-13 by dev-techshlok