aws/amazon-freertos #1999: Porting AWS FreeRTOS to STM32F746G_Discovery board missing files

ahmedwahdan, 2020-05-12:
**Describe the bug**
I'm following the porting guide to port AWS FreeRTOS to STM32F746G_Discovery board board, and I managed to list most of the missing including path, but currently I'm stuck with the following files:

#include "funky.h"
#include "stanky.h"
#include <Windows.h>
Do I need to install other libraries?

Thank you!

muneebahmed10, 2020-05-12:
Hi @ahmedwahdan 

I assume you are referring to the porting guide [here](https://docs.aws.amazon.com/freertos/latest/portingguide/porting-guide.html)? Could you list specifically which files include the missing headers?

ahmedwahdan, 2020-05-12:
freertos_plus/aws/ota/test/aws_test_ota_cbor.c
#include <windows.h>
#include <strsafe.h>
libraries/3rdparty/unity/expectdata/*.c
#include "funky.h"
#include "stanky.h"

muneebahmed10, 2020-05-12:
Hi @ahmedwahdan 

In what step are you trying to build those files? Are you trying to port OTA as one of the libraries as well?

For the Unity files, those seem to be part of the tests when building Unity, not tests that use the Unity framework. Are you trying to test Unity for your toolchain? If not, you should not need to build those files while porting.

The lwip file seems to be referring to the [check unit testing framework](https://libcheck.github.io/check/). However, you should not need to include the `test/` directory for lwip either: https://docs.aws.amazon.com/freertos/latest/portingguide/porting-lwip.html#porting-testing-lwip

ahmedwahdan, 2020-05-12:
I'm just trying to port to my board and use Ethernet till it's up and running correctly, then switch to PPPOS, but I'm kind of lost here.
I reached that point of the porting https://docs.aws.amazon.com/freertos/latest/portingguide/afr-porting-config.html.
Thanks for helping.

muneebahmed10, 2020-05-12:
Hi @ahmedwahdan 

You are likely including files that do not need to be built. Can you recheck [this step](https://docs.aws.amazon.com/freertos/latest/portingguide/porting-create-project.html) for IDE projects or [this step](https://docs.aws.amazon.com/freertos/latest/portingguide/porting-cmake-setup.html) for CMake to make sure you are not importing unnecessary test files?

ahmedwahdan, 2020-05-12:
Step 6 Import all of the files in freertos/libraries and its subdirectories into the aws_tests IDE project.
and then a note for not importing the unnecessary modules.
I don't know which is required and which is not. For sure I removed modules I don't need, but this is porting for testing so I kept the testing files.
I need FreeRTOS + AWS SDK (MQTT) + Lwip, could you clear what is really required to accomplish this task?
I'm playing around to figure out what to keep and what to remove.

htibosch, 2020-05-13:
Hi @ahmedwahdan I assume that you port will also use FreeRTOS+TCP?

You will find the latest sources and network interface [here](https://github.com/FreeRTOS/FreeRTOS/tree/master/FreeRTOS-Plus/Source/FreeRTOS-Plus-TCP).
[STM32Fxx](https://github.com/FreeRTOS/FreeRTOS/tree/master/FreeRTOS-Plus/Source/FreeRTOS-Plus-TCP/portable/NetworkInterface/STM32Fxx) is a combined driver for any STM23Fxx. You just have to #define the correct part.
The driver uses a generic PHY driver, found in Common/phyHandling.c
I have both a STM32F4 and a STM32F7 so I can do some co-testing.
Have you decided about the IDE to use? I'm using the Open STM32 tools, but I am open to any other tool .

ahmedwahdan, 2020-05-13:
Hi @htibosch 
I need to use lwip stack because I will use it's PPPOS interface later.
I'm currently working on CubeIDE.
Thanks for helping.

muneebahmed10, 2020-05-14:
Hi @ahmedwahdan 

Thank you for pointing out the issue in our documentation. We will work on updating it to clarify which modules are not required. In the meantime, you should not need to include test files that are from `libraries/3rdparty`, as those test the libraries and not FreeRTOS.

ahmedwahdan, 2020-05-16:
Hi @muneebahmed10 
I tried to do porting the other way, using generated project from cubemx with lwip integrated.
I added the mqtt libraries and all the dependencies required.
The code is highly dependent on the RTOS(i.e Task Pool) which make it very hard to debug.
Currently I'm stuck at this point:
IotMqtt_Connect > _IotMqtt_ScheduleOperation > IotTaskPool_ScheduleDeferred
The operation got blocked on TASKPOOL_ENTER_CRITICAL().
note:
I just started from the Run_Demo function, and initialized all the required functions.
[https://github.com/ahmedwahdan/STM32_Lwip_PPPOS](url)

ahmedwahdan, 2020-05-16:
I found the task pool is not initialized, the problem now is with the thread stack size.
The default value is 5000 which is an enormous number!, Could this size be minimized?

sarenameas, 2020-05-21:
Hello @ahmedwahdan 

Are you still having issues? Where are you seeing the default stack size as 5000? You can definitely minimize the default size of the task pool stack. It is different depending on the platform and how much space is available. The demo code has the default as:

```
#define IOT_THREAD_DEFAULT_STACK_SIZE    2048
```

ahmedwahdan, 2020-05-24:
Hello @sarenameas 
I'm set a reasonable size for the stack size,
Currently I'm facing a problem with TLS handshake
`00> PSH ACK F:/Wahdan/Repos/LwIp_Mqtt/LwIP_MQTT/Middlewares/Third_Party/mbedTLS/library/ssl_tls.c:4233: dumping 'input record from network' (5416 bytes)
00> F:/Wahdan/Repos/LwIp_Mqtt/LwIP_MQTT/Middlewares/Third_Party/mbedTLS/library/ssl_tls.c:4233: 0000:  16 03 03 15 23 02 00 00 51 03 03 ce c2 e9 a6 2f  ....#...Q....../
00> F:/Wahdan/Repos/LwIp_Mqtt/LwIP_MQTT/Middlewares/Third_Party/mbedTLS/library/ssl_tls.c:4233: 0010:  21 90 0b bb e3 0f cf e3 f0 fa ea 90 b5 20 13 dc  !............ ..
00> F:/Wahdan/Repos/LwIp_Mqtt/LwIP_MQTT/Middlewares/Third_Party/mbedTLS/library/ssl_tls.c:4233: 0020:  51 33 73 cc 21 2f ba 4f 7a 69 42 20 5f 8a 5a 32  Q3s.!/.OziB _.Z2
00> F:/Wahdan/Repos/LwIp_Mqtt/LwIP_MQTT/Middlewares/Third_Party/mbedTLS/library/ssl_tls.c:4233: 0030:  e2 7a bf 03 38 33 4b 39 0c 0e 0c 6b d1 25 c4 1d  .z..83K9...k.%..
00> F:/Wahdan/Repos/LwIp_Mqtt/LwIP_MQTT/Middlewares/Third_Party/mbedTLS/library/ssl_tls.c:4233: 0040:  dd 39 ad 22 6f de f5 83 94 8d f0 9d c0 2f 00 00  .9."o......../..
00> F:/Wahdan/Repos/LwIp_Mqtt/LwIP_MQTT/Middlewares/Third_Party/mbedTLS/library/ssl_tls.c:4233: 0050:  09 ff 01 00 01 00 00 17 00 00 0b 00 13 0f 00 13  ................
00> F:/Wahdan/Repos/LwIp_Mqtt/LwIP_MQTT/Middlewares/Third_Party/mbedTLS/library/ssl_tls.c:4233: 0060:  0c 00 05 a4 30 82 05 a0 30 82 04 88 a0 03 02 01  ....0...0.......
00> F:/Wahdan/Repos/LwIp_Mqtt/LwIP_MQTT/Middlewares/Third_Party/mbedTLS/library/ssl_tls.c:4233: 0070:  02 02 10 04 b1 f9 df c2 bd 23 88 18 a5 2c be 73  .........#...,.s
00> F:/Wahdan/Repos/LwIp_Mqtt/LwIP_MQTT/Middlewares/Third_Party/mbedTLS/library/ssl_tls.c:4233: 0080:  31 5d dd 30 0d 06 09 2a 86 48 86 f7 0d 01 01 0b  1].0...*.H......
00> F:/Wahdan/Repos/LwIp_Mqtt/LwIP_MQTT/Middlewares/Third_Party/mbedTLS/library/ssl_tls.c:4233: 0090:  05 00 30 46 31 0b 30 09 06 03 55 04 06 13 02 55  ..0F1.0...U....U
00> F:/Wahdan/Repos/LwIp_Mqtt/LwIP_MQTT/Middlewares/Third_Party/mbedTLS/library/ssl_tls.c:4233: 00a0:  53 31 0f 30 0d 06 03 55 04 0a 13 06 41 6d 61 7a  S1.0...U....Amaz
00> F:/Wahdan/Repos/LwIp_Mqtt/LwIP_MQTT/Middlewares/Third_Party/mbedTLS/library/ssl_tls.c:4233: 00b0:  6f 6e 31 15 30 13 06 03 55 04 0b 13 0c 53 65 72  on1.0...U....Ser
00> F:/Wahdan/Repos/LwIp_Mqtt/LwIP_MQTT/Middlewares/Third_Party/mbedTLS/library/ssl_tls.c:4233: 00c0:  76 65 72 20 43 41 20 31 42 31 0f 30 0d 06 03 55  ver CA 1B1.0...U
00> F:/Wahdan/Repos/LwIp_Mqtt/LwIP_MQTT/Middlewares/Third_Party/mbedTLS/library/ssl_tls.c:4233: 00d0:  04 03 13 06 41 6d 61 7a 6f 6e 30 1e 17 0d 32 30  ....Amazon0...20
00> F:/Wahdan/Repos/LwIp_Mqtt/LwIP_MQTT/Middlewares/Third_Party/mbedTLS/library/ssl_tls.c:4233: 00e0:  30 32 31 37 30 30 30 30 30 30 5a 17 0d 32 31 30  0217000000Z..210
00> F:/Wahdan/Repos/LwIp_Mqtt/LwIP_MQTT/Middlewares/Third_Party/mbedTLS/library/ssl_tls.c:4233: 00f0:  31 32 38 31 32 30 30 30 30 5a 30 28 31 26 30 24  128120000Z0(1&0$
00> F:/Wahdan/Repos/LwIp_Mqtt/LwIP_MQTT/Middlewares/Third_Party/mbedTLS/library/ssl_tls.c:4233: 0100:  06 03 55 04 03 0c 1d 2a 2e 69 6f 74 2e 75 73 2d  ..U....*.iot.us-
00> F:/Wahdan/Repos/LwIp_Mqtt/LwIP_MQTT/Middlewares/Third_Party/mbedTLS/library/ssl_tls.c:4233: 0110:  65 61 73 74 2d 32 2e 61 6d 61 7a 6f 6e 61 77 73  east-2.amazonaws
00> F:/Wahdan/Repos/LwIp_Mqtt/LwIP_MQTT/Middlewares/Third_Party/mbedTLS/library/ssl_tls.c:4233: 0120:  2e 63 6f 6d 30 82 01 22 30 0d 06 09 2a 86 48 86  .com0.."0...*.H.
00> F:/Wahdan/Repos/LwIp_Mqtt/LwIP_MQTT/Middlewares/Third_Party/mbedTLS/library/ssl_tls.c:4233: 0130:  f7 0d 01 01 01 05 00 03 82 01 0f 00 30 82 01 0a  ............0...
00> F:/Wahdan/Repos/LwIp_Mqtt/LwIP_MQTT/Middlewares/Third_Party/mbedTLS/library/ssl_tls.c:4233: 0140:  02 82 01 01 00 bc 11 e2 59 b8 30 2a 4d a3 1c fd  ........Y.0*M...
00> F:/Wahdan/Repos/LwIp_Mqtt/LwIP_MQTT/Middlewares/Third_Party/mbedTLS/library/ssl_tls.c:4233: 0150:  e4 e3 58 3e 5c 77 4f 93 ff 33 0b 56 ca 1c 29 9f  ..X>\wO..3.V..).
00> F:/Wahdan/Repos/LwIp_Mqtt/LwIP_MQTT/Middlewares/Third_Party/mbedTLS/library/ssl_tls.c:4233: 0160:  fa a9 78 6a b2 a8 1d c4 9e 79 ed 3f 5e f3 20 42  ..xj.....y.?^. B
00> F:/Wahdan/Repos/LwIp_Mqtt/LwIP_MQTT/Middlewares/Third_Party/mbedTLS/library/ssl_tls.c:4233: 0170:  26 9d 37 66 57 93 f8 92 e5 e4 b9 41 11 64 55 2e  &.7fW......A.dU.
00> F:/Wahdan/Repos/LwIp_Mqtt/LwIP_MQTT/Middlewares/Third_Party/mbedTLS/library/ssl_tls.c:4233: 0180:  12 b8 c3 58 b1 cf d9 75 50 ea 1f be a7 d1 12 b2  ...X...uP.......
00> F:/Wahdan/Repos/LwIp_Mqtt/LwIP_MQTT/Middlewares/Third_Party/mbedTLS/library/ssl_tls.c:4233: 0190:  cb e1 1b 7b c0 a2 21 19 1a 1d 1d 71 ef 5c 8e b3  ...{..!....q.\..
00> F:/Wahdan/Repos/LwIp_Mqtt/LwIP_MQTT/Middlewares/Third_Party/mbedTLS/library/ssl_tls.c:4233: 01a0:  61 be 88 01 6d 71 55 04 7a bf 0d 78 84 62 11 6c  a...mqU.z..x.b.l
00> F:/Wahdan/Repos/LwIp_Mqtt/LwIP_MQTT/Middlewares/Third_Party/mbedTLS/library/ssl_tls.c:4233: 01b0:  42 04 4d 96 d4 15 b5 83 5b 1c 07 55 9c 39 f3 7b  B.M.....[..U.9.{
00> F:/Wahdan/Repos/LwIp_Mqtt/LwIP_MQTT/Middlewares/Third_Party/mbedTLS/library/ssl_tls.c:4233: 01c0:  6a a6 ff 3f 57 9e 5c 5c 14 5f 50 aa b8 8b 14 16  j..?W.\\._P.....
00> F:/Wahdan/Repos/LwIp_Mqtt/LwIP_MQTT/Middlewares/Third_Party/mbedTLS/library/ssl_tls.c:4233: 01d0:  82 f7 73 9d c1 14 f0 da 5a 6c f8 0c 47 17 43 9b  ..s.....Zl..G.C.
[... cut ...]
00> F:/Wahdan/Repos/LwIp_Mqtt/LwIP_MQTT/Middlewares/Third_Party/mbedTLS/library/ssl_tls.c:3426: dumping 'output record sent to network' (7 bytes)
00> F:/Wahdan/Repos/LwIp_Mqtt/LwIP_MQTT/Middlewares/Third_Party/mbedTLS/library/ssl_tls.c:3426: 0000:  15 03 03 00 02 02 2d                             ......-
00> F:/Wahdan/Repos/LwIp_Mqtt/LwIP_MQTT/Middlewares/Third_Party/mbedTLS/library/ssl_tls.c:2755: => flush output
00> F:/Wahdan/Repos/LwIp_Mqtt/LwIP_MQTT/Middlewares/Third_Party/mbedTLS/library/ssl_tls.c:2774: message length: 7, out_left: 7`

Any thoughts?

ahmedwahdan, 2020-05-24:
[image]
[image]
[image]

sarenameas, 2020-05-26:
So in your log it says `x5089verify_cert() returned -9984 (-0x2700)`
That error is in x509.h: MBEDTLS_ERR_X509_CERT_VERIFY_FAILED
The server certificate could not be verified. 

If you are connecting to AWS IoT Core your port should be using the certificates in aws_default_root_certificate.h to verify the server. This is done automatically in libraries/freertos_plus/standard/tls/src/iot_tls.c, see TLS_Connect(). 

I just want to verify that you ported iot_tls.c?

lundinc2, 2020-07-23:
Hello @ahmedwahdan,

Are you still facing issues porting FreeRTOS, or can I close this issue?

Thanks

yanjos-dev, 2020-08-12:
@ahmedwahdan This topic has become inactive so I'm going to close the issue. Please reopen this if you need any further assistance.

closed 2020-08-12 by yanjos-dev