aws/amazon-freertos #2001: Receive MQTT packet with payload larger than 8KB

EtienneMdv, 2020-05-13:
Hello,

I've been trying to send MQTT packets larger than 8KB through AWS IOT from the broker to my device (ESP32 board). The content I'm trying to send to my device is a JPG image encoded in base64. My tests work well whenever the content I'm sending is smaller than 8KB. In that case I'm able to decode the base64 data and save it. However, the packet isn't received by my device when this size gets greater than 8KB. I would like to know if there is any restriction implied by AFR that would limit the size of a MQTT packet received. I'm using the IOT Serializer library in order to decode the data.
After reading the documentation I've seen that the payload maximum size is 128KB.  Moreover, I read that the maximum size for a JSON device shadow file is 8KB. 
Being relatively new on that topic I hope the information I'm giving will be enough to identify the issue.

Thank you for help!

muneebahmed10, 2020-05-13:
Hi @EtienneMdv 

It sounds like the buffer is not large enough to receive the packet. Can you check the value of the `MBEDTLS_SSL_MAX_CONTENT_LEN` macro and try increasing it?

EtienneMdv, 2020-05-14:
Hi @muneebahmed10 

Thank you very much for your answer! I managed to change the value of the `MBEDTLS_SSL_MAX_CONTENT_LEN` macro, that was initially equal to 8KB, to 16KB. I saw online that this size is the maximum that I can use. This would imply that I couldn't make use of the 128KB allowed by AWS IOT for a MQTT packet.

In any case I thank you warmly for your answer that helped me increase the size of the messages I receive. It was indeed a buffer issue.

mingyue86010, 2020-05-17:
Hi @EtienneMdv,

I'm wondering if your problem get resolved? After you increased the MBEDTLS_SSL_MAX_CONTENT_LEN are you able to send a MQTT packet upto 128KB? The MBEDTLS_SSL_MAX_CONTENT_LEN is the maxinum fragment length in bytes of the TLS record layer. If you pass a packet from upper layer larger than that, the TLS record lay will fragment it to mutiple chuncks. This process is transparent to the application layer. So it shouldn't revent you from sending a larger payload.

EtienneMdv, 2020-05-17:
Hello @mingyue86010!
I didn't have time to work on that topic lately but I will give you more information in the days to come. My last results were that I couldn't receive packets larger than 16KB. I will give you more details on the reason why the connection shuts down whenever the size gets too large and provide you screenshots that will surely be more helpful than my explanations. From what I recall it was due to a failure when my device was trying to send a PINGREQ packet.

sarenameas, 2020-05-26:
Hello @EtienneMdv,

Are you still having issues in your software after increasing the SSL content length? I am going to close this issue for now. If there are any other concerns feel free to open a new issue. Thanks.

closed 2020-05-26 by sarenameas

EtienneMdv, 2020-05-29:
Hi!
I'm very sorry for taking so much time to give updates. Unfortunately I still cannot receive data up to 128KB. I am wondering if the packet I am sending is indeed fragmented into chunks as @mingyue86010 shared. I am trying to send JSON data containing my JPG image. A key is dedicated to the image encoded in base64. The total size for the JSON data is around 74kB.
I get the following error :
`15 2044 [NetRecv] [ERROR][MQTT][20440] (MQTT connection 0x3ffd42d4) Failed to allocate buffer of length 74163 for incoming packet type 48.`
The size supposed to be allocated corresponds to the total size of the data I am trying to send.