aws/amazon-freertos #2002: How to implement WiFi module on FreeRTOS

ppedro1996, 2020-05-13:
Hi everyone,
I have a B-L475E-IOT01A2 board and I'm developing a project using STM32CubeIDE.
I would need to implement the Wi-Fi module within the project, using FreeRTOS (CMSIS-V2).
I found this Amazon project, which implements the Wi-Fi module using FreeRTOS (CMSIS-V1).
I imported the following files into my project:
- es_wifi_conf.h
- es_wifi_io.c
- es_wifi_io.h
- es_wifi.c
- es_wifi.h
- iot_network_types.h
- iot_wifi.c
- iot_wifi.h
- aws_wifi_config.h

I had to adapt wait and post functions of semaphores and mutexes to make them compatible with the new version of FreeRTOS (CMSIS_V2).
The problem is that, going to run the project, the module is not initialized correctly and the WIFI_On() call executes until it stops at a certain point.
I noticed that in the es_wifi_io.h file a WIFI_RESET_MODULE() function is defined which has two HAL_Delay(). The first is 10 ms, but the second is 500 ms.
If I modify the second Delay and set it, for example, to 20 ms, the point at which the execution of the WIFI_On() stops is different.
I tried to configure FreeRTOS as indicated by the configuration files provided by the example of the Amazon project, but I still cannot solve.

Finally, I tried to create a project using FreeRTOS (CMSIS-V1), then I imported the same files listed above.
I didn't have to adapt any functions for semaphores and mutex, as it is the same version used for the Amazon project.
And I had the same problem.

I tried to make the call to WIFI_On() both from the main and from the vApplicationDaemonTaskStartupHook() function, but the result is the same.
Can you help me?

Thanks to anyone who wants to help me

aggarg, 2020-05-14:
Do you mind sharing your project so that we can better help you? Also, do you only need WiFi functionality or anything more too?

Thanks.

cerbero-dj, 2020-05-14:
Hi,
thank you for replying.
If you want and if you have time, I can send you the entire project on google drive / by email / or in the best way you want.
To carry out my project I would also need to use the microphone, because the project I want to carry out is a translator.

Meanwhile, thank you and looking forward to your reply, I wish you a good day

aggarg, 2020-05-14:
You can share using google drive. Another option is to use the Contact link ***Contact us for business related inquiries.*** to email the project. Please mention in the email that it is for aggarg@. 

My intention of the question about other libraries was to determine what all libraries from `FreeRTOS` you will need. 

Thanks.

cerbero-dj, 2020-05-14:
Scusami, ma non ho capito bene come posso inviarti tutto evitando google drive...

cerbero-dj, 2020-05-14:
Sorry, but I didn't understand how I can send you the project avoiding google drive...

aggarg, 2020-05-14:
Please feel free to use Google Drive.

Thanks.

cerbero-dj, 2020-05-14:
Below I am attaching the project created with STM32CubeIDE.
[https://drive.google.com/file/d/19FIKZWf_Wv1FvXugC-KUgCJOayrGubI1/view?usp=sharing](url)

I thank you for your time and patience.

aggarg, 2020-05-17:
Here is the project which should help you get started: [WiFi.zip](https://github.com/aws/amazon-freertos/files/4641253/WiFi.zip).

I generated a fresh project using STM32Cube IDE and I needed to make the following major changes:

* Port the es_wifi driver to use CMSIS_V2 APIs.
* Delete the default task as its stack was too small and was causing memory corruption.
* The default generated implementation of `_sbrk` is buggy and causes a Hard Fault when es_wifi driver calls strtok. The cause of the Fault is that the default `_sbrk` implementation reads "SP" to check if heap has grown into stack. This does not work in case of FreeRTOS because it results in reading "PSP" while the intention of the code is to read "MSP". I updated the `_sbrk` implementation to read "MSP" explicitly.
* Set `configUSE_NEWLIB_REENTRANT` to 1 in FreeRTOSConfig.h and provide implementation of `__malloc_lock` and `__malloc_unlock` - This is needed to ensure that malloc calls from within the implementation of strtok are thread safe.

In addition to the above, I also pulled in our secure sockets layer - the es_wifi driver APIs are not thread safe but our secure sockets APIs are. I have written a small echo demo to show how to connect, send and receive data from an echo server. To run the demo, you need to configure the following:

* `clientcredentialWIFI_SSID` in `Config/aws_clientcredential.h`.
* `clientcredentialWIFI_PASSWORD` in `Config/aws_clientcredential.h`.
* `configECHO_SERVER_ADDR0` in `Core/Inc/FreeRTOSConfig.h`.
* `configECHO_SERVER_ADDR1` in `Core/Inc/FreeRTOSConfig.h`.
* `configECHO_SERVER_ADDR2` in `Core/Inc/FreeRTOSConfig.h`.
* `configECHO_SERVER_ADDR3` in `Core/Inc/FreeRTOSConfig.h`.
* `configTCP_ECHO_CLIENT_PORT` in `Core/Inc/FreeRTOSConfig.h`.

Hope that helps.

Thanks.

cerbero-dj, 2020-05-18:
Hi,
thank you very much for the time you have dedicated to me and for the exhaustive description of the various problems.
I have tried and everything works as you described!
I thank you again and wish you a good day

closed 2020-05-18 by cerbero-dj

aggarg, 2020-05-18:
Glad to know that it worked for you. One additional point - you can avoid patching the `_sbrk` implementation by redirecting `malloc` and `free` calls to FreeRTOS `pvPortMalloc` and `vPortFree` using the linker `wrap` feature. This project shows how to do that: [WiFi-2.zip](https://github.com/aws/amazon-freertos/files/4645767/WiFi-2.zip)

Thanks.

cerbero-dj, 2020-05-18:
I also thank you for this second version.
I will most likely consider the latter because the latter variation is very interesting.
Good evening and thank you for the time you have dedicated to me.