aws/amazon-freertos #2018: Integrate Voice control SDK on ESP32

elieDaan, 2020-05-18:
**Is your feature request related to a problem? Please describe.**  
Is it possible to integrate a voice control system (Alexa or another ) on ESP32 running with AFR ? 
 
**Describe the solution you would like.**  
Using the amazon freertos with ESP32 like the [Espressif's Voice Assistant SDK](https://github.com/espressif/esp-va-sdk). 

Thank you!

aggarw13, 2020-06-25:
Hi @elieDaan, currently, we do not have an example of using a voice assistant SDK with Espressif in our repository. We will add it to our list of feature requests. 
There is an existing Amazon qualified solution for Alexa voice control on the Espressif ESP32-Vaquita-DSPG board using Espressif’s AVS for AWS IoT SDK. You can find more information about the board and ESP Voice Assistant SDK for Alexa [here]( https://www.espressif.com/en/solutions/audio-solutions/esp-avs-for-aws-iot). This solution uses the [freertos component](https://github.com/espressif/esp-idf/tree/master/components/freertos) distributed through the ESP-IDF SDK.

yuhui-zheng, 2020-06-25:
Hi @elieDaan, I'm closing this thread for now. Should you run into any difficulty related to our SDK or integrated samples, please don't hesitate to open another issue.

closed 2020-06-25 by yuhui-zheng