aws/amazon-freertos #2024: AWS OTA Agent config flag to allow firmware version downgrades

danahern, 2020-05-19:
We are building a number of completely closed devices which our main way of testing outside of the hardware team is through OTA.  As part of our QA and as a fail safe for our consumer release process we would like the ability to downgrade the firmware.  

Right now in the aws_iot_ota_agent.c there is a check:
```
if( C->ulUpdaterVersion > xAppFirmwareVersion.u.ulVersion32 )
                        {
                            /* The running firmware is older than the firmware that performed the update so reject the job. */
                            OTA_LOG_L1( "[%s] Rejecting image because version is older than previous.\r\n", OTA_METHOD_NAME );
                            ( void ) prvSetImageStateWithReason( eOTA_ImageState_Rejected, kOTA_Err_DowngradeNotAllowed );
                        }
                        else /* Version reported is the same as the running version. */
                        {
                            if( ( xOTA_Agent.pcClientTokenFromJob == NULL ) ||
                                ( strtoul( ( const char * ) xOTA_Agent.pcClientTokenFromJob, NULL, 0 ) == 0U ) ) /*lint !e9007 We don't provide a modifiable variable to strtoul. */
                            {
                                /* The version is the same so either we're not actually the new firmware or
                                 * someone messed up and sent firmware with the same version. In either case,
                                 * this is a failure of the OTA update so reject the job. */
                                OTA_LOG_L1( "[%s] Failing job. We rebooted and the version is still the same.\r\n", OTA_METHOD_NAME );
                                ( void ) prvSetImageStateWithReason( eOTA_ImageState_Rejected, kOTA_Err_SameFirmwareVersion );
                            }
                            else
                            {
                                OTA_LOG_L1( "[%s] Ignoring job. Device must be rebooted first.\r\n", OTA_METHOD_NAME );
                            }
                        }
```

What I would like to suggest is adding a flag to the aws_ota_agent_config.h flags along the lines of otaconfigALLOW_DOWNGRADE which would turn off this check.  Also allowing the same version would be nice too.  

The reason we can't just go in and change the code directly is because we are using the AFR as an external library.

If there is interest I could probably put together the change and offer a PR.

Thanks

pvyawaha, 2020-05-20:
Hello,

We are reviewing this change request and I will update the status here when the change is merged.

pvyawaha, 2020-06-02:
The Pull Request is merged to master.

Closing this issue , please reopen if you have more questions regarding this.

closed 2020-06-02 by pvyawaha