aws/amazon-freertos #2026: Undefined symbol ulTCPWindowTxSack (referred from freertos_tcp_ip.o)

cyliangtw, 2020-05-21:
**Describe the bug**

in FreeRTOS_TCP_IP.c, `prvReadSackOption()` will call `ulTCPWindowTxSack()`. 
However, if platform not enable "ipconfigUSE_TCP_WIN", it will encounter "Undefined symbol ulTCPWindowTxSack" in linking.

Snippet of `prvReadSackOption()` as below:
```
static void prvReadSackOption( const uint8_t * const pucPtr,
							   size_t uxIndex,
							   FreeRTOS_Socket_t * const pxSocket )
{
uint32_t ulFirst = ulChar2u32( &( pucPtr[ uxIndex ] ) );
uint32_t ulLast  = ulChar2u32( &( pucPtr[ uxIndex + 4U ] ) );
uint32_t ulCount = ulTCPWindowTxSack( &( pxSocket->u.xTCP.xTCPWindow ), ulFirst, ulLast );;
```

sarenameas, 2020-05-22:
Hello @cyliangtw ,

Thanks for bringing this up. We are looking into fixing this.

htibosch, 2020-05-22:
Thanks for posting this. We were aware of it, but it was a busy week, we worked on a big release.
Right now I am preparing a PR to repair this.

Background: the TCP SACK (selective ACK) option is only used when both `ipconfigUSE_TCP` and `ipconfigUSE_TCP_WIN` are enabled.

htibosch, 2020-05-26:
As I submitted a PR #2029 to solve this problem, I think that this issue can be closed?
Hein

closed 2020-05-26 by sarenameas