aws/amazon-freertos #2037: AWS FreeRTOS Windows Simulator demo issue about NETWORK ERROR

shengyi123, 2020-05-26:
Hi,
The following is my log:
`9 6062 [iot_thread] [INFO ][DEMO][6062] ---------STARTING DEMO---------

10 6063 [iot_thread] [INFO ][INIT][6063] SDK successfully initialized.
11 6063 [iot_thread] [INFO ][DEMO][6063] Successfully initialized the demo. Network type for the demo: 4
12 6063 [iot_thread] [INFO ][MQTT][6063] MQTT library successfully initialized.
13 6063 [iot_thread] [INFO ][DEMO][6063] MQTT demo client identifier is IotThing (length 8).
14 6618 [iot_thread] DNS[0x91D3]: The answer to 'a2kq59c1ukpv1s-ats.iot.us-east-2.amazonaws.com' (3.22.227.144) will be stored
15 11620 [iot_thread] [ERROR][NET][11620] Failed to establish new connection. Socket status: -116.
16 11620 [iot_thread] [ERROR][MQTT][11620] Failed to establish new MQTT connection, error NETWORK ERROR.
17 11620 [iot_thread] [ERROR][DEMO][11620] MQTT CONNECT returned error NETWORK ERROR.
18 11620 [iot_thread] [INFO ][MQTT][11620] MQTT library cleanup done.
19 11620 [iot_thread] [INFO ][DEMO][11620] memory_metrics::freertos_heap::before::bytes::2085856
20 11620 [iot_thread] [INFO ][DEMO][11620] memory_metrics::freertos_heap::after::bytes::2057664
21 11620 [iot_thread] [INFO ][DEMO][11620] memory_metrics::demo_task_stack::before::bytes::1908
22 11620 [iot_thread] [INFO ][DEMO][11620] memory_metrics::demo_task_stack::after::bytes::1908
23 11620 [iot_thread] [ERROR][DEMO][11620] Error running demo.
24 11623 [iot_thread] [INFO ][INIT][11623] SDK cleanup done.
25 11623 [iot_thread] [INFO ][DEMO][11623] -------DEMO FINISHED-------`

**System information**
-  [FreeRTOS Windows Simulator](url) (Connect_to_AWS_IoT_-_Windows)
- Visual Studio 2019
- Windows 10
- npcap 0.9991

I use fast connect in aws but get networ error.
I have tried many times.
I guest it's related to  DNS lookup
[image]
 
Can anybody has any idea about this?
Thank you very much!

sarenameas, 2020-05-26:
Hello @shengyi123 

The DNS lookup passed as evident by this line:
```
14 6618 [iot_thread] DNS[0x91D3]: The answer to 'abcdefg-ats.iot.us-east-2.amazonaws.com' (3.22.227.144) will be stored
```

A -116 error is a timeout on trying to connect. A timeout can occur if AWS IoT refuses your connection and drops the TLS handshake altogether.

- Please make sure that you have the correct device credentials under: https://github.com/aws/amazon-freertos/blob/master/demos/include/aws_clientcredential_keys.h

- Check that your policy associated with your credentials, allows MQTT connections, publishes, subscribes, and receives. See the policy in step 5 here: https://docs.aws.amazon.com/freertos/latest/userguide/get-started-freertos-thing.html

- Make sure your certificates are also in the activated state from the IoT Core dashboard.

shengyi123, 2020-05-27:
Hi @sarenameas ,
First,thank you so much for your reply,then:
1. I have checked aws_clientcredential_key.h again,and I sure I have the correct 
    credentials device.
[image]
[image]

2. I sure my policy associated with my credentials
[image]

[image]

[image]
3. I sure my certificates are also in the activated state

and then bacause I use [quick connect](https://us-east-2.console.aws.amazon.com/iot/home?region=us-east-2#/freertos/quickconnect/Connect_to_AWS_IoT_-_Windows-Simulator?type=predefined),I don't think it should be a problem

shengyi123, 2020-05-28:
I think this problem is related to the firewall

closed 2020-05-28 by shengyi123

aogrcs, 2021-03-06:
I follow the quick connect, there is the error:
0 524 [Tmr Svc] WiFi module initialized.
1 544 [Tmr Svc] [ERROR] [PKCS11] [core_pkcs11_mbedtls.c:1374] 2 550 [Tmr Svc] Failed destroying object. Received a different handle from the PAL when writing to the same label.3 560 [Tmr Svc]
4 567 [Tmr Svc] Write certificate...
5 583 [Tmr Svc] [ERROR] [PKCS11] [core_pkcs11_mbedtls.c:2122] 6 589 [Tmr Svc] Failed to create certificate. Could not save the certificate to the PKCS #11 PAL.7 597 [Tmr Svc]
8 599 [Tmr Svc] ERROR: Failed to provision device certificate. 49
9 3895 [Tmr Svc] WiFi connected to AP TP-LINK_CF68F8.
10 3900 [Tmr Svc] IP Address acquired 192.168.0.103.
11 3905 [Tmr Svc] WiFi firmware version is: C3.5.2.3.BETA9
12 3911 [Tmr Svc] [WARN] WiFi firmware needs to be updated.
13 3917 [iot_thread] [INFO ][DEMO][3917] ---------STARTING DEMO---------

14 3925 [iot_thread] [INFO ][INIT][3925] SDK successfully initialized.
15 7787 [iot_thread] [INFO ][DEMO][7787] Successfully initialized the demo. Network type for the demo: 1
16 7797 [iot_thread] [INFO] [MQTT_MutualAuth_Demo] [mqtt_demo_mutual_auth.c:745] 17 7804 [iot_thread] Creating a TLS connection to a2dmqd9fmooofy-ats.iot.us-east-2.amazonaws.com:8883.18 7813 [iot_thread]
19 7941 [iot_thread] [ERROR] [Transport_Secure_Sockets] [transport_secure_sockets.c:323] 20 7949 [iot_thread] Failed to establish new connection. secureSocketStatus=-1.21 7956 [iot_thread]
22 7969 [iot_thread] [WARN] [MQTT_MutualAuth_Demo] [mqtt_demo_mutual_auth.c:753] 23 7976 [iot_thread] Connection to the broker failed. Attempting connection retry after backoff delay.24 7985 [iot_thread]
25 8219 [iot_thread] [INFO] [MQTT_MutualAuth_Demo] [mqtt_demo_mutual_auth.c:689] 26 8226 [iot_thread] Retry attempt 2 out of maximum retry attempts 5.27 8232 [iot_thread]
28 8234 [iot_thread] [INFO] [MQTT_MutualAuth_Demo] [mqtt_demo_mutual_auth.c:745] 29 8241 [iot_thread] Creating a TLS connection to a2dmqd9fmooofy-ats.iot.us-east-2.amazonaws.com:8883.30 8250 [iot_thread]
31 8318 [iot_thread] [ERROR] [Transport_Secure_Sockets] [transport_secure_sockets.c:323] 32 8325 [iot_thread] Failed to establish new connection. secureSocketStatus=-1.33 8332 [iot_thread]
34 8345 [iot_thread] [WARN] [MQTT_MutualAuth_Demo] [mqtt_demo_mutual_auth.c:753] 35 8352 [iot_thread] Connection to the broker failed. Attempting connection retry after backoff delay.36 8361 [iot_thread]
37 8643 [iot_thread] [INFO] [MQTT_MutualAuth_Demo] [mqtt_demo_mutual_auth.c:689] 38 8650 [iot_thread] Retry attempt 3 out of maximum retry attempts 5.39 8656 [iot_thread]
40 8658 [iot_thread] [INFO] [MQTT_MutualAuth_Demo] [mqtt_demo_mutual_auth.c:745] 41 8665 [iot_thread] Creating a TLS connection to a2dmqd9fmooofy-ats.iot.us-east-2.amazonaws.com:8883.42 8674 [iot_thread]
43 8744 [iot_thread] [ERROR] [Transport_Secure_Sockets] [transport_secure_sockets.c:323] 44 8751 [iot_thread] Failed to establish new connection. secureSocketStatus=-1.45 8758 [iot_thread]
46 8771 [iot_thread] [WARN] [MQTT_MutualAuth_Demo] [mqtt_demo_mutual_auth.c:753] 47 8778 [iot_thread] Connection to the broker failed. Attempting connection retry after backoff delay.48 8787 [iot_thread]
49 8857 [iot_thread] [INFO] [MQTT_MutualAuth_Demo] [mqtt_demo_mutual_auth.c:689] 50 8864 [iot_thread] Retry attempt 4 out of maximum retry attempts 5.51 8870 [iot_thread]
52 8872 [iot_thread] [INFO] [MQTT_MutualAuth_Demo] [mqtt_demo_mutual_auth.c:745] 53 8879 [iot_thread] Creating a TLS connection to a2dmqd9fmooofy-ats.iot.us-east-2.amazonaws.com:8883.54 8888 [iot_thread]
55 8956 [iot_thread] [ERROR] [Transport_Secure_Sockets] [transport_secure_sockets.c:323] 56 8963 [iot_thread] Failed to establish new connection. secureSocketStatus=-1.57 8971 [iot_thread]
58 8983 [iot_thread] [WARN] [MQTT_MutualAuth_Demo] [mqtt_demo_mutual_auth.c:753] 59 8990 [iot_thread] Connection to the broker failed. Attempting connection retry after backoff delay.60 8999 [iot_thread]