aws/amazon-freertos #2039: Want to add AWS rtos features like MQTT and OTA in my ESP-IDF example.

dev-techshlok, 2020-05-27:
Hello,

I am trying to add an RTOS example in a code I made in ESP-IDF using ESP-ADF.  My only question is how to do it.

There are a few methods that I have tried.
Method 1. Using these steps written here.
https://docs.aws.amazon.com/freertos/latest/userguide/getting_started_espressif.html
Under this section 
**Using FreeRTOS in your own CMake project for ESP32**
```
If you want to consume FreeRTOS in your own CMake project, you can set it up as a subdirectory and build it together with your application. First, get a copy of FreeRTOS either from GitHub, or from the FreeRTOS console. If you're using git, you can also set it up as a git submodule with the following command so it's easier to update in the future.

git submodule add -b release https://github.com/aws/amazon-freertos.git freertos
If a newer version is released, you can update your local copy with these commands.
```
Method 2 - 
Example given here 
https://github.com/tgsong/amazon-freertos-examples
But in this exmaple not sure how to add my code. 

There are so many doubts from so many people .
can you please let me know some steps here, with some examples, please.

alfred2g, 2020-05-28:
Hi @dev-techshlok ,

Thanks for your interest in FreeRTOS, I will be looking forward in providing you with more information.

Alfred

dev-techshlok, 2020-05-28:
Hello @alfred2g .
Thank you for your support.
Please let me know in comments.

dev-techshlok, 2020-05-28:
@alfred2g 
If i be more specific about the problem statement. i added this example in a folder compiled it using c make with these commands.
It compiled ok. using method 2.
```
cmake -S . -B build -DCMAKE_TOOLCHAIN_FILE=freertos/tools/cmake/toolchains/xtensa-esp32.cmake -GNinja
cmake --build build
```
I also observed that there is a component called `foo`. I added one more component `driver `. tried led blinking it worked ok.

Now the main problem is that I want to include all the components of ESp-IDF and ESP-ADF.
So the code i have build combining both compiles.

but when in example code from method 2, i included these components 
```
#include "esp_log.h"
#include "audio_element.h"
#include "audio_pipeline.h"
#include "audio_event_iface.h"
#include "audio_mem.h"
#include "audio_common.h"
#include "i2s_stream.h"
#include "mp3_decoder.h"
```

So it gave this error.

`./src/main.c:83:27: fatal error: audio_element.h: No such file or directory
`
which is fine because I did not add it in the CMake file like component  `foo`. 
But is there any way i can add all the components from ESP-IDF and ESP-ADF at once and be done with it.

So any header file i added in ESP-IDF and ESP-ADF project gets compiled with aws-rtos too.

alfred2g, 2020-05-28:
As far as I know you will have to add them manually, but I can double check for you in the morning, whether there is a better/easier method to add all the files in ESP-IDF and ESP-ADF in one shot.

Alfred

dev-techshlok, 2020-05-28:
Thank you @alfred2g .

Please let me know in the morning. I ll be waiting for you response.

Thank You

alfred2g, 2020-05-28:
What do you mean by ESP-ADF ?

alfred2g, 2020-05-28:
Hi,

For our example esp-idf should be included, for esp-adf how to do it is explained in their documentation here
https://docs.espressif.com/projects/esp-adf/en/latest/get-started/index.html#get-started-get-esp-adf
and here is their cmakelists
https://github.com/espressif/esp-adf/blob/0f2fa97c64ad9c39486a594d5894175c44226c8b/examples/get-started/play_mp3/CMakeLists.txt

so that example cmake file just include the esp-adf top level cmake file, and this line is what you need,

https://github.com/espressif/esp-adf/blob/0f2fa97c64ad9c39486a594d5894175c44226c8b/CMakeLists.txt#L9
which corresponds to this line in our example repo,
https://github.com/tgsong/amazon-freertos-examples/blob/b72b620d16243c8cdeba99e1fda80ea6fc300084/CMakeLists.txt#L12

Hope this helps,

Alfred

dev-techshlok, 2020-05-29:
Hello @alfred2g any luck finding solution.

alfred2g, 2020-05-29:
Hi @dev-techshlok,

I have sent you a comment, i forgot to tag your username.

Regards,
Alfred

dev-techshlok, 2020-05-29:
Hello @alfred2g 
Thank for the reply.

I am following the steps given by you.

1. I replaced my CMake file of `aws-rtos-example` with this.
I added these lines from steps given by you

_**set(EXTRA_COMPONENT_DIRS $ENV{ADF_PATH}/components)
include($ENV{ADF_PATH}/CMakeLists.txt)**_
CMakefile.txt
```
cmake_minimum_required(VERSION 3.13)
project(freertos_examples)
add_executable(afr_demo src/main.c)

include($ENV{ADF_PATH}/CMakeLists.txt)

# Tell IDF build to link against this target.
set(IDF_PROJECT_EXECUTABLE afr_demo)
# Add some extra components. IDF_EXTRA_COMPONENT_DIRS is an variable used by ESP-IDF
# to collect extra components.

set(EXTRA_COMPONENT_DIRS $ENV{ADF_PATH}/components)

get_filename_component(
    EXTRA_COMPONENT_DIRS
    "components/foo" ABSOLUTE
)
list(APPEND IDF_EXTRA_COMPONENT_DIRS ${EXTRA_COMPONENT_DIRS})
# As of now there's no offical way to redefine config files outside of FreeRTOS source tree.
# This is a temporary approach to inject an include path so that this takes precedence over the
# config file directory inside FreeRTOS.
include_directories(BEFORE freertos-configs)
# Add freertos as an subdirectory. AFR_BOARD tells which board to target.
set(AFR_BOARD espressif.esp32_devkitc CACHE INTERNAL "")
add_subdirectory(freertos)
# Link against the mqtt demo so that we can use it. Dependencies of this demo are transitively
# linked.
target_link_libraries(afr_demo PRIVATE AFR::demo_mqtt)

```

This compiles successfully using this command.
`cmake -S . -B build -DCMAKE_TOOLCHAIN_FILE=freertos/tools/cmake/toolchains/xtensa-esp32.cmake -GNinja
`

2. Now I have not modified anything in `src/main.c` file yet just trying to compile it. Using this command.
`cmake --build build`

Then I get the following error.

```
[163/1122] Building C object freertos/esp-idf/newlib/CMakeFiles/idf_component_newlib.dir/locks.c.obj
FAILED: freertos/esp-idf/newlib/CMakeFiles/idf_component_newlib.dir/locks.c.obj
C:\PROGRA~1\ESPRES~1\ESP-ID~1\tools\bin\XT8994~1.EXE -DADF_VER=\"b72b620-dirty\" -DESP_PLATFORM -DGCC_NOT_5_2_0=0 -DHAVE_CONFIG_H -DIDF_VER=\"v3.3.2-107-g722043f73\" -IC:/Users/Hitesh/esp/esp
ples/get-started/amazon-freertos-examples/freertos-configs -IC:/Users/Hitesh/esp/esp-adf/esp-idf/components/newlib/platform_include -IC:/Users/Hitesh/esp/esp-adf/esp-idf/components/newlib/inc
nfig -IC:/Users/Hitesh/esp/esp-adf/esp-idf/components/vfs/include -IC:/Users/Hitesh/esp/esp-adf/esp-idf/components/esp32/include -IC:/Users/Hitesh/esp/esp-adf/esp-idf/components/driver/includ
ers/Hitesh/esp/esp-adf/esp-idf/components/esp_ringbuf/include -IC:/Users/Hitesh/esp/esp-idf/examples/get-started/amazon-freertos-examples/freertos/vendors/espressif/boards/esp32/components/fr
pip/tcpip_adapter/include -IC:/Users/Hitesh/esp/esp-idf/examples/get-started/amazon-freertos-examples/freertos/libraries/freertos_plus/standard/freertos_plus_tcp/include -IC:/Users/Hitesh/esp
examples/get-started/amazon-freertos-examples/freertos/libraries/freertos_plus/standard/freertos_plus_tcp/source/portable/Compiler/GCC -IC:/Users/Hitesh/esp/esp-adf/esp-idf/components/esp_eve
e -IC:/Users/Hitesh/esp/esp-adf/esp-idf/components/log/include -IC:/Users/Hitesh/esp/esp-adf/esp-idf/components/efuse/include -IC:/Users/Hitesh/esp/esp-adf/esp-idf/components/efuse/esp32/incl
Users/Hitesh/esp/esp-idf/examples/get-started/amazon-freertos-examples/freertos/vendors/espressif/boards/esp32/components/freertos/include -IC:/Users/Hitesh/esp/esp-idf/examples/get-started/a
ertos-examples/freertos/freertos_kernel/include -IC:/Users/Hitesh/esp/esp-idf/examples/get-started/amazon-freertos-examples/freertos/freertos_kernel/portable/ThirdParty/GCC/Xtensa_ESP32/inclu
sers/Hitesh/esp/esp-idf/examples/get-started/amazon-freertos-examples/freertos/vendors/espressif/boards/esp32/aws_demos/config_files -IC:/Users/Hitesh/esp/esp-adf/esp-idf/components/heap/incl
Users/Hitesh/esp/esp-adf/esp-idf/components/soc/esp32/include -IC:/Users/Hitesh/esp/esp-adf/esp-idf/components/soc/include -mlongcalls   -mlongcalls -Og -ffunction-sections -fdata-sections -f
latile-bitfields -nostdlib -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-pa
Wno-sign-compare -ggdb -std=gnu99 -Wno-old-style-declaration -MD -MT freertos/esp-idf/newlib/CMakeFiles/idf_component_newlib.dir/locks.c.obj -MF freertos\esp-idf\newlib\CMakeFiles\idf_compone
.dir\locks.c.obj.d -o freertos/esp-idf/newlib/CMakeFiles/idf_component_newlib.dir/locks.c.obj   -c C:/Users/Hitesh/esp/esp-adf/esp-idf/components/newlib/locks.c
C:/Users/Hitesh/esp/esp-adf/esp-idf/components/newlib/locks.c: In function 'lock_acquire_generic':
C:/Users/Hitesh/esp/esp-adf/esp-idf/components/newlib/locks.c:140:10: error: implicit declaration of function 'xPortCanYield' [-Werror=implicit-function-declaration]
     if (!xPortCanYield()) {
          ^
cc1.exe: some warnings being treated as errors
[168/1122] Building C object freertos/libraries/3rdparty/CMakeFiles/afr_3rdparty_mbedtls.dir/mbedtls/library/ssl_tls.c.obj
ninja: build stopped: subcommand failed.

```

Now I am not sure why this error came.

Also thanks for the information that ESp-IDF is already included it compiles ok when I take anything from ESP-IDF example.

please let me know your thoughts on this.

dev-techshlok, 2020-06-03:
Hello @alfred2g 

Sorry, I took a little while but this time I am sure that this is the main problem we need to solve.
Because whenever I fix one library dependency it asks me to add for another. Also, my main.c file only contains the code from freertos-aws example so nothing is there which I have included in the main file. 

The libraries are linked with the other libraries which are not automatically included. 

So please let me know if we can solve this problem.

alfred2g, 2020-06-04:
Hi @dev-techshlok,

Can you share the errors you are getting?
It is hard for me to guess without seeing from where the failures are generated.

Thanks,
Alfred

dev-techshlok, 2020-06-22:
Hello @alfred2g 
I am sorry for the late reply as I was trying all the methods I can get my hands on.

hello @pvyawaha ,  I hope you are here to help me now.

So, I have reached some basic points in the process of finding a way to implement the ADF components to work with AWS-RTOS. 

When in my CMakefile.txt file I add this line
`include($ENV{ADF_PATH}/CMakeLists.txt)
`
which is same Cmake file in this reply. 
https://github.com/aws/amazon-freertos/issues/2039#issuecomment-635919099

it gives the same error in the answer.
**Then I get the following error.**.

Now when I remove this line it gives the error in this replay https://github.com/aws/amazon-freertos/issues/2039#issuecomment-635144043

So now I am stuck at this point where I was before and finding this ```
error: implicit declaration of function 'xPortCanYield' [-Werror=implicit-function-declaration]
     if (!xPortCanYield()) {
```

error is not an option for me.

Please can u suggest a way to implement all the ESP-ADF components at once and so I can include any file from the ESP-ADF example and get them run?

alfred2g, 2020-06-23:
Can you post the smallest possible code that would show the failure somehow ?
It is a bit tricky figuring out whats going on in the dark

Alfred

dev-techshlok, 2020-06-23:
Hello @alfred2g 

This is my CMake file.

[CMakeLists.txt](https://github.com/aws/amazon-freertos/files/4818332/CMakeLists.txt)

This is my source file which is in src/main.c 

[code.txt](https://github.com/aws/amazon-freertos/files/4818337/code.txt)

aggarw13, 2020-06-24:
Hi @dev-techshlok, thank you for providing the CMake and code source files.

Looking at the build logs you shared in https://github.com/aws/amazon-freertos/issues/2039#issuecomment-635919099, the build error about `xPortCanYield` for the `esp-idf/components/newlib/locks.c` file can exist only if the file does not include the appropriate header. However, the [lock.c file on the official ESP-IDF repository](https://github.com/espressif/esp-idf/blob/722043f734fa556d66d57473ac266fb1d0ec5ad2/components/newlib/locks.c#L24) includes `freertos/portable.h` file which contains the definition of [`xPortCanYield`](https://github.com/espressif/esp-idf/blob/722043f734fa556d66d57473ac266fb1d0ec5ad2/components/freertos/include/freertos/portable.h#L221-L236) function.

Can you make sure that your local version of `esp-idf/components/freertos/include/freertos/portable.h` contains definition of the  `xPortCanYield` function? Can you also check whether there are any other `portable.h` files in your workspace that could be interfering with the ESP-IDF's implementation of `portable.h` file? 
It could be that the `portable.h` from the FreeRTOS source clone from the `aws/amazon-freertos` repository is taking precedence in the build inclusion.

yanjos-dev, 2020-08-12:
@dev-techshlok This topic has become inactive so I'm going to close the issue. Please reopen this if you have any questions or need any further assistance.

closed 2020-08-12 by yanjos-dev