aws/amazon-freertos #2049: How to implement Microphone on FreeRTOS

cerbero-dj, 2020-05-29:
Hi everyone,
I need your help for a project I'm trying to carry out, based on STM32CubeIDE and using FreeRTOS.
I have an STM32L475 board and I would like to create a translator that, starting from the voice recorded through the board's microphone, converts the voice to text (using some online API).
Unfortunately I couldn't find much online.
Have any of you ever managed to make the microphone on this board work and make it work for this purpose?
Thank you in advance

alfred2g, 2020-05-29:
Hi @git-cerbero,
you are almost in the right spot, we do support such service with aws called Lex https://aws.amazon.com/lex/, that connects an iot device to the cloud.
Please check it out

I will be closing the issue, but if you have any further questions please reopen the issue
Regards,
Alfred

closed 2020-05-29 by alfred2g

htibosch, 2020-05-31:
Hi @git-cerbero :
It is interesting what you want to do.
In case you have technical questions about using TCP and FreeRTOS drivers, there is the [FreeRTOS forum](https://forums.freertos.org).

git-cerbero, 2020-06-08:
Hi,
@alfred2g , thank you for pointing out Lex, I wasn't aware of it.
Do you know if there are some projects for the board that I have and from which I could take inspiration?
@htibosch , thank you and I just opened a thread on the freertos forum.
I found a github repository that uses my board. [https://github.com/janjongboom/b-l475e-iot01a-audio-mbed](https://github.com/janjongboom/b-l475e-iot01a-audio-mbed)
The problem is that I can't implement it in my project, where I use FreeRTOS. This is because I have some errors due to the fact that mbedos is used.
Do you have any indications to give me in order to integrate this project into a FreeRTOS project?
Have a nice day.