aws/amazon-freertos #2067: ESP32 Port SSID Max Length only 31 bytes due to strlcpy

danahern, 2020-06-02:
On an ESP32 board trying to connect to a 32 byte SSID fails.  

We have created an access point named “iiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiK”32 ‘i' characters and a final 'K’ characters.

In this function, the function esp_wifi_scan_get_ap_records is used to fetch the scan access points, and then a loop is performed to copy records between structures.

When we print the SSID just before the strlcpy, we have the following output (in raw hex format):

I (20941) WIFI: 69 69 69 69 69 69 69 69 69 69 69 69 69 69 69 69 
I (20947) WIFI: 69 69 69 69 69 69 69 69 69 69 69 69 69 69 69 *4B*

When we print the output of the value copied in the structure right after the strlcpy, we have the following output:

W (20954) WIFI: 69 69 69 69 69 69 69 69 69 69 69 69 69 69 69 69
W (20961) WIFI: 69 69 69 69 69 69 69 69 69 69 69 69 69 69 69 *00*

https://github.com/aws/amazon-freertos/blob/master/vendors/espressif/boards/esp32/ports/wifi/iot_wifi.c#L565

Thanks!
Dan

shubhamkulkarni97, 2020-06-03:
Hi @danahern,

Have created a [PR](https://github.com/aws/amazon-freertos/pull/2073) to fix the issue.

While debugging I also observed that `sizeof` operator is used to calculate SSID length. Due to use of sizeof operator, NULL termination is also counted in the length. However, the [documentation](https://github.com/aws/amazon-freertos/blob/f264005442069bc7c5ba78ce2fa6405b144a6cd7/libraries/abstractions/wifi/include/iot_wifi.h#L110) states that length should not include NULL termination.

So, this should also be fixed in all the places.

danahern, 2020-06-04:
Fix looks good to me!

aggarw13, 2020-06-18:
The PR #2073 for the fix has been merged. 
@danahern, please let us know if you have any further questions about the issue or fix.

closed 2020-06-22 by aggarw13