aws/amazon-freertos #2107: How to establish a MQTT connection outside the demo MQTT

AbdeenM, 2020-06-06:
Device: ESP32-DevC

I am trying to connect to the AWS IOT server independent of the config header files and MQTT demo but i can not seem to be able to get it working, to start i copied the same code as the demo and configured my app to run from that file after i established a wifi connection.
After connecting to the wifi i call the function that is just the RunMqttDemo renamed, below is my modified _establishMqttConnection function code, i believe i have everything setup correctly aside from the pNetworkInterface which am not sure is correct.

Your help is highly appreciated!!

```
#define ROOT_CERTIFICATE_PEM "the root cert from aws"
#define CLIENT_CERTIFICATE_PEM "the device cert"
#define CLIENT_PRIVATE_KEY_PEM "the device private key" 

static int _establishMqttConnection( bool awsIotMqttMode,
                                     const char * pIdentifier,
                                     //void * pNetworkServerInfo,
                                     //void * pNetworkCredentialInfo,
                                     //IotNetworkInterface_t pNetworkInterface,
                                     IotMqttConnection_t *pMqttConnection )
{
    int status = EXIT_SUCCESS;
    IotMqttError_t connectStatus = IOT_MQTT_STATUS_PENDING;
    IotMqttNetworkInfo_t networkInfo = IOT_MQTT_NETWORK_INFO_INITIALIZER;
    IotMqttConnectInfo_t connectInfo = IOT_MQTT_CONNECT_INFO_INITIALIZER;
    IotMqttPublishInfo_t willInfo = IOT_MQTT_PUBLISH_INFO_INITIALIZER;
    char pClientIdentifierBuffer[ CLIENT_IDENTIFIER_MAX_LENGTH ] = { 0 };

    // Custom declares.
    IotNetworkServerInfo_t xServerInfo = { 0 };
    IotNetworkCredentials_t xCredentials = AWS_IOT_NETWORK_CREDENTIALS_AFR_INITIALIZER;
    IotNetworkInterface_t *pNetworkInterface = IOT_NETWORK_INTERFACE_AFR;

    /* Set the server info. */
    xServerInfo.pHostName = "broker-end-point.amazonaws.com";
    xServerInfo.port = 8883;

    /* Set the server certificate for a secured connection. Other credentials
     * are set by the initializer. */
    xCredentials.pRootCa = ROOT_CERTIFICATE_PEM;
    xCredentials.rootCaSize = sizeof( ROOT_CERTIFICATE_PEM );
    /* Disable SNI. */
    xCredentials.disableSni = true;
    /* ALPN is not needed. */
    xCredentials.pAlpnProtos = NULL;

    /* Strong mutual authentication to authenticate both the broker and
    * the client. */
    xCredentials.pClientCert = CLIENT_CERTIFICATE_PEM;
    xCredentials.clientCertSize = sizeof( CLIENT_CERTIFICATE_PEM );
    xCredentials.pPrivateKey = CLIENT_PRIVATE_KEY_PEM;
    xCredentials.privateKeySize = sizeof( CLIENT_PRIVATE_KEY_PEM );

    /* Set the members of the network info not set by the initializer. This
     * struct provided information on the transport layer to the MQTT connection. */
    networkInfo.createNetworkConnection = true;
    networkInfo.u.setup.pNetworkServerInfo = &xServerInfo;
    networkInfo.u.setup.pNetworkCredentialInfo = &xCredentials;
    networkInfo.pNetworkInterface = pNetworkInterface;

    #if ( IOT_MQTT_ENABLE_SERIALIZER_OVERRIDES == 1 ) && defined( IOT_DEMO_MQTT_SERIALIZER )
        networkInfo.pMqttSerializer = IOT_DEMO_MQTT_SERIALIZER;
    #endif

    /* Set the members of the connection info not set by the initializer. */
    connectInfo.awsIotMqttMode = awsIotMqttMode;
    connectInfo.cleanSession = true;
    connectInfo.keepAliveSeconds = KEEP_ALIVE_SECONDS;
    connectInfo.pWillInfo = &willInfo;

    /* Set the members of the Last Will and Testament (LWT) message info. The
     * MQTT server will publish the LWT message if this client disconnects
     * unexpectedly. */
    willInfo.pTopicName = WILL_TOPIC_NAME;
    willInfo.topicNameLength = WILL_TOPIC_NAME_LENGTH;
    willInfo.pPayload = WILL_MESSAGE;
    willInfo.payloadLength = WILL_MESSAGE_LENGTH;

    /* Use the parameter client identifier if provided. Otherwise, generate a
     * unique client identifier. */
    if( ( pIdentifier != NULL ) && ( pIdentifier[ 0 ] != '\0' ) )
    {
        connectInfo.pClientIdentifier = pIdentifier;
        connectInfo.clientIdentifierLength = ( uint16_t ) strlen( pIdentifier );
    }
    else
    {
        /* Every active MQTT connection must have a unique client identifier. The demos
         * generate this unique client identifier by appending a timestamp to a common
         * prefix. */
        status = snprintf( pClientIdentifierBuffer,
                           CLIENT_IDENTIFIER_MAX_LENGTH,
                           CLIENT_IDENTIFIER_PREFIX "%lu",
                           ( long unsigned int ) IotClock_GetTimeMs() );

        /* Check for errors from snprintf. */
        if( status < 0 )
        {
            IotLogError( "Failed to generate unique client identifier for demo." );
            status = EXIT_FAILURE;
        }
        else
        {
            /* Set the client identifier buffer and length. */
            connectInfo.pClientIdentifier = pClientIdentifierBuffer;
            connectInfo.clientIdentifierLength = ( uint16_t ) status;

            status = EXIT_SUCCESS;
        }
    }

    /* Establish the MQTT connection. */
    if( status == EXIT_SUCCESS )
    {
        IotLogInfo( "MQTT demo client identifier is %.*s (length %hu).",
                    connectInfo.clientIdentifierLength,
                    connectInfo.pClientIdentifier,
                    connectInfo.clientIdentifierLength );

        connectStatus = IotMqtt_Connect( &networkInfo,
                                         &connectInfo,
                                         MQTT_TIMEOUT_MS,
                                         pMqttConnection );

        if( connectStatus != IOT_MQTT_SUCCESS )
        {
            IotLogError( "MQTT CONNECT returned error %s.",
                         IotMqtt_strerror( connectStatus ) );

            status = EXIT_FAILURE;
        }
    }

    return status;
}
```

pvyawaha, 2020-06-08:
Hello,

The `pNetworkInterface` should be set to the `IOT_NETWORK_INTERFACE_AFR` so that change should be good. What error are you getting while connecting to the AWS IoT ? Can you please share console logs?

AbdeenM, 2020-06-09:
Thank you for the assist, i wasn't sure what pNetworkInterface was suppose to be.
Closing the issue now

closed 2020-06-09 by AbdeenM