aws/amazon-freertos #2119: Publish MQTT messages to AWS IoT Core using the Greengrass Discovery library.

BasemTarek, 2020-06-09:
****
I am using ESP-32 as my FreeRTOS device, and Raspberry Pi3 as my Greengrass Core device.

And I am trying to publish MQTT messages to Greengrass core device then to AWS IoT Core, so I used the Greengrass Discovery library to discover the Greengrass Core device and the discovery process was done successfully, but when the code reach the part of publishing the message, errors appeared.

****

I am using this project [](https://github.com/teuteuguy/amazon-freertos-workshop) to develop my application.
And we edited the _establishMqttConnection() and lab_run() functions in the lab_connection.c file to be able to discover the Greengrass Core device and to publish my messages to my topic.

Then, the Greengrass device was discovered successfully, but no messages were received and the following errors appeared:

> 54 8555 [iot_thread] [ERROR][NET][85550] Error -27648 while sending data.
> 55 8555 [iot_thread] [ERROR][MQTT][85550] (MQTT connection 0x3ffb5f44) Failed to send PINGREQ.
> 56 8555 [iot_thread] [WARN ][NET][85550] Failed to close connection.
> 

- Lab1 was enabled in the workshop configuration.
- The Thing and the topic are configured in the Greengrass group subscriptions.

- This is my edited lab_connection.c file (https://gist.github.com/BasemTarek/10e64a16cce284ac2fd579cc0504d2ec)
- This is the logs file of ESP-32 when running the code (https://gist.github.com/BasemTarek/8320eb0919d065273722cc7e87f68252)

And to make sure that there is no problem with the Greengrass Core device itself, I tried to run AWS IoT Greengrass discovery demo application on my FreeRTOS device (ESP-32), and it worked successfully.

I expected to receive a message when I subscribed to the configured topic, but this error `Failed to send PINGREQ` appeared and no message was received.

Thank you!

dachalco, 2020-06-11:
Hello @BasemTarek,

I'm unfamiliar with the Lab_1 project you've baselined, though from what I've read thus far, it appears that application uses FreeRTOS to implement a button, and publishes the state of the button to IoT Core. To double-check, this application worked as intended?

Secondly, you were able to run the [GreenGrass discovery demo](https://docs.aws.amazon.com/freertos/latest/userguide/gg-demo.html) successfully?

BasemTarek, 2020-06-22:
Thank you @dachalco, your comment helped me but I didn't catch the root cause yet, also there are an error message `54 8555 [iot_thread] [ERROR][NET][85550] Error -27648 while sending data.` which precedes PINGREQ error, I found that this error comes from this [IotNetworkAfr_Send()](https://github.com/aws/amazon-freertos/blob/4eec30ceb3fad7542298e06082f596fbeffc0eeb/libraries/abstractions/platform/freertos/iot_network_freertos.c#L475), so I am trying to catch the cause of this error.

I found that these two errors may be related to each other, because this [->send](https://github.com/aws/amazon-freertos/blob/master/libraries/c_sdk/standard/mqtt/src/iot_mqtt_operation.c#L707) actually is a call to 
this [IotNetworkAfr_Send()](https://github.com/aws/amazon-freertos/blob/4eec30ceb3fad7542298e06082f596fbeffc0eeb/libraries/abstractions/platform/freertos/iot_network_freertos.c#L475).

Also if you check the attached logs file, there is an info message `I (85847) lab_connection: LABCONNECTION_MQTT_DISCONNECTED` inform that the MQTT connection disconnected which precedes all the error messages, I had tried to find the cause of this "disconnecting action", it is likely that this disconnecting action might be the reason of the mentioned errors..!
Still don't know the cause of this disconnecting action.

muneebahmed10, 2020-06-29:
Hi @BasemTarek,

I notice that your logs indicate that the connection to your Greengrass Core [was successful](https://gist.github.com/BasemTarek/8320eb0919d065273722cc7e87f68252#file-hardcopy_logs-txt-L149). Also, the failure happens when trying to send a keep alive PINGREQ, which would happen after the expiry of the keep alive interval of 60 seconds. If no MQTT packet is being sent for 60 seconds, is your application getting stuck somewhere? 
Additionally, I'd suggest enabling debug logs, and either running the mqtt demo, or replacing the Greengrass endpoint and cert in your code with that of the AWS IoT broker, in order to determine if the issue lies with the use of Greengrass discovery, or elsewhere.

BasemTarek, 2020-07-01:
Thank you @muneebahmed10, 

> is your application getting stuck somewhere?

 sure it stuck somewhere, can you help on that..!?

> running the mqtt demo, or replacing the Greengrass endpoint and cert in your code with that of the AWS IoT broker, in order to determine if the issue lies with the use of Greengrass discovery, or elsewhere.

I don't know how this will be helpful, the Greengrass demo worked successfully, and the mqtt demo worked successfully, but separately.

BasemTarek, 2020-07-05:
Are there any updates..

muneebahmed10, 2020-07-07:
Hi @BasemTarek ,

If the MQTT and Greengrass discovery demos are each working successfully for you, then I only have enough information to offer suggestions in debugging the errors you're encountering in your implementation. 

Have you stepped through the code in a debugger, and found anywhere it's getting stuck? Could your code be waiting on a mutex or semaphore that isn't being incremented? I notice that after you call MQTT's publish, your code waits on a [clean up semaphore](https://gist.github.com/BasemTarek/10e64a16cce284ac2fd579cc0504d2ec#file-lab_connection-c-L729), that only gets posted to when there's a [LABCONNECTION_MQTT_DISCONNECTED event](https://gist.github.com/BasemTarek/10e64a16cce284ac2fd579cc0504d2ec#file-lab_connection-c-L204). Could your application be stuck there?

leegeth, 2020-07-21:
@BasemTarek ,

Did you get a chance to follow the steps mentioned in the [above comment](https://github.com/aws/amazon-freertos/issues/2119#issuecomment-655184319)? Please let us know if there are any updates and if you need any further assistance on this issue.

yanjos-dev, 2020-08-12:
@BasemTarek This topic has become inactive so I'm going to close the issue. Please reopen this if you have any questions or need any further assistance.

closed 2020-08-12 by yanjos-dev