aws/amazon-freertos #2134: Create Official FreeRTOS Release

ricky-sb, 2020-06-11:
We've not had an official FreeRTOS release for 4 months. Pre-COVID, it looked like we were on a 2-month release cycle. Is that still the case?  

I see that development is still very active, and we're merging almost daily.

rashedtalukder, 2020-06-17:
Hi @ricky-sb,

We understand the importance of a regular release schedule for members of our community to align their own testing/development schedules. We develop on FreeRTOS and the included libraries on a regular cadence (as you have seen) and will continue to do so. We have strict quality bar for our releases that must be met and the team here works relentlessly to meet and surpass that bar. Additionally, we've been hard at work to deliver our first LTS release and want to make sure we get it right for our partners and customers.

I apologize for any issues yourself and any other member of the community has felt due to us not meeting the release schedule that you saw. I hope that all the commits that you see, the support we continue to provide, and the ecosystem that continues to grow gives you confidence that we're here and we're working hard to deliver.

ricky-sb, 2020-06-17:
Thanks for the message @rashedtalukder! We really appreciate all the work the FreeRTOS team is putting into this :)

When do you expect the next release will be available?

rashedtalukder, 2020-06-17:
I can't give an exact date at the moment, and would hate to set expectations and disappoint. I realize it's not the answer you were hoping for, but many organizations setup critical schedules based on the information we release and we have to be mindful of that. So I would say just keep watching this repo and our blogs for official updates.

closed 2020-07-25 by ricky-sb