aws/amazon-freertos #2145: Using OTA agent to stream .jpg files

EtienneMdv, 2020-06-17:
Hi,

I have recently been trying to take advantage of the OTA agent features to stream .jpg files towards my ESP32 board. What I noticed is that I receive correctly all the packages. However, it seems like I am having issues writing those packets in the SPIFFS. As the packages don't arrive necessarily in order I'm writing their content at the corresponding offsets. I am using the `fwrite`function to write in the SPIFFS. Is it a good practice to write my blocks in the SPIFFS when receiving them? Is there some timing limitations between the packet "ingestion" and the reception of the following packet?

Thank you in advance for your answers.

pvyawaha, 2020-06-19:
Hello,

It should be possible to write blocks at correct offset while receiving the file blocks. The number of blocks received per request is defined by the configuration parameter `otaconfigMAX_NUM_BLOCKS_REQUEST` and the number of buffers is defined by `otaconfigMAX_NUM_OTA_DATA_BUFFERS`. If you are experiencing slower file write and dropping file blocks because of no data buffers available you can try setting these configuration parameters same.

If this is not the issue can you please share details about the issue you are facing and the logs?

pvyawaha, 2020-06-25:
Hello,

Dis changing the configurations worked for you or you still having this issue?

EtienneMdv, 2020-06-26:
Hello @pvyawaha,
Thank you very much for your help. I am now able to receive my files correctly!

pvyawaha, 2020-06-29:
To slow down the download process lowest settings for  `otaconfigMAX_NUM_BLOCKS_REQUEST`, which is number of file blocks received per request from the service is 1. The `otaconfigFILE_REQUEST_WAIT_MS` is the timeout between requests are sent and can be set to higher value like 10 Seconds. This timeout is only used when the requested blocks are not received.

If you have modified the OTA pal to use the file system , is the task stack size also updated? I suspect the crash you are observing is due to a stack overflow. You can try increasing the stack size in the config file by updating `otaconfigSTACK_SIZE`. 

More information about stack checking - 
https://www.freertos.org/uxTaskGetSystemState.html
https://www.freertos.org/Stacks-and-stack-overflow-checking.html

pvyawaha, 2020-07-08:
Hello,

Do you still having issue running this demo?

EtienneMdv, 2020-07-09:
Hello @pvyawaha,

My app is now running fine. The errors came from a poor management of the system memory.

Thank you for your help.

pvyawaha, 2020-07-10:
Thank you for sharing update that it worked for you, closing this issue now.

closed 2020-07-10 by pvyawaha