aws/amazon-freertos #2161: Storing email in NVS

AbdeenM, 2020-06-20:
Device: ESP32 DevkitC

So am trying to save an email “abdeen.mohamed@outlook.com” to the devices memory using the function nvs_set_str() with a key “user_email” but the string fails to be written to memory, meaning after I store the email and fetch it from memory and log it I receive weird symbols mostly question marks as in “??/     ????” Or something similar. Even if I restart the device and fetch the key value I get the same thing. 

Am positive the way am storing or my code is correct as I am storing other details along with the email and they are stored and fetched successfully. 

My question is, is there a different or specific way to handle storing emails? 
I suspect my issue could be because the email in question has a “.” or “@“ if that’s the cause is there a specific way to handle these characters? Because I need to store any given user email that may be supplied. 

Sorry for no code as I am on my mobile device. 

On another note, is 5 GHz WiFi bands supported currently? I have read the old forum posts about it being not supported but that was since 2018 am wondering if it is supported now. Am using AWS freertos if that is of any use knowing. 

Many thanks!!!

aggarw13, 2020-06-21:
Hi @AbdeenM, as your questions are specific about the ESP-IDF SDK, that contains the `nv_set_str()` function, and the HW specification of the board, we recommend inquiring about directly with Espressif on their [official GitHub repository](https://github.com/espressif/esp-idf).

AbdeenM, 2020-06-21:
Hi @aggarw13 thanks for the response, okay noted closing the issue now.

closed 2020-06-21 by AbdeenM