aws/amazon-freertos #2172: Race condition when running the TCP echo client demo

miteshdedhia7, 2020-06-23:
**Describe the bug**
There is a race condition in the TCP echo client demo that causes "Demo Runner" thread to finish before the "Echo" thread has finished. `DEMO_RUNNER_RunDemos` creates a thread that runs the `runDemoTask` function. `runDemoTask` then runs the `DEMO_entryFUNCTION` which in the case of the TCP echo demo is `vStartTCPEchoClientTasks_SingleTasks` which creates "Echo" thread that runs the demo and returns immediately. The "Demo Runner" thread and the "Echo" thread are at the same priority and the `runDemoTask` does not wait for the "Echo" thread to finish before calling `_cleanup` causing the network to be disabled before the "Echo" thread runs. 

Up till the 201910.00 branch the example runs fine in spite of the race condition because the `_cleanup` function did not disable the network but after https://github.com/aws/amazon-freertos/commit/4b7ca8864f8e8772d959f364c4915bfed31a3f71 the example stopped working because the network is disabled before the "Echo" thread starts communication.

**System information**
- Which hardware board or part numbers?
**CY8CPROTO_062_4343W**
- IDE used
**ModusToolbox**
- Operating System 
**MacOS**
- Version of FreeRTOS (run ``git describe --tags`` to find it)
**202002.00**
- Project 
**TCP Echo**

**Expected behavior**
The demo runs successfully with sending and receiving the message.

**Screenshots or console output**
If appropriate, please cut-and-paste the console output to explain the issue. If possible, include the call stack. 
[image]
Note that this screenshot was taken from the 201910.00 branch and hence runs the demo fine but notice that the "----DEMO FINISHED----" string printed in the `runDemoTask` appears even before the echo thread has run.

**To reproduce**
1. Run the TCP echo demo example on 202002.00 branch
2. Observe output.

Thank you!

aggarw13, 2020-06-24:
Hi @miteshdedhia7, thank you for informing us about the issue along with your analysis of the cause. We are looking into the issue, and will provide you an update soon.

AniruddhaKanhere, 2020-06-25:
Hello @miteshdedhia7, thanks for pointing out this issue.
I have created a PR https://github.com/aws/amazon-freertos/pull/2181 to resolve this race-condition. You might want to look at the PR and if you would like, try it out as well to see if your issue is resolved.
If the issue persists, tell us about it in comments below.

miteshdedhia7, 2020-06-25:
Hi @AniruddhaKanhere  Thanks for the quick fix. The changes look good and should fix the problem.

closed 2020-06-25 by miteshdedhia7