aws/amazon-freertos #2182: BLE connection is not possible with Android SDK Demo App.

tobaro-4, 2020-06-25:
Hello.

I am trying to do MQTT communication with AWS with nRF52840 Device.
<https://docs.aws.amazon.com/freertos/latest/userguide/getting_started_nordic.html>
I am doing this while looking at the guide of this link.
However, it seems that the Android SDK Demo App and BLE are not connected.

Debug Log on nRF52840 is as follows. :
```
0 0 [iot_thread] [INFO ][DEMO][0] ---------STARTING DEMO---------

The RAM must start at 200044b0, it's size must be 3bb50
1 0 [iot_thread] [INFO ][INIT][0] SDK successfully initialized.
2 2 [iot_thread] [INFO ][DEMO][20] No networks connected for the demo. Waiting for a network connection. 
3 370 [BLE] [ERROR][BLE_HAL][3700] sd_ble_gattc_exchange_mtu_request() on connection 0x0 returned busy, will retry.
4 370 [BLE] [INFO ][DEMO][3700] BLE Connected to remote device, connId = 0

5 370 [iot_thread] [INFO ][DEMO][3700] Successfully initialized the demo. Network type for the demo: 2
6 370 [iot_thread] [INFO ][MQTT][3700] MQTT library successfully initialized.
7 370 [iot_thread] [INFO ][DEMO][3700] MQTT demo client identifier is MyFMSBLE1 (length 9).
8 416 [InputTask] Numeric comparison:787566
9 416 [InputTask] Press 'y' to confirm
10 3370 [iot_thread] [ERROR][NET_BLE][33700] Failed to create BLE network connection after 30000 milliseconds.
11 3370 [iot_thread] [ERROR][MQTT][33700] Failed to establish new MQTT connection, error NETWORK ERROR.
12 3370 [iot_thread] [ERROR][DEMO][33700] MQTT CONNECT returned error NETWORK ERROR.
13 3370 [iot_thread] [INFO ][MQTT][33700] MQTT library cleanup done.
14 3370 [iot_thread] [INFO ][DEMO][33700] memory_metrics::freertos_heap::before::bytes::125128
15 3370 [iot_thread] [INFO ][DEMO][33700] memory_metrics::freertos_heap::after::bytes::96752
16 3370 [iot_thread] [INFO ][DEMO][33700] memory_metrics::demo_task_stack::before::bytes::19024
17 3370 [iot_thread] [INFO ][DEMO][33700] memory_metrics::demo_task_stack::after::bytes::18104
18 3370 [iot_thread] [ERROR][DEMO][33700] Error running demo.
19 3376 [BLE] [ERROR][BLE_HAL][33760] Failed to secure connection. Disconnecting.
20 3376 [BLE] [INFO ][DEMO][33760]  Disconnected from BLE device. Stopping the counter update 

21 3376 [iot_thread] [INFO ][INIT][33760] SDK cleanup done.
22 3376 [iot_thread] [INFO ][DEMO][33760] -------DEMO FINISHED-------
```

I think BLE connect is disconnected due to Timeout because'y' msg does not come from Android App. Is that correct?
Can you solve this problem? Please.

ravibhagavandas, 2020-06-25:
From the logs its seems the device is not receiving the input for confirmation for numeric comparison from the terminal:
```
9 416 [InputTask] Press 'y' to confirm
```
Are you using Segger embedded studio terminal ? If so the terminal does not accept input characters as mentioned in the doc:
```
Note

The Segger embedded studio terminal emulator does not support an input capability. For this, use a terminal emulator like PuTTy, Tera Term, or GNU Screen. Configure the terminal to connect to your board by a serial connection as instructed in Installing a terminal emulator.
```
As mentioned you can use other terminals like TeraTerm or Putty and press `y` to confirm numeric comparison on the device.

tobaro-4, 2020-06-26:
Thanks for your kind answer.

What I want to know is that BLE connection is possible only by receiving'Y' MSG. 
I don't understand. If this is not a problem, I would like to know why the BLE connection is not working.

ravibhagavandas, 2020-06-26:
@tobaro-4 
By default device (peripheral) is configured to run with secure connection only mode, means encrypted connection with authentication using MITM (Man in the middle protection). This requires a numeric comparison code displayed at both device side and mobile phone and user need to confirm code on both sides.
 You can choose to override this config parameters by setting Just Works Pairing mode (Unauthenticated paring with encryption) or no security as shown in this doc: https://docs.aws.amazon.com/freertos/latest/userguide/freertos-ble-library.html#freertos-ble-restrictions

closed 2020-06-29 by tobaro-4