aws/amazon-freertos #2199: Cannot run AWS FreeRTOS MQTT demo on ESP32-DevkitC : nothing is displaying on Serial Monitor after " "I (927) WIFI: SYSTEM_EVENT_STA_START""

Maliha03, 2020-06-29:
**Describe the bug**
I am trying to run the MQTT demo on ESP32. I followed this tutorial : https://docs.aws.amazon.com/freertos/latest/userguide/getting_started_espressif.html 
Its configuration is done and build files are successfully generated. But when i run the command to flash and run FreeRTOS : " ./vendors/espressif/esp-idf/tools/idf.py flash monitor -p /dev/ttyUSB0" on serial monitor it shows "Starting Demo" and get stuck at
 "I (927) WIFI: SYSTEM_EVENT_STA_START"
 and there is no output after that, nothing is displayed on the monitor after that..not even an error. I cannot figure out what kind of error it could be.
There's is another error :  E (297) PKCS11: failed nvs get file size 4354 0 

**System information**
- Hardware : ESP32- DevkitC
- Operating System Linux-Ubuntu
- Version of FreeRTOS : 202002.00

**Console Output**

ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:7216
ho 0 tail 12 room 4
load:0x40078000,len:12620
load:0x40080400,len:6708
entry 0x40080778
I (75) boot: Chip Revision: 1
I (76) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (41) boot: ESP-IDF v3.3-163-g601a03e 2nd stage bootloader
I (41) boot: compile time 00:33:37
I (41) boot: Enabling RNG early entropy source...
I (46) boot: SPI Speed      : 40MHz
I (51) boot: SPI Mode       : DIO
I (55) boot: SPI Flash Size : 4MB
I (59) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 nvs              WiFi data        01 02 00010000 00006000
I (77) boot:  1 otadata          OTA data         01 00 00016000 00002000
I (84) boot:  2 phy_init         RF data          01 01 00018000 00001000
I (92) boot:  3 ota_0            OTA app          00 10 00020000 00177000
I (99) boot:  4 ota_1            OTA app          00 11 001a0000 00177000
I (107) boot:  5 storage          WiFi data        01 02 00317000 00010000
I (114) boot: End of partition table
I (119) boot: ota rollback check done
I (123) boot_comm: chip revision: 1, min. application chip revision: 0
I (130) esp_image: segment 0: paddr=0x00020020 vaddr=0x3f400020 size=0x23504 (144644) map
I (190) esp_image: segment 1: paddr=0x0004352c vaddr=0x3ffbdb60 size=0x030e0 ( 12512) load
I (195) esp_image: segment 2: paddr=0x00046614 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at /home/maliha/RGS/freertos-1592506346911/FreeRTOS/freertos_kernel/portable/ThirdParty/GCC/Xtensa_ESP32/xtensa_vectors.S:1713

I (197) esp_image: segment 3: paddr=0x00046a1c vaddr=0x40080400 size=0x095f4 ( 38388) load
I (222) esp_image: segment 4: paddr=0x00050018 vaddr=0x400d0018 size=0xa70d0 (684240) map
0x400d0018: _flash_cache_start at ??:?

I (462) esp_image: segment 5: paddr=0x000f70f0 vaddr=0x400899f4 size=0x0ce70 ( 52848) load
0x400899f4: r_rwble_isr at ??:?

I (497) boot: Loaded app from partition at offset 0x20000
I (497) boot: Disabling RNG early entropy source...
I (498) cpu_start: Pro cpu up.
I (501) cpu_start: Application information:
I (506) cpu_start: Project name:     esp-idf
I (511) cpu_start: App version:      1
I (516) cpu_start: Compile time:     Jun 29 2020 00:29:05
I (522) cpu_start: ELF file SHA256:  32b3a7452d4b58e6...
I (528) cpu_start: ESP-IDF:          v3.3-163-g601a03e
I (534) cpu_start: Single core mode
I (538) heap_init: Initializing. RAM available for dynamic allocation:
I (545) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (551) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (557) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (563) heap_init: At 3FFBDB5C len 00000004 (0 KiB): DRAM
I (569) heap_init: At 3FFCCC68 len 00013398 (76 KiB): DRAM
I (575) heap_init: At 3FFE0440 len 0001FBC0 (126 KiB): D/IRAM
I (582) heap_init: At 40078000 len 00008000 (32 KiB): IRAM
I (588) heap_init: At 40096864 len 0000979C (37 KiB): IRAM
I (594) cpu_start: Pro cpu start user code
I (277) cpu_start: Starting scheduler on PRO CPU.
I (287) uart: queue free spaces: 10
0 1 [main] Initializing FreeRTOS TCP stack
I (297) PKCS11: Initializing NVS partition: "storage"
E (297) PKCS11: failed nvs get file size 4354 0
1 1 [main] Write certificate...
2 18 [iot_thread] [INFO ][DEMO][180] ---------STARTING DEMO---------

I (467) BTDM_INIT: BT controller compile version [9891bc6]
3 18 [iot_thread] [INFO ][INIT][180] SDK successfully initialized.
I (477) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (567) phy: phy_version: 4102, 2fa7a43, Jul 15 2019, 13:06:06, 0, 0
GAP procedure initiated: stop advertising.
GAP procedure initiated: advertise; disc_mode=2 adv_channel_map=0 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=480 adv_itvl_max=960
I (867) wifi: wifi driver task: 3ffd8fb0, prio:23, stack:3584, core=0
I (867) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (867) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (887) wifi: wifi firmware version: c94f3e6
I (887) wifi: config NVS flash: enabled
I (887) wifi: config nano formating: disabled
I (887) wifi: Init dynamic tx buffer num: 32
I (897) wifi: Init data frame dynamic rx buffer num: 32
I (897) wifi: Init management frame dynamic rx buffer num: 32
I (907) wifi: Init management short buffer num: 32
I (907) wifi: Init static rx buffer size: 1600
I (907) wifi: Init static rx buffer num: 10
I (917) wifi: Init dynamic rx buffer num: 32
I (927) wifi: mode : sta (24:0a:c4:c6:39:a0)
I (927) WIFI: SYSTEM_EVENT_STA_START

Its my first time using AWS FreeRTOS...Can anyone help? 

Thank you!

yuhui-zheng, 2020-07-01:
Would it be possible to step the code and see where it got stuck?

(May need to refer to the debug setup in the getting started guide you've linked.)

shubhamkulkarni97, 2020-07-01:
@Maliha03, This happens when WiFi credentials are not configured.

To fix the issue, you need to configure [aws_clientcredential.h](https://github.com/aws/amazon-freertos/blob/master/demos/include/aws_clientcredential.h) and [aws_clientcredential_keys.h](https://github.com/aws/amazon-freertos/blob/master/demos/include/aws_clientcredential_keys.h).

lundinc2, 2020-07-23:
Hello @Maliha03,

I am going to close this issue as there has been no response for some time. Please let us know if we can assist you further by re-opening this issue.

closed 2020-07-23 by lundinc2