aws/amazon-freertos #2211: iot defender unit test lack memory region maximum size check

TheSilentDawn, 2020-07-01:
**Describe the bug**
The function **_copyDataCallbackFunction()**, as shown in line 480-523 of aws_iot_tests_defender_system.c is used to copy data from MQTT callback to local buffer. Inside the function, it tries to copy from **pCallbackInfo->pPayload** and  **pCallbackInfo->pMetricsReport** to **_callbackInfo.pPayload** and **_callbackInfo.pMetricsReport** as shown in line 513 and 518 of aws_iot_tests_defender_system.c. However, it doesn't check the maximum size of the two buffer **_callbackInfo.pPayload** and **_callbackInfo.pMetricsReport** which pointed to two global array **_payloadBuffer** and **_metricsBuffer**.

**System information**
- NXP TWR-K64F?
- none
- Windows
- master
- Device Defender

**Expected behavior**
If using a global array as the buffer, the structure AwsIotDefenderCallbackInfo_t should define two members indicate the maximum size of **pMetricsReport** and **pPayload** or use dynamic memory allocation instead.

**To reproduce**
Steps to reproduce the behavior:

1. feed a large MQTT packet
2. debug and check the memcpy() function

**Code to reproduce the bug**
The unit test file aws_iot_tests_defender_system.c

Thank you!

ravibhagavandas, 2020-07-02:
`PAYLOAD_MAX_SIZE ` and `METRICS_MAX_SIZE`  are defined as config values which estimates maximum size required for the received buffers based on the current tests. The config values can be changed as per the test requirements.  Does having an assert on the condition `if payload size exceeds maximum defined size` before `memcpy()`, will be sufficient here ?

In real applications user can either do a dynamic allocation or estimate a maximum possible buffer size and allocate it statically if there are memory constraints.

yanjos-dev, 2020-08-12:
@TheSilentDawn This topic has become inactive so I'm going to close the issue. Please reopen this if you have any questions or need any further assistance.

closed 2020-08-12 by yanjos-dev