aws/amazon-freertos #2216: support for ti cc3200 board

taufeeq25, 2020-07-02:
hello ,

I want to add support for ti cc3200 support, but so far no idea how to do that. Could you please tell me, if its possible to add TI CC3200 board support in aws freertos, if yes please point me where to start.

Thank you!

ravibhagavandas, 2020-07-02:
Hi @taufeeq25 

FreeRTOS has a [qualified port](https://docs.aws.amazon.com/freertos/latest/userguide/getting_started_ti.html) for TI CC3220 ,the second generation of  the CC32xx series. Did you consider this as an option?

To start with a kernel port, you can refer to the [kernel porting guide ](https://www.freertos.org/FreeRTOS-porting-guide.html) for more information. Both CC3220 and CC3200 belong to same processor family (ARM_CM4), so I think this should be straightforward, you can use the existing port for CC3220 as reference. 

Also for a fully qualified amazon freertos port you need to refer this [guide ](https://github.com/aws/amazon-freertos/blob/master/doc/freertos_port_qual/FreeRTOS_Porting_Guide.pdf) for porting additional modules.

taufeeq25, 2020-07-10:
Hi @ravibhagavandas 

Thank you so much for reply and sharing porting guide.

1. Yes i do consider to use cc32xx as cc3200 belongs to the same series but so far TI have kept separate SDK for CC3200 and CC3220SF.

2. I compiled CC3220SF aws freertos SDK by changing board and build target was able to build and flash that binary in CC3200 but looks like board is not booting up.
 
In CC3200 case, if you check CC3200.cmd file which is not using flash and looks like aws have made little bit changes to this file. I have attached all .cmd file, please have a look. 

MEMORY
{
    /* Application uses internal RAM for program and data */
    SRAM_CODE (RWX) : origin = 0x20004000, length = 0x30D3F
    SRAM_DATA (RWX) : origin = 0x20037450, length = 0x19000
}

My question is that should i go with CC32xx sdk (which is not supported for CC3200 according to TI documentation) or should i start porting using CC3200 SDK?

[memory layout.zip](https://github.com/aws/amazon-freertos/files/4901552/memory.layout.zip)

Thank you.

yanjos-dev, 2020-08-13:
Hello @taufeeq25, I apologize for the late reply. 

> I compiled CC3220SF aws freertos SDK by changing board and build target was able to build and flash that binary in CC3200 but looks like board is not booting up.

This is an expected behavior because the image was built for the CC3220SF board and then flashed to the CC3200 board. This failure is likely due to the differences between these two platforms. 

To answer your question, I have two recommendations for moving forward. The first option would be to use hardware that has already been qualified and ported to the libraries in this repository. As you found, the CC3220SF board from TI is one of the qualified boards that can be used out of the box. You can find a more complete [list of the qualified boards in this link.](https://docs.aws.amazon.com/freertos/latest/userguide/freertos-hardware.html)

If you wish to use the CC3200 board, then another option is to add support for it by following the [guide for porting new boards](https://github.com/awsdocs/aws-freertos-docs/blob/master/doc_source/portingguide/index.md). The port will end up being different compared to the CCS220SF port due to the differences in the platforms. If you choose to port the board, you can use any version of the SDK that you are interested in using.

yanjos-dev, 2020-10-23:
@taufeeq25 I'm going to close this issue for now. Feel free to reopen this if you have any further questions.

closed 2020-10-23 by yanjos-dev