aws/amazon-freertos #2218: unknown type name 'CK_ATTRIBUTE' on 'remotes/origin/feature/esp32-wroom-32se' branch

paloky, 2020-07-02:
Hi.
I try to start a project with ESP32-WROOM-32SE with AWS with this tutorial link:
https://docs.aws.amazon.com/es_es/freertos/latest/userguide/getting_started_esp32wroom-32se.html

I change to "remotes/origin/feature/esp32-wroom-32se" branch.

I can compile FreeRTOS correctly without errors using the command:
`cmake -DVENDOR=espressif -DBOARD=esp32_plus_ecc608a_devkitc -DCMAKE_BUILD_TYPE=Debug flag -DCOMPILER=xtensa-esp32 -GNinja -S . -B xxxxx`

But when try to compile the Application with "ninja", I get a lot of errors in the "iot_pkcs11.h" library.
I don't find the issue.

```
C:/Users/paloky/Documents/Projectes/Claitec__0/ESP32_firmware/amazon-freertos/libraries/freertos_plus/standard/pkcs11/include/iot_pkcs11.h:156:5: error: unknown type name 'CK_ATTRIBUTE'
     CK_ATTRIBUTE xObjectClass;     /* CKA_CLASS, set to CKO_CERTIFICATE. */
     ^
C:/Users/paloky/Documents/Projectes/Claitec__0/ESP32_firmware/amazon-freertos/libraries/freertos_plus/standard/pkcs11/include/iot_pkcs11.h:157:5: error: unknown type name 'CK_ATTRIBUTE'
     CK_ATTRIBUTE xSubject;         /* CKA_SUBJECT, this parameter is required by the PKCS #11 standard. */
     ^
C:/Users/paloky/Documents/Projectes/Claitec__0/ESP32_firmware/amazon-freertos/libraries/freertos_plus/standard/pkcs11/include/iot_pkcs11.h:158:5: error: unknown type name 'CK_ATTRIBUTE'
     CK_ATTRIBUTE xCertificateType; /* CKA_CERTIFICATE_TYPE, set to CKC_X_509. */
     ^
C:/Users/paloky/Documents/Projectes/Claitec__0/ESP32_firmware/amazon-freertos/libraries/freertos_plus/standard/pkcs11/include/iot_pkcs11.h:159:5: error: unknown type name 'CK_ATTRIBUTE'
     CK_ATTRIBUTE xValue;           /* CKA_VALUE, the DER byte array of the certificate contents. */
     ^
C:/Users/paloky/Documents/Projectes/Claitec__0/ESP32_firmware/amazon-freertos/libraries/freertos_plus/standard/pkcs11/include/iot_pkcs11.h:160:5: error: unknown type name 'CK_ATTRIBUTE'
     CK_ATTRIBUTE xLabel;           /* CKA_LABEL. */
     ^
C:/Users/paloky/Documents/Projectes/Claitec__0/ESP32_firmware/amazon-freertos/libraries/freertos_plus/standard/pkcs11/include/iot_pkcs11.h:161:5: error: unknown type name 'CK_ATTRIBUTE'
     CK_ATTRIBUTE xTokenObject;     /* CKA_TOKEN. */
     ^
C:/Users/paloky/Documents/Projectes/Claitec__0/ESP32_firmware/amazon-freertos/libraries/freertos_plus/standard/pkcs11/include/iot_pkcs11.h:171:1: error: unknown type name 'CK_RV'
 CK_RV xInitializePKCS11( void );
 ^
C:/Users/paloky/Documents/Projectes/Claitec__0/ESP32_firmware/amazon-freertos/libraries/freertos_plus/standard/pkcs11/include/iot_pkcs11.h:188:1: error: unknown type name 'CK_RV'
 CK_RV xGetSlotList( CK_SLOT_ID ** ppxSlotId,
 ^
C:/Users/paloky/Documents/Projectes/Claitec__0/ESP32_firmware/amazon-freertos/libraries/freertos_plus/standard/pkcs11/include/iot_pkcs11.h:188:21: error: unknown type name 'CK_SLOT_ID'
 CK_RV xGetSlotList( CK_SLOT_ID ** ppxSlotId,
                     ^
C:/Users/paloky/Documents/Projectes/Claitec__0/ESP32_firmware/amazon-freertos/libraries/freertos_plus/standard/pkcs11/include/iot_pkcs11.h:189:21: error: unknown type name 'CK_ULONG'
                     CK_ULONG * pxSlotCount );
                     ^
C:/Users/paloky/Documents/Projectes/Claitec__0/ESP32_firmware/amazon-freertos/libraries/freertos_plus/standard/pkcs11/include/iot_pkcs11.h:200:1: error: unknown type name 'CK_RV'
 CK_RV xInitializePkcs11Session( CK_SESSION_HANDLE * pxSession );
 ^
C:/Users/paloky/Documents/Projectes/Claitec__0/ESP32_firmware/amazon-freertos/libraries/freertos_plus/standard/pkcs11/include/iot_pkcs11.h:200:33: error: unknown type name 'CK_SESSION_HANDLE'
 CK_RV xInitializePkcs11Session( CK_SESSION_HANDLE * pxSession );
                                 ^
C:/Users/paloky/Documents/Projectes/Claitec__0/ESP32_firmware/amazon-freertos/libraries/freertos_plus/standard/pkcs11/include/iot_pkcs11.h:208:1: error: unknown type name 'CK_RV'
 CK_RV xInitializePkcs11Token( void );
 ^
C:/Users/paloky/Documents/Projectes/Claitec__0/ESP32_firmware/amazon-freertos/libraries/freertos_plus/standard/pkcs11/include/iot_pkcs11.h:226:1: error: unknown type name 'CK_RV'
 CK_RV xFindObjectWithLabelAndClass( CK_SESSION_HANDLE xSession,
 ^
C:/Users/paloky/Documents/Projectes/Claitec__0/ESP32_firmware/amazon-freertos/libraries/freertos_plus/standard/pkcs11/include/iot_pkcs11.h:226:37: error: unknown type name 'CK_SESSION_HANDLE'
 CK_RV xFindObjectWithLabelAndClass( CK_SESSION_HANDLE xSession,
                                     ^
C:/Users/paloky/Documents/Projectes/Claitec__0/ESP32_firmware/amazon-freertos/libraries/freertos_plus/standard/pkcs11/include/iot_pkcs11.h:228:37: error: unknown type name 'CK_OBJECT_CLASS'
                                     CK_OBJECT_CLASS xClass,
                                     ^
C:/Users/paloky/Documents/Projectes/Claitec__0/ESP32_firmware/amazon-freertos/libraries/freertos_plus/standard/pkcs11/include/iot_pkcs11.h:229:37: error: unknown type name 'CK_OBJECT_HANDLE_PTR'
                                     CK_OBJECT_HANDLE_PTR pxHandle );
                                     ^
C:/Users/paloky/Documents/Projectes/Claitec__0/ESP32_firmware/amazon-freertos/libraries/freertos_plus/standard/pkcs11/include/iot_pkcs11.h:250:1: error: unknown type name 'CK_RV'
 CK_RV vAppendSHA256AlgorithmIdentifierSequence( uint8_t * x32ByteHashedMessage,
 ^
In file included from C:/Users/paloky/Documents/Projectes/Claitec__0/ESP32_firmware/amazon-freertos/libraries/abstractions/pkcs11/ecc608a/iot_pkcs11_secure_element.c:39:0:
C:/Users/paloky/Documents/Projectes/Claitec__0/ESP32_firmware/amazon-freertos/libraries/abstractions/pkcs11/include/iot_pkcs11_pal.h:36:1: error: unknown type name 'CK_OBJECT_HANDLE'
 CK_OBJECT_HANDLE PKCS11_PAL_SaveObject( CK_ATTRIBUTE_PTR pxLabel,
 ^
C:/Users/paloky/Documents/Projectes/Claitec__0/ESP32_firmware/amazon-freertos/libraries/abstractions/pkcs11/include/iot_pkcs11_pal.h:36:41: error: unknown type name 'CK_ATTRIBUTE_PTR'
 CK_OBJECT_HANDLE PKCS11_PAL_SaveObject( CK_ATTRIBUTE_PTR pxLabel,
                                         ^
C:/Users/paloky/Documents/Projectes/Claitec__0/ESP32_firmware/amazon-freertos/libraries/abstractions/pkcs11/include/iot_pkcs11_pal.h:43:1: error: unknown type name 'CK_RV'
 CK_RV PKCS11_PAL_DestroyObject( CK_OBJECT_HANDLE xHandle );
 ^
C:/Users/paloky/Documents/Projectes/Claitec__0/ESP32_firmware/amazon-freertos/libraries/abstractions/pkcs11/include/iot_pkcs11_pal.h:43:33: error: unknown type name 'CK_OBJECT_HANDLE'
 CK_RV PKCS11_PAL_DestroyObject( CK_OBJECT_HANDLE xHandle );
                                 ^
C:/Users/paloky/Documents/Projectes/Claitec__0/ESP32_firmware/amazon-freertos/libraries/abstractions/pkcs11/include/iot_pkcs11_pal.h:48:1: error: unknown type name 'CK_OBJECT_HANDLE'
 CK_OBJECT_HANDLE PKCS11_PAL_FindObject( uint8_t * pLabel,
 ^
C:/Users/paloky/Documents/Projectes/Claitec__0/ESP32_firmware/amazon-freertos/libraries/abstractions/pkcs11/include/iot_pkcs11_pal.h:56:39: error: unknown type name 'CK_OBJECT_HANDLE'
 BaseType_t PKCS11_PAL_GetObjectValue( CK_OBJECT_HANDLE xHandle,
                                       ^
C:/Users/paloky/Documents/Projectes/Claitec__0/ESP32_firmware/amazon-freertos/libraries/abstractions/pkcs11/include/iot_pkcs11_pal.h:59:39: error: unknown type name 'CK_BBOOL'
                                       CK_BBOOL * xIsPrivate );
```

There is someone who has used the ESP32-Wroom-32SE?
You could help me.
Thank's a lot.

lundinc2, 2020-07-02:
Hello @paloky,

The ESP32-Wroom-32SE is now available on the "release-candidate" branch and tested nightly on that branch. My recommendation is to use the release-candidate until an official release is available that supports the ESP32-Wroom-32SE.

I will investigate the issue on the feature branch and comment here when a fix is available.

lundinc2, 2020-07-02:
Hello @paloky,

Unfortunately I was unable to reproduce the issue. 

Are you able to build with the regular ESP32? Swap -DBOARD=esp32_plus_ecc608a_devkitc to -DBOARD=esp32_devkitc to compile the other board. 

The errors are for symbols defined in "pkcs11.h" that appear to be missing. Can you confirm that "libraries/3rdparty/pkcs11/pkcs11.h" exists?

paloky, 2020-07-03:
Hi Carl. (@lundinc2 )
Thanks a lot.  With the "release-candidate" branch I can success compile the examples.

closed 2020-07-03 by paloky

lundinc2, 2020-07-06:
I'm happy to hear that @paloky. Once a tagged release has been created that includes this board, I will update here.

lundinc2, 2020-07-20:
Hello Paloky,

There is now a tagged release with support for the ESP32-WROOM-32SE https://github.com/aws/amazon-freertos/releases/tag/202007.00.