aws/amazon-freertos #2229: Can't reconnect to broker after disconnecting

gearsmotion789, 2020-07-04:
**Describe the bug**
I noticed that when I disconnect power to the Wifi, the device attempts to reconnect to the broker. When I reconnect power to the Wifi, the device still tries to attempt to reconnect to the broker but never does. I'm not sure how to fix this. I am using the OTA demo. Here is the log output from when the wifi disconnects. (Note: I've manually changed the retry delay to 10s)

```
85 35182 [iot_thread] [INFO ][DEMO][351820] State: WaitingForJob  Received: 1   Queued: 0   Processed: 0   Dropped: 0

86 35682 [iot_thread] [INFO ][DEMO][356820] State: WaitingForJob  Received: 1   Queued: 0   Processed: 0   Dropped: 0

87 36182 [iot_thread] [INFO ][DEMO][361820] State: WaitingForJob  Received: 1   Queued: 0   Processed: 0   Dropped: 0

88 36682 [iot_thread] [INFO ][DEMO][366820] State: WaitingForJob  Received: 1   Queued: 0   Processed: 0   Dropped: 0

89 36723 [iot_thread] [ERROR][NET][367230] Error -27648 while sending data.
90 36723 [iot_thread] [ERROR][MQTT][367230] (MQTT connection 0x3ffb9114) Failed to send PINGREQ.
91 36723 [iot_thread] [WARN ][NET][367230] Failed to close connection.
92 36723 [iot_thread] [INFO ][MQTT][367230] (MQTT connection 0x3ffb9114) Network connection closed.
93 36723 [iot_thread] [INFO ][DEMO][367230] Mqtt disconnected due to Keep-alive response not received.

94 37182 [iot_thread] [INFO ][DEMO][371820] State: WaitingForJob  Received: 1   Queued: 0   Processed: 0   Dropped: 0

95 37182 [OTA Agent Task] [prvSuspendHandler] OTA Agent is suspended.
96 37182 [OTA Agent Task] [prvOTAAgentTask] Called handler. Current State [WaitingForJob] Event [Suspend] New state [Suspended] 
97 37682 [iot_thread] [INFO ][DEMO][376820] Retrying network connection in 10 Secs 
98 38682 [iot_thread] [INFO ][DEMO][386820] Connecting to broker...

99 38682 [iot_thread] [INFO ][DEMO][386820] MQTT demo client identifier is 240ac450c454 (length 12).
100 39313 [iot_thread] [ERROR][NET][393130] Failed to resolve <aws iot endpoint>.
101 39313 [iot_thread] [ERROR][MQTT][393130] Failed to establish new MQTT connection, error NETWORK ERROR.
102 39313 [iot_thread] [ERROR][DEMO][393130] MQTT CONNECT returned error NETWORK ERROR.
103 39313 [iot_thread] [ERROR][DEMO][393130] ERROR:  MQTT_AGENT_Connect() Failed.

104 39313 [iot_thread] [INFO ][DEMO][393130] Retrying network connection in 10 Secs 
105 40313 [iot_thread] [INFO ][DEMO][403130] Connecting to broker...

106 40313 [iot_thread] [INFO ][DEMO][403130] MQTT demo client identifier is 240ac450c454 (length 12).
107 41013 [iot_thread] [ERROR][NET][410130] Failed to resolve <aws iot endpoint>.
108 41013 [iot_thread] [ERROR][MQTT][410130] Failed to establish new MQTT connection, error NETWORK ERROR.
109 41013 [iot_thread] [ERROR][DEMO][410130] MQTT CONNECT returned error NETWORK ERROR.
110 41013 [iot_thread] [ERROR][DEMO][410130] ERROR:  MQTT_AGENT_Connect() Failed.

111 41013 [iot_thread] [INFO ][DEMO][410130] Retrying network connection in 10 Secs
112 42013 [iot_thread] [INFO ][DEMO][420130] Connecting to broker...

113 42013 [iot_thread] [INFO ][DEMO][420130] MQTT demo client identifier is 240ac450c454 (length 12).
114 42713 [iot_thread] [ERROR][NET][427130] Failed to resolve <aws iot endpoint>.
115 42713 [iot_thread] [ERROR][MQTT][427130] Failed to establish new MQTT connection, error NETWORK ERROR.
116 42713 [iot_thread] [ERROR][DEMO][427130] MQTT CONNECT returned error NETWORK ERROR.
117 42713 [iot_thread] [ERROR][DEMO][427130] ERROR:  MQTT_AGENT_Connect() Failed.

118 42713 [iot_thread] [INFO ][DEMO][427130] Retrying network connection in 10 Secs
```

**System information**
- Which hardware board or part numbers?
ES32 DevKitCv4
- IDE used
VSCode
- Operating System [Windows|Linux|MacOS]
Windows
- Version of FreeRTOS (run ``git describe --tags`` to find it)
release, release-candidate
- Project [Custom Application|MQTT|Shadow|Greengrass|OTA|Device Defender|TCP Echo|Sub pub Demo|Simple TCP Server]
OTA

**Expected behavior**
When disconnecting from the Wifi network, the device should try reconnecting to the network and then reconnecting with the MQTT broker.

**To reproduce**
1. Setup and install the OTA sample
2. Wait for the device to connect to Wifi and wait for OTA updates
3. Disconnect the Wifi power
4. Wait until the device lost connection
5. Reconnect the Wifi network
6. The device doesn't reconnect

Thank you!

ravibhagavandas, 2020-07-05:
@gearsmotion789 
From the logs, it seems that after wifi powers back up, the OTA application is trying to  re-establish TCP connection but underlying WIFI connectivity is still not established. For establishing WiFi connectivity you can listen to WiFi network down events from network manager and trigger a re-connection to your WiFi network using `WIFI_ConnectAP` API. A sample task which listens to network disabled events from network manager and connects to provisioned WiFi networks in flash, is shown in this demo: https://github.com/aws/amazon-freertos/blob/master/demos/wifi_provisioning/aws_wifi_connect_task.c .

gearsmotion789, 2020-07-07:
@ravibhagavandas Thank you very much. It works now. The problem indeed was that the device didn't reconnect to Wifi before attempting to reconnect to the broker. I implemented a Wifi callback in order to reconnect back to the Wifi, like the code you linked.

ravibhagavandas, 2020-07-08:
Glad that it works.
I am closing this.
Please open a new issue if you have any questions.

closed 2020-07-08 by ravibhagavandas