aws/amazon-freertos #2244: AP mode Provisioning

gearsmotion789, 2020-07-08:
**Describe the bug**
I wanted to know if the amazon-freertos has a function to start an ESP32 in AP mode provisioning, like the example from https://github.com/espressif/esp-idf/tree/release/v4.2/examples/provisioning/legacy/softap_prov. I know that there is a function called "WIFI_Provision()" in iot_wifi.c that allows for smart config provisioning, where the device is put into station mode. However, this mode only works if the user is connected to a 2.4G network. So I need AP mode provisioning as another option to provision the device. How can I do this with the amazon-freertos?

**System information**
- Which hardware board or part numbers?
ESP32DevKitCv4
- IDE used
VSCode
- Operating System [Windows|Linux|MacOS]
Windows
- Version of FreeRTOS (run ``git describe --tags`` to find it)
release, release-candidate

Thank you!

ravibhagavandas, 2020-07-08:
@gearsmotion789 
Up untill now we only support WiFi provisioning over BLE using a companion mobile app. Vendors like Espressif have added support in their SDK through these PRs PR 1519 and PR 1571, but we don't have a formal demo showing WiFi provisioning using soft AP on all platforms. We will consider this request into prioritizing support for soft AP support in our WiFi abstraction.

ricky-sb, 2020-09-15:
>we don't have a formal demo showing WiFi provisioning using soft AP on all platforms  

I understand that there's no formal demo, but does the code already exist in AWS FreeRTOS to allow this?