aws/amazon-freertos #2269: WiFi Provisioning Service - CBOR parser interprets NULL character in payload and throws Unexpected End of File error

rrivadeneira-rsc, 2020-07-14:
**Describe the bug**
This error (CborErrorUnexpectedEOF) seems to occur for any `00` (or NULL) character that may be present in a CBOR payload to parse. I specifically found this when using the **WiFi provisioning service** and setting up a WiFi profile within our application with an Arris TG862G router. Its MAC is: 00:1D:D3:8F:84, in which the 00 octet is interpreted as a NULL character by the CBOR decoder (byte string), causing it to exit and not finish the add network request.

On our application, we were seeing a debugging error, in which the SSID (the first value the deserializer tries to find from a `deserializeAddNetworkRequest`) could not be interpreted:
`88 8594 [Btc_task] Failed to get SSID parameter, error = 7, value type = 0`

This in turn led us to see we were seeing an IOT_SERIALIZER_INTERNAL_FAILURE (7), prompted on what we suspected was a parsing error rather than finding the value of the SSID. Upon further digging, we see this error stems from the function `cbor_value_map_find_value` from the IoT Serializer function `_find` under `iot_serializer_tinycbor_decoder.c`. When successful, given a decoder object and key, this function returns a success and a decoder object with the respective value OR a "not found" error. However, this operation fails and returns a CborErrorUnexpectedEOF on the CBOR call, and finally an IOT_SERIALIZER_INTERNAL_FAILURE on the _find function. This ends on an error on the characteristic, where deserialization stops and request to add network is not processed.

Due to this deserialization/parsing error, no networks with a `00` BSSID octet are able to successfully WiFi provision, as the request is never successfully processed.

**System information**
- Which hardware board or part numbers? - ESP32-WROOM-32
- IDE used - Eclipse
- Operating System - MacOS
- Version of FreeRTOS (run ``git describe --tags`` to find it) - **202002.00-12-g446f9d081** <- This commit hash is a CUSTOM version derived from commit hash 45021e9f91e43 on FreeRTOS
- Project - Custom Application
- If your project is a Custom Application, please add the relevant code snippet in the section `Code to reproduce the bug`.

**Expected behavior**
A CBOR payload for a WiFi network profile _will_ see `00` characters. For instance, the BSSID / MAC address of the network may contain one or more `00` octets. Many router manufacturers may contain octets within that range [as seen here](https://gist.github.com/aallan/b4bb86db86079509e6159810ae9bd3e4).

_Moreover_, if the WiFi network profile is an open network, the option for an open security network is enum 0 (int) - which is also interpreted as a `00` character by the CBOR parser.

**Screenshots or console output**
Looking at a [CBOR decoder](http://cbor.me/), I took the binary input of the CBOR encoded payload our mobile application gives the firmware to interpret:
```
A5 61 62 46 00 1D D3 8F 84 30 61 67 20 61 6D 6C 44 47 38 36 30 41 38 46 38 34 33 32 61 71 03 61 72 68 44 47 38 36 30 41 33 32
```
And confirming this is a valid CBOR encoded payload that is successfully decoded as follows **(NOTE: "b" = BSSID _byte string_, "g" = index _int_, "m" = network psk _string_, "q" = network type _int_, "r" = network SSID _string_)**:
```
{"b": h'001DD38F8430', "g": -1, "m": "DG860A8F8432", "q": 3, "r": "DG860A30"}
```
 As you see, the 5th character of the binary output above is where the 00 octet / character is located.

**To reproduce**
In order to reproduce, use the WiFi provisoning service and the Save Network characteristic with:
1. A demo WiFi provisioning application 
2. A network with these characteristics (`00` octet BSSID), OR
3. Force a BSSID `00` octet by modifying the networks sent by the List Network characteristic to send , and the `_serializeNetwork()` 

**Code to reproduce the bug**
As per item (3), modify the `_serializeNetwork()` function and force a BSSID `00` octet to:
```c
if( IS_VALID_SERIALIZER_RET( ret, pBuffer ) )
{
    value.type = IOT_SERIALIZER_SCALAR_BYTE_STRING;
    value.value.u.string.pString = ( uint8_t * ) pNetworkInfo->pBSSID;
    *(value.value.u.string.pString) = 0x00; // or + n if trying to force on any other octet
    value.value.u.string.length = pNetworkInfo->BSSIDLength;
    ret = IOT_BLE_MESG_ENCODER.appendKeyValue( &networkMap, IOT_BLE_WIFI_PROV_BSSID_KEY, value );
}
```
This will replicate networks with a `00` BSSID octet, and the provisioning application will take that BSSID and return it back to the firmware to parse through the Add Network characteristic

**Additional context**
Add any other context about the problem here.

Thank you!

leegeth, 2020-07-21:
@rrivadeneira-rsc ,

Thanks for bringing this issue to our attention and the detailed steps for reproducing the issue. We are looking into this issue and will get back to you as soon as we have some updates.

lundinc2, 2020-08-03:
Hello @rrivadeneira-rsc, to provide a quick update, I verified that the value was getting copied properly by TinyCBOR using the unit tests on visual studio. I did not have access to my Espressif board but I will continue to debug now that I have it.

rrivadeneira-rsc, 2020-08-03:
Thank you for the update @lundinc2. Looking forward to further tests and see what you find. Let me know if you need anything from me.

lundinc2, 2020-08-07:
Hi @rrivadeneira-rsc sorry for the late reply. I was unable to reproduce this issue, using said Hash and the iOS app. 

I was able to decode the payload written above on my device, as well as manually adding some 0x00 hex chars to the BSSID. 

The logs mention an error in the SSID, are there any special characters in your SSID? In the above cbor they seem ASCII based to me, but I am curious if there is something missing regarding the SSID that the CBOR is failing to decode. 

One thing I did find is the iOS app will crash when trying to use a UTF-8 encoded SSID, so I will create a patch for that soon.

sarenameas, 2020-09-20:
Hello @rrivadeneira-rsc ,

Are you still having issues? Could you follow up with my colleague @lundinc2's post?

sarenameas, 2020-09-21:
Hello @rrivadeneira-rsc ,

I am going to close this ticket for now. Please feel free to reopen or open a new issue if you are still having problems.

Thanks,
Sarena

closed 2020-09-21 by sarenameas