aws/amazon-freertos #2276: MIT License and future possibility of removal of MIT

amitkulkarnis, 2020-07-15:
Hi,

This is a general question on MIT license on amazon-freertos.
As of today amazon-freertos is open-source with MIT license, is there a possibility that Amazon would remove the MIT lincense and put a commercial license (making it close-source) on the amazon-freertos in the future? 
or will the current MIT license protect it from doing so?

Thanks.

leegeth, 2020-07-20:
@amitkulkarnis ,
Thanks for bringing this question to our attention.  There is no plan to change the MIT license for amazon-freertos. Hope this answered this your question.

lundinc2, 2020-07-28:
Hello @amitkulkarnis,

I will close your issue, as I assume it is answered. Please re-open this issue if you have further tickets regarding licenses

closed 2020-07-28 by lundinc2